chan-sccp/chan-sccp #495: "Temp Fail" displayed when call is connected

mpaland, 2019-11-28:
Tested with asterisk 16.6.1 and latest dev chan-sccp as of today.

I have a button as line monitor for a group of other phones belonging to Manfred on my desk phone like:
`button = speeddial, Manfred, 12, 81@ext-hints`

hint 81 in the dialplan is like:
`exten => 81,hint,PJSIP/11&SCCP/12&SCCP/13`

When a call is coming in on the this line it shows in my display:
2132232 ->Manfred

When the call is connected it shows:
Manfred
or
22123322 <=> Manfred
or
"Temp Fail" <=> Manfred

No number is displayed when the call is handled by the PJSIP phone, but I saw "Temp fail" a few times on asterisk 16 lately and am wondering why?

When a call is going out, it always shows:
323232 <- Manfred (when SCCP phone is used)
and just
Manfred (when PJSIP phone is used)

dkgroot, 2019-11-28:
Ooh that is not good. To be honest i did a little bit of 'optimization' in the sccp_hint.c file a couple of 1-2 weeks ago. But i don't remember if i checked the result, i just looked logical. I will look it up and revert the change and give it a look. Sorry about this  and thanks for reporting the issue.

dkgroot, 2019-11-28:
Reverted commit 8239b6d
Please let me know if this took care of the problem, assuming it does.

mpaland, 2019-11-29:
I reviewed your changes, all the fall-through optimizations should be fine. Must be some other change.
Anyway I retest the actual version.

By the way, is the explicit number display configurable?
Sometimes, it would be better not to expose the call details on a monitoring phone, the status of the hint (LED, Line symbol) would be sufficient.

dkgroot, 2019-11-29:
> I reviewed your changes, all the fall-through optimizations should be fine. Must be some other change.

That's what i thought as well. It was just supposed to be a little cleanup.

> By the way, is the explicit number display configurable?

Eeh yes, but currently it's an all or nothing `./configure --disable-dynamic-speeddial-cid` option. It could be made into a per line, per device or per button kind solution, but that is not present at this time. Would need some discussion and work i guess.

dkgroot, 2019-11-29:
@mpaland 
Had another look at the current code. There is the device->privacy sccp.conf entry. It controls hint cid information. privacy=full means all call info is hidden on remote hinted speeddials, privacy=on means it can be controlled using the 'private' softkey. privacy=off means call info is always shared. I forgot this was added something like 2 years ago. Maybe having this at the device level is not 100% what you were looking for, but it is a start i guess.

commit 4e54fc9c (2019-11-28): Revert 8239b6d "Minimal potential change in hint processing (regarding SCCP_CHANNELSTATE_BUSY)"

Issue with hints reported by mpaland
Fixes: #495
closed 2020-04-07 by dkgroot