chan-sccp/chan-sccp #496: Unexpected phones disconnect

ErmakovDmitriy, 2019-12-05:
Hi, everybody.
I noticed the problem some weeks ago. Sometimes (no obvious reason) Cisco phones disconnect/loose session from Asterisk server.
Some weeks ago we (my colleague and me) thought that the situation was related to some incorrect logic during config reload (we thought that it happened just after "sccp reload").
Today we saw the problem without any obvious reason.

**Reproduction Steps:**

1. Install FreePBX
2. Install chan-sccp driver
3. Configure Cisco phones to work with sccp
4. Wait for some time

**chan-sccp version:** Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 1c5cc89 (built by 'root' on '2019-11-05 11:52:58 UTC')

**asterisk version:** Asterisk 13.27.1 built by mockbuild @ jenkins7 on a x86_64 running Linux on 2019-09-06 21:11:59 UTC

**Expected behavior:**

Phones stay online, update their registration without any errors.

**Observed behavior:**

Phones can not register and the Asterisk server writes error in log:

`
[2019-12-05 10:32:00] VERBOSE[34368] :     -- <?xml version="1.0" encoding="UTF-8"?><x-cisco-alarm><Alarm Name="LastOutOfServiceInformation"><ParameterList><String name="DeviceName">SEP6886a7431fa7</String><String name="DeviceIPv4Address">10.23.16.46/32</String><String name="IPv4DefaultGateway">10.23.16.1</String><String name="DeviceIPv6Address"></String><String name="IPv6DefaultGateway"></String><String name="ModelNumber">CP-6921</String><String name="NeighborIPv6Address"></String><String name="NeighborIPv4Address">10.23.32.13</String><String name="NeighborDeviceID">CO-SW03.ACT.LOCAL</String><String name="NeighborIPortID">FastEthernet0/41</String><Enum name="DHCPv4Status">1</Enum><Enum name="DHCPv6Status">1</Enum><Enum name="TFTPCfgStatus">1</Enum><Enum name="DNSStatusCUCM1">4</Enum><String name="VoiceVLAN">2016</String><String name="CUCMIPAddress">10.93.1.21</String><String name="LocalPort">35906</String><String name="TimeStamp">1575541888</String><Enum name="ReasonForOutOfService">17</Enum><String name="LastProtocolEventSent">0x0</String><String name="LastProtocolEventReceived">0x100</String></ParameterList></Alarm></x-cisco-alarm>
[2019-12-05 10:32:00] VERBOSE[34368] :  SEP6886A7431FA7: is registering, Instance: 1, UserId: 0, Type: Cisco 6921 (495), Version: 22 (loadinfo 'SCCP69xx.9-3-1-3')
[2019-12-05 10:32:00] WARNING[34368] sccp_actions.c: SEP6886A7431FA7: Cleaning previous session, come back later, state:OK
`
The affected phones are not able to register until somebody restarts Asterisk.
The problem does not affect all phones instantly (it starts from some set of phones and then spreads).

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: No (It started after Asterisk update from major 11 version to 13 during FreePBX migration from legacy (based on CentOS 6) to actual (CentOS 7 based) version.
- Problem can be reliably reproduced, doesn't happen randomly: No
- Platform:
  - Intel(R) Xeon(R) CPU E5-2620 0 @ 2.00GHz ( VM on Microsoft Hyper-V server)
  - 64 Bit
  - Small Endian
  - Operating System: FreePBX Distro (CentOS 7 based)
  - Operating System Version: Sangoma Linux release 7.5.1805 (Core)
- Direct Contact information: demonihin@gmail.com
- Log Files:  [Asterisk log](https://pastebin.com/xjR2Nn4w);  [phone status when Ok](https://pastebin.com/e62wpVxn)

dkgroot, 2019-12-05:
@ErmakovDmitriy
Thank you for you issue report. The Unexpected Disconnect and Register has been in chan_sccp for quite some time, and we have not been able to figure out the real reason for this yet. We have made a little progress, fixing the looping behaviour after the initial disconnect in Issue #488. I have however not hear back from @rofferman yet on the latest findings. But his initial results were positive. Maybe you can switch to the issues/488 branch and help me check that the 'looping behaviour' does not occur with this branch.

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/488
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

dkgroot, 2019-12-05:
@rofferman just reported back on the issue/488 that it looks like it fixed the devices getting stuck in an infinite loop. So merging issue/488 back to develop. You can pull the latest develop branch in 10 min.

dkgroot, 2019-12-05:
Closing the issue report for now. If the issue does continue please reopen this one or issue/488

dkgroot, 2019-12-05:
Closing this issue report for now.
If the issue does come back / is not resolved, then please reopen this issue or issue/488.

ErmakovDmitriy, 2019-12-10:
Thank you @dkgroot. 

I tested the changes you offered.
It worked from Friday evening to Tuesday morning (its important to notice that my PBX restarts every night at 00:00).

Today my colleague got the same problem.
Here is the [log and status](https://pastebin.com/uZZxs43S).

`sccp show ...` outputs are stripped to include only some phones (for which I got "Cleaning previous session, come back later, state:OK" log records) but I can send you full output (in a private email, if possible).

dkgroot, 2019-12-10:
@ErmakovDmitriy Darn. I was hoping it was resolved !
It looks like the stuck channels are the cause for the continues restarts. We are having trouble cleaning them up, and getting in this loop
You can send mail to ddegroot@talon.nl if you like.

Please include 'sccp show refcount' as well if you still have it.
I will review the log.

ErmakovDmitriy, 2019-12-10:
Thank you. I have just sent full log to ddegroot@talon.nl

dkgroot, 2019-12-10:
Thanks i will dig through it tomorrow !

dkgroot, 2019-12-11:
Regarding: I"SCCP11.8-4-2S : Invalid SCCP message! : ID :87:"
Please read #424 issue report.
This phones problem seems to be solved by changing the phones firmware.

dkgroot, 2019-12-11:
Problem 2: "Debug Log"
`sccp_actions.c: SEP6886XXXXXXX: Cleaning previous session, come back later, state:**OK**`

The state "OK" is causing an endless loop. Fix: We change the registration state to FAILED during cleanup and disconnect the session from the device forcefully. Committed a fix in a separate branch:

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/496
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

ErmakovDmitriy, 2019-12-11:
Thank you for the fix.
I will try it as soon as possible.

ErmakovDmitriy, 2019-12-17:
Thank you @dkgroot for your help!

I compilled issues/496 branch about 5 days ago.
I have not noticed any problems with sessions since the update.

Now I can see:

```shell
[2019-12-17 06:10:09] VERBOSE[2451] :     -- SCCP: Alarm Message: Severity: Informational (2), 10: Name=SEP3037A617E137Load= SCCP11.8-4-2S Last=TCP-timeout [0/0]
[2019-12-17 06:10:09] VERBOSE[2451] :  SEP3037A617E137: is registering, Instance: 0, UserId: 0, Type: Cisco 7906 (369), Version: 17 (loadinfo 'SCCP11.8-4-2S')
[2019-12-17 06:10:09] WARNING[2451] sccp_session.c: Session(0x7fbd5c00ac60) and Device Session(0x7fbd5c014130) are out of sync.
[2019-12-17 06:10:09] WARNING[2451] sccp_actions.c: SEP3037A617E137: Cleaning previous session, come back later, state:OK
[2019-12-17 06:10:09] VERBOSE[2451] :   == 10.93.1.21:52635: Session 0x7fbd5c014360 needs to be closed!
[2019-12-17 06:10:09] VERBOSE[2451] :     -- SEP3037A617E137: Registration State is 'Failed'
```

and after that the phone registers.

```shell
kdast03*CLI> core show uptime
System uptime: 11 hours, 38 minutes, 28 seconds
Last reload: 11 hours, 38 minutes, 28 seconds

kdast03*CLI> sccp show device SEP3037A617E137

--- SCCP channel driver device settings ----------------------------------------------------------------------------------
MAC-Address                                    : SEP3037A617E137
Protocol Version                               : Supported '17', In Use '17'
Protocol In Use                                : SCCP Version 17
Device Features                                : 0x7285,000000000111001010000101
Tokenstate                                     : None
Keepalive                                      : 60
Registration state                             : OK
State                                          : On Hook
Addons                                         : 
MWI state                                      : off (0/0)
MWI light-type                                 : On
MWI During call                                : turn off
Description                                    : #608
Config Phone Type                              : 7906
Skinny Phone Type                              : Cisco 7906(369)
Softkey support                                : yes
Softkeyset                                     : default => default (0x41e7110)
BTemplate support                              : yes
linesRegistered                                : yes
Image Version                                  : SCCP11.8-4-2S
Timezone Offset                                : -1
Audio Capabilities                             : [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a]
Audio Preferences                              : [ulaw/64k, ulaw/56k, alaw/64k, alaw/56k]
Audio TOS                                      : 184
Audio COS                                      : 6
Video TOS                                      : 136
Video COS                                      : 5
DND Feature enabled                            : yes
DND Status                                     : Disabled
DND Action                                     : Silent
Can Transfer                                   : on
Can Park                                       : on
Can CFWDALL                                    : on
Can CFWBUSY                                    : off
Can CFWNOANSWER                                : off
Allow ringin notification                      : no
Private softkey                                : off
Dtmf mode                                      : RFC2833
Nat                                            : (Auto)Off
Videosupport?                                  : no
Direct RTP                                     : off
Trust phone ip (deprecated)                    : off
Bind Address                                   : 10.53.4.20:30843
Server Address                                 : 10.93.1.21:58487
[... 26 lines cut ...]
+--- SpeeddialButtons -------------------------------------------------------------------------------------------------+
| Id   Name                    Number                 Hint                                                             |
+ ==== ======================= ====================== ================================================================ +
+----------------------------------------------------------------------------------------------------------------------+

+--- FeatureButtons ---------------------------------------------------------------------------------------------------+
| Id   Name                    Status                 Options                                                          |
+ ==== ======================= ====================== ================================================================ +
+----------------------------------------------------------------------------------------------------------------------+

+--- ServiceURLButtons ------------------------------------------------------------------------------------------------+
| Id   Name                   URL                                                                                      |
+ ==== ====================== ======================================================================================== +
+----------------------------------------------------------------------------------------------------------------------+

+--- CallStatistics ---------------------------------------------------------------------------------------------------+
[... cut ...]
+ ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== ======== +
| LAST     0        0        0        0        0        0        0.000000 0.000000 0.000000 0.000000 0.000000 0        |
| AVG      0        0        0        0        0        0        0.000000 0.000000 0.000000 0.000000 0.000000 0        |
+----------------------------------------------------------------------------------------------------------------------+

```

I believe that the problem is solved.

dkgroot, 2019-12-17:
I have merged issues/496 back into the develop branch.
Thanks you for reporting the issue and testing the solution.

closed 2019-12-17 by dkgroot

duartero84, 2021-05-10:
Hello guys!
I am having this kind of issue too. I am running freepbx with chan-sccp and I would like to apply this fix.

The phone not full restarted, just the sccp session restarted ( according to the attached photo). 
[image]

Please, can you guide me to apply this fix?
Rodolfo.

dkgroot, 2021-05-13:
#
***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/496 origin/issues/496
git checkout issues/496
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

duartero84, 2021-05-15:
Thanks for your supporting!

I have applyed the fix and than I will have monitoring so far for about the next three days. 
Soon I will let you know to close this issue.

dkgroot, 2021-05-24:
@duartero84 The fix mentioned above has been applied to the develop branch and it no longer available separately. I am hopeing everything is ok with it. You should be able to switch back to the develop branch.

duartero84, 2021-05-26:
Thanks @dkgroot for helping! All phones are working fine! I have a question for new installation of freepbx and chan-sccp. Should I apply the fix or will the released for the next version?

dkgroot, 2021-07-11:
@duartero84 This branch was already merged

commit 239d0e99 (2019-12-17): Fix crossover device session: with state OK, causing endless loop

Cleaning previous session, come back later, state:OK

Fixes: #496
References: #488
References: #333