chan-sccp/chan-sccp #497: Asterisk stops working after Cisco 7925 (WiFi) tries to reregister.

anutator, 2019-12-10:
After you made changes in chan_sccp the problem we also had great problems with Asterisk and shared lines, but 3-4 times in a month. I decided to make separate phones without shared lines and we still have problems - all phones don't work, IVR not working — nothing.

alarm:
```
[Dec 10 17:53:08] VERBOSE[2758] pbx_variables.c: Setting global variable 'SIPDOMAIN' to '10.112.0.103'
[Dec 10 17:53:24] VERBOSE[1600] :     -- SCCP: Accepted Client Connection from 10.112.14.21:4367
[Dec 10 17:53:24] VERBOSE[1600] :     -- SCCP: Connected on server via 10.112.0.103:51944
[Dec 10 17:53:25] VERBOSE[2797] :     -- SCCP: Alarm Message: Severity: Informational (2), Name=SEP30F70D4D19FD Load=CP7925G-1.4.8SR1.5 Last=TCP-timeout [67584/353267722]
[Dec 10 17:53:25] VERBOSE[2797] :  SEP30F70D4D19FD: is registering, Instance: 1, UserId: 0, Type: Cisco 7925 (484), Version: 16 (loadinfo '')
[Dec 10 17:53:25] WARNING[2797] sccp_session.c: Session(0x7f89380a3b20) and Device Session(0x7f893807d2e0) are out of sync.
[Dec 10 17:53:25] WARNING[2797] sccp_actions.c: SEP30F70D4D19FD: Cleaning previous session, come back later, state:OK
[Dec 10 17:53:25] VERBOSE[2797] :   == 10.112.0.103:51944: Session 0x7f893807d510 needs to be closed!
[Dec 10 17:53:57] VERBOSE[1600] :     -- SCCP: Accepted Client Connection from 10.112.14.21:4369
[Dec 10 17:53:57] VERBOSE[1600] :     -- SCCP: Connected on server via 10.112.0.103:38554
[Dec 10 17:53:57] VERBOSE[2803] :     -- SCCP: Alarm Message: Severity: Informational (2), Name=SEP30F70D4D19FD Load=CP7925G-1.4.8SR1.5 Last=TCP-timeout [67584/353267722]
[Dec 10 17:53:57] VERBOSE[2803] :  SEP30F70D4D19FD: is registering, Instance: 1, UserId: 0, Type: Cisco 7925 (484), Version: 16 (loadinfo '')
[Dec 10 17:53:57] WARNING[2803] sccp_session.c: Session(0x7f8938051640) and Device Session(0x7f893807d2e0) are out of sync.
[Dec 10 17:53:57] WARNING[2803] sccp_actions.c: SEP30F70D4D19FD: Cleaning previous session, come back later, state:OK
[Dec 10 17:53:57] VERBOSE[2803] :   == 10.112.0.103:38554: Session 0x7f893807d510 needs to be closed!
[Dec 10 17:54:15] VERBOSE[1600] :     -- SCCP: Accepted Client Connection from 10.112.14.21:4371
[Dec 10 17:54:15] VERBOSE[1600] :     -- SCCP: Connected on server via 10.112.0.103:53167
[Dec 10 17:54:15] VERBOSE[2807] :     -- SCCP: Alarm Message: Severity: Informational (2), Name=SEP30F70D4D19FD Load=CP7925G-1.4.8SR1.5 Last=TCP-timeout [67584/353267722]
[Dec 10 17:54:15] VERBOSE[2807] :  SEP30F70D4D19FD: is registering, Instance: 1, UserId: 0, Type: Cisco 7925 (484), Version: 16 (loadinfo '')
```
Asterisk 16.6.0-rc2. 
Skinny Client Control Protocol (SCCP). Release: 4.3.2 issues/438 - 666a88e (built by 'root' on '2019-10-22 17:56:18 UTC')

I noticed 7911 and  7925 have different SCCP version inside firmware. SEP30F70D4D19FD — Cisco 7925.

```
--- SCCP channel driver device settings ----------------------------------------------------------------------------------
MAC-Address                                    : SEP30F70D4D19FD
Protocol Version                               : Supported '16', In Use '16'
Protocol In Use                                : SCCP Version 16
Device Features                                : 0x6085,000000000110000010000101
Tokenstate                                     : None

--- SCCP channel driver device settings ----------------------------------------------------------------------------------
MAC-Address                                    : SEP04C5A4B0856E
Protocol Version                               : Supported '22', In Use '22'
Protocol In Use                                : SCCP Version 22
Device Features                                : 0x7285,000000000111001010000101
Tokenstate                                     : None
Keepalive                                      : 60

```

dkgroot, 2019-12-11:
@anutator
In issue #496 i have committed an fix for the issue you report above.

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/496
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

This should at least take care of the endless looping.

dkgroot, 2019-12-17:
@anutator The issue/496 branch was merged, which apparently solves some of the re-register/looping issue. Closing this issue report for now. Please re-open if/when necessary.

closed 2019-12-17 by dkgroot