chan-sccp/chan-sccp #498: chan_sccp fails to start when tcp is enabled in chan_sip

j217, 2019-12-19:
If "tcpenabled=yes" is in sip.conf then chan_sccp fails to start and loops with errors like 
[Dec 19 18:44:38] ERROR[100371] sccp_session.c: Error accepting new socket Software caused connection abort on accept_sock:11
[Dec 19 18:44:38] ERROR[100371] sccp_session.c: Error accepting new socket Bad file descriptor on accept_sock:11

**Reproduction Steps:**

1. [First Step]
2. [Second Step]
3. [Other Steps...]

**chan-sccp version:** chan_sccp-4.3.2_1 (chan-sccp-chan-sccp-v4.3.2-gamma_GH0) as in FreeBSD ports at 2019-Sep-01

**asterisk version:** 16.5.0 as in FreeBSD ports at 2019-Sep-01 + usecallmanager patch from usecallmanager.nz

**Expected behavior:**
Normal start independently of chan_sip settings

**Observed behavior:**
Setting of "tcpenable=yes" in sip.conf prevents chan_sccp to start.
After chan_sccp has started I can set tcpenable to yes and do "sip reload" without affecting chan_sccp.

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes]
- Platform:
FreeBSD 12.1-RELEASE r354233 GENERIC  amd64
running in KVM on
CPU: Common KVM processor (1995.29-MHz K8-class CPU)

- Direct Contact information: j217@dnepro.net
- Log Files in attach
- Comments: 
The version of chan_sccp was taken to fit asterisk 16.5.0 in ports because that's the version to apply usecallmanager patch to.

j217, 2019-12-19:
sorry, attached files were lost

j217, 2019-12-19:
OK, I can't attach files here, so let it be on pastebin:
https://pastebin.com/raw/VqE2vNAB

j217, 2019-12-19:
Confirmed also on fresh packages in FreeBSD:
aster-sccp*CLI> core show version 
Asterisk 16.6.2 built by root @ 120amd64-quarterly-job-01 on a amd64 running FreeBSD on 2019-12-10 09:56:14 UTC
with asterisk-chan_sccp-4.3.2_1 :
aster-sccp*CLI> sccp show version 
Skinny Client Control Protocol (SCCP). Release: 4.3.1 TRUNK - unknown (built by 'wheel' on '2019-12-10 22:47:09 UTC')
and even with chan-sccp built from ports (chan-sccp-chan-sccp-v4.3.2-epsilon_GH0.tar.gz)
aster-sccp*CLI> sccp show version
Skinny Client Control Protocol (SCCP). Release: 4.3.2 TRUNK - unknown (built by 'wheel' on '2019-12-19 20:11:00 UTC')

j217, 2019-12-19:
Not reproduced on asterisk13. 
On asterisk13 chan-sccp starts without errors no matter what is the value of tcpenable in sip.conf.
Also tried noloading of different modules but it still starts fine.
aster-sccp*CLI> core show version
Asterisk 13.29.2 built by root @ 120amd64-quarterly-job-11 on a amd64 running FreeBSD on 2019-12-10 04:42:26 UTC
aster-sccp*CLI> sccp show version 
Skinny Client Control Protocol (SCCP). Release: 4.3.2 TRUNK - unknown (built by 'wheel' on '2019-12-19 20:45:51 UTC')

dkgroot, 2019-12-20:
I don't understand how that could be caused
Except if chan_skinny is also loaded or chan_sip is using port 2000.

j217, 2019-12-21:
Chan_skinny is noload. That was the first thing I checked. Settings in sip.conf are as you show except port is now set in tcpbindaddr after as 1.2.3.4:5060.
The additional important detail is that chan-sccp fails only if the SIP phone on TCP is reachable when asterisk starts.
1) Start Asterisk when SIP phone on TCP is reachable, tcpenable=yes - chan-sccp fails to start.
2) Start Asterisk when SIP phone on TCP is reachable, tcpenable=no - chan-sccp starts.
3) Start Asterisk when SIP phone on TCP is unreachable, tcpenable=yes - chan-sccp starts.
4) If SIP phone on TCP becomes reachable after chan-sccp has started then it's OK.
5) If I set tcpenable=yes and do "sip reload" after chan-sccp has started then it's OK.

I suppose it has something to do with module initialization order but there are no settings to change it.

dkgroot, 2019-12-21:
Are you running a SIP softphone on the asterisk server ? What port is it using ?
You can turn of module autoloading and load all the modules using "load=", that should load them in order (i think), but would be quite a bit of work to figure out the required order. Might be better to figure out what is blocking port 2000. Maybe have a look at `lsof`

j217, 2019-12-21:
Here are open sockets when SIP phone was turned on after sccp-chan started successfully:
USER     COMMAND    PID   FD PROTO  LOCAL ADDRESS         FOREIGN ADDRESS      
asterisk asterisk   10228 6  stream /var/run/asterisk/asterisk.ctl
asterisk asterisk   10228 9  udp4   *:20721               *:*
asterisk asterisk   10228 10 tcp4   192.168.220.71:5060   192.168.220.171:51294
asterisk asterisk   10228 13 udp6   *:50547               *:*
asterisk asterisk   10228 14 udp4   *:5060                *:*
asterisk asterisk   10228 15 tcp4   *:5060                *:*
asterisk asterisk   10228 16 tcp4   *:2000                *:*
asterisk asterisk   10228 17 tcp4   192.168.220.71:2000   192.168.220.176:4339
asterisk asterisk   10228 23 tcp4   192.168.220.71:2000   192.168.220.175:2280

and here is what happens after I restarted asterisk and chan-sccp has started to loop with error
[Dec 21 18:44:45] ERROR[100637] sccp_session.c: Error accepting new socket Invalid argument on accept_sock:13

sockstat -v | grep aster
asterisk asterisk   10296 6  stream /var/run/asterisk/asterisk.ctl
asterisk asterisk   10296 9  udp4   *:50913               *:*
asterisk asterisk   10296 10 udp6   *:12054               *:*
asterisk asterisk   10296 11 udp4   *:5060                *:*
asterisk asterisk   10296 12 tcp4   *:5060                *:*
asterisk asterisk   10296 14 tcp4   192.168.220.71:2000   192.168.220.176:2052

and a minute later when SIP phone is reconnected:
sockstat -v | grep aster
asterisk asterisk   10296 6  stream /var/run/asterisk/asterisk.ctl
asterisk asterisk   10296 9  udp4   *:50913               *:*
asterisk asterisk   10296 10 udp6   *:12054               *:*
asterisk asterisk   10296 11 udp4   *:5060                *:*
asterisk asterisk   10296 12 tcp4   *:5060                *:*
asterisk asterisk   10296 13 tcp4   192.168.220.71:5060   192.168.220.171:52639
asterisk asterisk   10296 14 tcp4   192.168.220.71:2000   192.168.220.176:2052

I see that SIP phone connects to chan_sip on handle 13, and this is the same handle that chan_sccp errs on.

dkgroot, 2019-12-21:
But also handle 14 is still open, when the module is not loaded, that is very curious. Looks like the client port is not being shutdown  / terminated correctly.
This could be caused by some of the subtle differences between ip model implementation i guess.

If i can get a connection to your system, we should be able to analyze both issue and figure out a remedy. If need be i can boot up my own FreeBSD system, but i would require a bit of time to set it all up and if that works it would not help you (except for showing that it can work), if it doesn't work i would be able to work on it locally though.

j217, 2019-12-22:
bindaddr=192.168.x.x solved it for now
and also fixed the one-way audio
Fixes for 0.0.0.0 binding coming

dkgroot, 2019-12-26:
Hi John,
The issues/bsd branch has been merged back into the develop tree (and has been deleted). So please checkout develop and ths issue should be resolved.
Thanks you for reporting the issue and creating a test machine for me to fix the issue.

commit f03c9505 (2019-12-23): Partial revert to remedy regression (Linux testing: @darkponay)
Fix compile warning
Fix socklen_t issue related to BSD
Fix socklen_t by using sockaddr_un from sys/un.h

Fixes: #498
Fixes: #499
commit 5eebacc6 (2019-12-26): Fix BSD socklen_t connect() issue
Fix compile warning
Fix socklen_t issue related to BSD
Fix socklen_t by using sockaddr_un from sys/un.h

Fixes: #498
Fixes: #499
closed 2019-12-26 by dkgroot