chan-sccp/chan-sccp #499: cisco 6901 one-way audio

j217, 2019-12-20:
Cisco 6901 only receives audio, does not send.

**chan-sccp version:**
Skinny Client Control Protocol (SCCP). Release: 4.3.2 TRUNK - unknown (built by 'wheel' on '2019-12-19 20:45:51 UTC')
built from chan-sccp-chan-sccp-v4.3.2-epsilon_GH0.tar.gz

**asterisk version:** 
13.29.2, 16.5.0, 16.6.2

**Expected behavior:**
Two-way audio

**Observed behavior:**
The phone plays audio received from asterisk but does not send. The phone has two-way audio on SIP FW but ignores dialplan.xml so I'd like to use it on SCCP. It is in one LAN with Asterisk, no NAT or firewalls involved. Tcpdump shows no RTP packet stream from phone to Asterisk.
Tested incoming and outgoing calls, SCCP to SCCP, SCCP to SIP and SIP to SCCP, always no outgoing audio on SCCP phone, giving complete silence in case of SCCP to SCCP.

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Don't know, this is my first experience with chan-sccp.
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
FreeBSD 12.1-RELEASE r354233 GENERIC  amd64
in KVM
- Direct Contact information: j217@dnepro.net
- Log Files 
https://pastebin.com/raw/eBuq01pa

dkgroot, 2019-12-20:
Can you set sccp.conf:
```
[general]
debug = core, channel, pbx, rtp, codec
```
And do
```
cd /usr/src
git clone https://github.com/chan-sccp/chan-sccp chan-sccp_trunk
cd chan-sccp_trunk
./configure
make -j4
make install
make reload
```

dkgroot, 2019-12-20:
Looking at your log i notice:
```
[Dec 20 03:08:41] VERBOSE[100516][C-00000007] :     -- SEPC4B36A6DF772: Getting the active channel on device.
[Dec 20 03:08:41] VERBOSE[100516][C-00000007] :     -- SCCP: getnameinfo(): Address family not recognized 
[Dec 20 03:08:41] VERBOSE[100516][C-00000007] :     -- SEPC4B36A6DF772: Set the active channel 7 on device
[Dec 20 03:08:41] VERBOSE[100516][C-00000007] :     -- SEPC4B36A6DF772: Set the active line 21
[Dec 20 03:08:41] VERBOSE[100516][C-00000007] :     -- SCCP: getnameinfo(): Address family not recognized 
```
Could it be that your phone is running in ipv6 mode and the server is in ipv4 ?
You can set sccp.conf
```
[general]
bindaddr = ::
```
To switch the server to ipv6 mode, just to make sure/test.

j217, 2019-12-21:
1) Setting
debug = core, channel, pbx, rtp, codec
directrtp = off
earlyrtp = progress
does not help. Still no sound. I've also tried other options of earlyrtp with no success.

2) My server has no IPv6 address on LAN, only on Loopback. 6901s are provided with IPv4 DHCP, TFTP and callmanager address in config is of course v4. Maybe this warning is because there is no DNS available?

3) I've built trunk version - still no sound on 
aster-sccp*CLI> sccp show version
Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 88cc7bb (built by 'wheel' on '2019-12-21 11:00:46 UTC')

dkgroot, 2019-12-21:
When i looked at the debug log, after the error i mentioned there was no attempt to start the auto sending part of the rtp stream, so it makes sense that the audio would be one way. We need to figure out where the error is coming from though. And if this issue is at all related to the #498 problem. I am sorry but i am a little stumped by both issues at the moment.

Do you have other skinny phones besides the 6901, and are they working normally ?

dkgroot, 2019-12-21:
Setting `bindaddr=::` should not have negative side effect, it just switches the server to Dual IPv4 and IPV6 mode, supporting both. It might get us past the 'family' issue.

dkgroot, 2019-12-21:
Ooh maybe we are getting an invalid ip-address/port combination back from "OpenReceiveChannel". We should set 'sccp debug core, channel, rtp, message, codec' to check that out. If necessary we can add some extra debug statement in the function parsing the 'OpenReceiveChannel' message.

j217, 2019-12-21:
If I set bindaddr to :: then 6901s do not connect to asterisk. sockstat -v shows the following:
USER     COMMAND    PID   FD PROTO  LOCAL ADDRESS         FOREIGN ADDRESS      
asterisk asterisk   10169 6  stream /var/run/asterisk/asterisk.ctl
asterisk asterisk   10169 9  udp4   *:30843               *:*
asterisk asterisk   10169 10 udp6   *:25010               *:*
asterisk asterisk   10169 11 udp4   *:5060                *:*
asterisk asterisk   10169 12 tcp4   *:5060                *:*
asterisk asterisk   10169 13 tcp6   *:2000                *:*

I think in this case we are only listening on IPv6 not both versions. May be that's because of difference between Linux and FreeBSD?
There are no problems in VM network setup (bridged to vlan), DHCP, TFTP, SIP (with two-way audio) and SSH all work on it.
Log of call between 2 6901 with debugs:
https://pastebin.com/raw/nDCqUsda
I have no skinny phones except 6901s at the moment but I'll try to reflash some SIP cisco phone to SCCP tomorrow.

dkgroot, 2019-12-21:
Oh i did not realize you were running FreeBSD. I thought it was just a typo (and you meant FreePBX), i am so sorry for that faulty assumption.
The sockstat is that from before of after chan_scccp module is loaded. If before, then we know the problem (2000 is already being used by something).
Ok there might be a difference in running DualStack in BSD. I have done most of by BSD testing on DragonFlyBSD. It is still a very curious thing, chan-sccp from being prevented to load when chan_sip is loaded. Have you tried changing the chan-sccp port to 12000 or something, just to see if that makes a difference ?
Would be nice if the issue is 6901 specific, or if all have the same problem.

dkgroot, 2019-12-21:
```
[Dec 21 18:02:58] NOTICE[100579][C-00000001] sccp_actions.c: SEP381C1ABA7F04: SCCP Handle Message: Open Receive Channel Acknowledge(0x0022) 48 bytes length
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :  SCCP: packet hex dump:
 00000000 - 30 00 00 00 14 00 00 00  22 00 00 00 00 00 00 00  - 0.......".......
 00000010 - 00 00 00 00 C0 A8 DC B0  00 00 00 00 00 00 00 00  - ................
 00000020 - 00 00 00 00 C8 43 00 00  FC FF FF FF 03 00 00 00  - .....C..........
 00000030 - 00 00 00 00 00 00 00 00                           - ........
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :     -- SEP381C1ABA7F04: Got OpenChannel ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.220.176:17352', Type:Indirect RTP, PassThruPartyId>
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :     -- SEP381C1ABA7F04: Getting the active channel on device.
```
Ok the openReceiveChannelAck come through nicely, and it has an appropiate ipv4 address and port. So that part is good.
And this time we are starting MediaTransmission correctly:
```
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :     -- SCCP/22-00000003: Using dtmf rtp_code : 101
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :     -- SEP381C1ABA7F04: >> Send message Start Media Transmission
[Dec 21 18:02:58] NOTICE[100579][C-00000001] sccp_session.c: SEP381C1ABA7F04: Send Message: Start Media Transmission(0x008A) 136 bytes length
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :  SCCP: packet hex dump:
 00000000 - 88 00 00 00 14 00 00 00  8A 00 00 00 03 00 00 00  - ................
 00000010 - FC FF FF FF 00 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000020 - 00 00 00 00 00 00 00 00  00 00 00 00 14 00 00 00  - ................
 00000030 - 02 00 00 00 B8 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000040 - 00 00 00 00 03 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000050 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000060 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000070 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000080 - 65 00 00 00 0A 00 00 00  00 00 00 00 00 00 00 00  - e...............
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :     -- SEP381C1ABA7F04: Tell Phone to send RTP/UDP media from 192.168.220.176:17352 to  (NAT: (Auto)Off)
```
And the phones seems to be ok with that too:
```
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :     -- SEP381C1ABA7F04: >> Got message Start Media Transmission Acknowledge (0x154)
[Dec 21 18:02:58] NOTICE[100579][C-00000001] sccp_actions.c: SEP381C1ABA7F04: SCCP Handle Message: Start Media Transmission Acknowledge(0x0154) 52 bytes length
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :  SCCP: packet hex dump:
 00000000 - 34 00 00 00 14 00 00 00  54 01 00 00 00 00 00 00  - 4.......T.......
 00000010 - 00 00 00 00 03 00 00 00  00 00 00 00 C0 A8 DC B0  - ................
 00000020 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
 00000030 - 00 00 00 00 00 00 00 00  00 00 00 00              - ............
[Dec 21 18:02:58] VERBOSE[100579][C-00000001] :     -- SEP381C1ABA7F04: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.220.176:0', Type:Indirect RTP, PassThru>
```

dkgroot, 2019-12-21:
Having another look at the beginning of the call, it seems to already go wrong here, when asking asterisk to setup an rtp port instance for us.
```
[Dec 21 18:02:48] VERBOSE[100579] :     -- SCCP/22-00000001: rtp server instance created at 0.0.0.0:10244
[Dec 21 18:02:48] VERBOSE[100579] :   == SCCP/22-00000001: update rtpmap: format:CISCO-DTMF, payload:101, mime:audio, rate:0
[Dec 21 18:02:48] VERBOSE[100579] :     -- SCCP/22-00000001: (sccp_rtp_createRtpServer) updated phone Audio RTP destination to : , family:IPv6, mapped: False
```

dkgroot, 2019-12-21:
Note: I had not realized we had been talking about FreeBSD all this time. I just assumed it was a typo for FreePBX, sorry :-)

dkgroot, 2019-12-21:
1. Improvements: We need to check these return values more precisely and fail early. So if the rtp instance creation fails we should have givem up immediatly at that point.
2. i think the difference between BSD and Linux might be causing this. We are not using all the asterisk netsock library and they might have already dealt with this issue in the past. I will have a look if i can see any BSD specific code in there, that might explain the problem.
3. i will fire up my own FreeBSD vm and see if it can replicate the issues you are seeing.
4. would still like to have ssh access if that is possible, in case my own machine does not show the same problem.

j217, 2019-12-21:
I've left you a message in gitter about ssh access.

j217, 2019-12-22:
bindaddr=192.1x.x.x fixed the one-way audio
Fixes for 0.0.0.0 binding coming

dkgroot, 2019-12-26:
Hi John,
The issues/bsd branch has been merged back into the develop tree (and has been deleted). So please checkout develop and ths issue should be resolved.
Thanks you for reporting the issue and creating a test machine for me to fix the issue.

commit f03c9505 (2019-12-23): Partial revert to remedy regression (Linux testing: @darkponay)
Fix compile warning
Fix socklen_t issue related to BSD
Fix socklen_t by using sockaddr_un from sys/un.h

Fixes: #498
Fixes: #499
commit 5eebacc6 (2019-12-26): Fix BSD socklen_t connect() issue
Fix compile warning
Fix socklen_t issue related to BSD
Fix socklen_t by using sockaddr_un from sys/un.h

Fixes: #498
Fixes: #499
closed 2019-12-26 by dkgroot