chan-sccp/chan-sccp #500: extension could not be found in context 'default'

ApertureDevelopment, 2019-12-22:
Hello,

First of all, this:  is related but didn't help me.

I am using chan-sccp and tried to add my phone to the config, while the phone connects and accepts incoming calls, outgoing calls are always rejected. ``extension could not be found in context 'default'`` is the error message I get in the asterisk debug log.

I am using FreePBX btw.

**Reproduction Steps:**

1. Install Chan_sccp and load it
2. Setup a phone
3. Make a call from your phone

**chan-sccp version:** current ( build today and can't find a version number )

**asterisk version:** 16.6.2

ApertureDevelopment, 2019-12-22:
Off topic:
Happy #500 issue!

dkgroot, 2019-12-23:
@ApertureDevelopment 

Which chan-sccp revision are you using & which FreePBX version, just to get the required information, and we are not trying to figure something out that was already fixed ? Note: I am not a freepbx user, but "default" is not a context that should be used, you should be using 'from-internal' as the standard context in the sccp.conf [general] section. See: https://github.com/chan-sccp/chan-sccp/wiki/FreePBX.

ApertureDevelopment, 2019-12-23:
The issue is that the context is set to from-internal.
I built the develop branch yesterday, but if you want the revision, I am at commit:
88cc7bbc9390814bd3a7b46bd201004410a68461
And I am using FreePBX 15

Here are the command returns:
https://gist.github.com/ApertureDevelopment/5dc098e87eccca8330349083580db798

dkgroot, 2019-12-23:
As you can see in the `sccp show line` excerpt at line 12, the line is using 'default' context and not 'from-internal' or 'from-internal-xfer'. If you correct that, then the phone should start working.

I am not sure if you are using the FreePBX `sccp_manager` plugin. If you are using config files, check them. If you are using sccp_manager, you have to have a look at the realtime db and the sccp_manager frontend, to see where it is reverting back to 'default'. If it is hard defined somewhere, then there is no way for us to fallback to the global definition, so this is not directly related to issue: #204.

BTW: we have a [gitter chat](https://gitter.im/chan-sccp/chan-sccp) and you can also reach the developer of the [sccp_manager](https://gitter.im/sccp_manager/community) plugin on there.

ApertureDevelopment, 2019-12-23:
omg I am sooo stupid. I didnt see that even thought it was right before my nose.

Thanks for the help and have a great day

closed 2019-12-23 by ApertureDevelopment

dkgroot, 2019-12-23:
Also funktioniert es jetzt ? Wenn ja, dan viel spass mit chan-sccp !

ApertureDevelopment, 2019-12-23:
Naja ich bin jetzt in das alte problem mit dem ton gestoßen ( ich benutze chan-sccp seit jahren, aber ein Update der telefonanlage hat eben diese zerstört, jetzt benutze ich freepbx, vorher habe ich Wazo benutzt ) aber das problem war wenn ich mich richtig erinnere nur das nicht genug ports offen waren für sccp

dkgroot, 2019-12-23:
Welches 'alte problem' mbz. ton ? Vielleicht besser wenn wir das uber der gitter chat machen.

(Stupid me, would be better to keep the issue list in english, so others can also follow :-))