chan-sccp/chan-sccp #501: Error loading module 'chan_sccp'

philis, 2020-01-06:
Asterisk starts allright but fails to load module "chan_sccp"

In the logs you find
`ERROR[20617]: loader.c:170 module_load_error: Error loading module 'chan_sccp': /usr/lib/asterisk/modules/chan_sccp.so: undefined symbol: ast_mwi_unsubscribe_and_join`

**Reproduction Steps:**

1. [First Step]
2. [Second Step]
3. [Other Steps...]

**chan-sccp version:** (include revision and/or branch if relevant)
develop branch

**asterisk version:** (include revision and/or branch if relevant)
Asterisk 16.2.1~dfsg-1+deb10u1 (Debian)

**Expected behavior:**

Module is loaded during start up
**Observed behavior:**

Module does not loadduring start up

**Addition / Optional Information:**

During compilation I get the following warnings.
The file `sccp_mwi.c` was just altered a couple of hours ago. Might be related.

```
  CC       libsccp_la-sccp_mwi.lo
In file included from pbx_impl/ast/ast.h:26,
                 from pbx_impl/pbx_impl.h:18,
                 from common.h:52,
                 from sccp_mwi.c:13:
sccp_mwi.c: In function ‘pbxMailboxSubscribe’:
pbx_impl/ast/define.h:215:38: warning: implicit declaration of function ‘ast_mwi_subscribe_pool’; did you mean ‘pbx_mwi_subscribe_pool’? [-Wimplicit-function-declaration]
 #define pbx_mwi_subscribe_pool       ast_mwi_subscribe_pool
                                      ^~~~~~~~~~~~~~~~~~~~~~
sccp_mwi.c:169:49: note: in expansion of macro ‘pbx_mwi_subscribe_pool’
  pbx_subscription = (pbx_event_subscription_t *)pbx_mwi_subscribe_pool(subscription->mailbox->uniqueid, pbx_mwi_event, subscription);
                                                 ^~~~~~~~~~~~~~~~~~~~~~
pbx_impl/ast/define.h:215:38: warning: nested extern declaration of ‘ast_mwi_subscribe_pool’ [-Wnested-externs]
 #define pbx_mwi_subscribe_pool       ast_mwi_subscribe_pool
                                      ^~~~~~~~~~~~~~~~~~~~~~
sccp_mwi.c:169:49: note: in expansion of macro ‘pbx_mwi_subscribe_pool’
  pbx_subscription = (pbx_event_subscription_t *)pbx_mwi_subscribe_pool(subscription->mailbox->uniqueid, pbx_mwi_event, subscription);
                                                 ^~~~~~~~~~~~~~~~~~~~~~
sccp_mwi.c:169:21: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  pbx_subscription = (pbx_event_subscription_t *)pbx_mwi_subscribe_pool(subscription->mailbox->uniqueid, pbx_mwi_event, subscription);
                     ^
In file included from pbx_impl/ast/ast.h:26,
                 from pbx_impl/pbx_impl.h:18,
                 from common.h:52,
                 from sccp_mwi.c:13:
sccp_mwi.c: In function ‘pbxMailboxUnsubscribe’:
pbx_impl/ast/define.h:218:38: warning: implicit declaration of function ‘ast_mwi_unsubscribe_and_join’; did you mean ‘pbx_mwi_unsubscribe_and_join’? [-Wimplicit-function-declaration]
 #define pbx_mwi_unsubscribe_and_join ast_mwi_unsubscribe_and_join
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
sccp_mwi.c:193:3: note: in expansion of macro ‘pbx_mwi_unsubscribe_and_join’
   pbx_mwi_unsubscribe_and_join((struct pbx_mwi_subscriber *)subscription->pbx_subscription);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
pbx_impl/ast/define.h:218:38: warning: nested extern declaration of ‘ast_mwi_unsubscribe_and_join’ [-Wnested-externs]
 #define pbx_mwi_unsubscribe_and_join ast_mwi_unsubscribe_and_join
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
sccp_mwi.c:193:3: note: in expansion of macro ‘pbx_mwi_unsubscribe_and_join’
   pbx_mwi_unsubscribe_and_join((struct pbx_mwi_subscriber *)subscription->pbx_subscription);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

dkgroot, 2020-01-06:
@philis 
Ooh this was a regression i introduced yesterday. I will add an extra check, to make sure i use the right version, depending on the asterisk-17 revision.
Will post a commit later today.
Sorry for the inconvenience. If need be you can use the revision before last.

dkgroot, 2020-01-06:
Ooh i missed you are running asterisk-16 and it got misdiagnosed as asterisk-17. correcting that now.

dkgroot, 2020-01-06:
Pushed an update, can you check if this fixes the ./configure recognition of asterisk-16

philis, 2020-01-06:
Hey Diederik,
thanks for the update. It installs nicely now.
Grüße Phili

dkgroot, 2020-01-06:
No problem, thanks for the issue report und feedback !
Closing the issue.

P.S. Spreche zwar auch Deutsch aber bin Holländer, nice try :-)

closed 2020-01-06 by dkgroot

Tiotimolin, 2024-03-26:
Hey.
While compilation "make -j2" error

In file included from pbx_impl/ast/ast.h:26,
                 from pbx_impl/pbx_impl.h:18,
                 from common.h:61,
                 from sccp_mwi.c:10:
sccp_mwi.c: In function ‘pbxMailboxSubscribe’:
pbx_impl/ast/define.h:237:38: error: implicit declaration of function ‘ast_mwi_subscribe_pool’; did you mean ‘pbx_mwi_subscribe_pool’? [-Werror=implicit-function-declaration]
  237 | #define pbx_mwi_subscribe_pool       ast_mwi_subscribe_pool
      |                                      ^~~~~~~~~~~~~~~~~~~~~~
sccp_mwi.c:166:49: note: in expansion of macro ‘pbx_mwi_subscribe_pool’
  166 |  pbx_subscription = (pbx_event_subscription_t *)pbx_mwi_subscribe_pool(subscription->mailbox->uniqueid, pbx_mwi_event, subscription);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[3]: *** [Makefile:971: libsccp_la-sccp_mwi.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/root/chan-sccp/src'
make[2]: *** [Makefile:1109: all-recursive] Error 1
make[2]: Leaving directory '/root/chan-sccp/src'
make[1]: *** [Makefile:668: all] Error 2
make[1]: Leaving directory '/root/chan-sccp/src'
make: *** [Makefile:615: all-recursive] Error 1

Asterisk Version: 18.20.0
FreePBX 16.0.40.7