chan-sccp/chan-sccp #502: UNDEFINED! displayed when cycling DND on 7690

Stevehans, 2020-01-23:
When I cycle the DND button on a test 7960 I'm seeing UNDEFINED! rather than >>>DND(Busy)<<<

**Reproduction Steps:**

1. Press DND

**chan-sccp version:** 4.3.2 develop - 63fe478

**asterisk version:** 13.29.2

**Expected behavior:**

DND status displayed.

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Unknown, upgrade from old hardware and OS etc
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - Intel
  - 64 Bit
  - FreePBX
- Comments:
Looks to be the same issues seen many years ago here:- https://sourceforge.net/p/chan-sccp-b/mailman/message/33065110/

The driver for this is that I'm using dnd = reject and dndFeature=on
So I only cycle between busy and off, not silent.

dkgroot, 2020-01-23:
@Stevehans Thanks for reporting the issue, I guess i overlooked that when implementing the new version of the DND Feature button. Will fix it tomorrow

Stevehans, 2020-01-23:
Long time no speak!

Many thanks for that.

dkgroot, 2020-01-23:
Happy to hear from you
I don't understand how that would have happened, but i will figure it out :-)

dkgroot, 2020-01-23:
BTW: if you need to have it work right now, you can switch to revision:da3fad using
```
git checkout da3fad
```

Stevehans, 2020-01-24:
Done:
Skinny Client Control Protocol (SCCP). Release: 4.3.2 (detached - da3fadf (built by 'root' on '2020-01-24 11:06:12 UTC')

1. I only have 7960's but it's the same on all (they are all at P0030801SR02)
2. Debug below:
3. I only configured the new server the other day so the first git clone was the latest release, the only other version I have running is from years ago.

<<<Press of DND to cycle from "Your current options" to reject, which actually displays "UNDEFINED!">>>

SCCP new debug status: (8441 -> 8430073) core,device,line,action,channel,config,feature,socket,event,message
    -- SEP003094C3D44A: >> Got message SoftKey Event Message (0x26)
[2020-01-24 11:09:03] NOTICE[11704]: sccp_actions.c:123 check_session_message_device: SEP003094C3D44A: SCCP Handle Message: SoftKey Event Message(0x0026) 24 bytes length
 SCCP: packet hex dump:
 00000000 - 18 00 00 00 00 00 00 00  26 00 00 00 16 00 00 00  - ........&.......
 00000010 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
    -- SEP003094C3D44A: Got Softkey: DND (63) line=0 callid=0
    -- SEP003094C3D44A: Feature dnd Change Event Scheduled
    -- SEP003094C3D44A: >> Send message Clear Prompt Status Message
[2020-01-24 11:09:03] NOTICE[11704]: sccp_session.c:1048 sccp_session_send2: SEP003094C3D44A: Send Message: Clear Prompt Status Message(0x0113) 12 bytes length
 SCCP: packet hex dump:
 00000000 - 0C 00 00 00 0B 00 00 00  13 01 00 00 00 00 00 00  - ................
 00000010 - 00 00 00 00                                       - ....
    -- SEP003094C3D44A: Clear the status prompt on line 0 and callid 0
    -- SEP003094C3D44A: ( sccp_device.c:2141:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '▒' for line 0 (0)
    -- SEP003094C3D44A: >> Send message Display Dynamic Prompt Status Message
[2020-01-24 11:09:03] NOTICE[11704]: sccp_session.c:1048 sccp_session_send2: SEP003094C3D44A: Send Message: Display Dynamic Prompt Status Message(0x0145) 20 bytes length
 SCCP: packet hex dump:
 00000000 - 14 00 00 00 0B 00 00 00  45 01 00 00 00 00 00 00  - ........E.......
 00000010 - 00 00 00 00 00 00 00 00  80 13 00 00              - ............
    -- SEP003094C3D44A: Display prompt on line 0, callid 0, timeout 0
    -- SEP003094C3D44A: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEP003094C3D44A: validKeyMask 4294967295
    -- SEP003094C3D44A: >> Send message Select SoftKeys Message
[2020-01-24 11:09:03] NOTICE[11704]: sccp_session.c:1048 sccp_session_send2: SEP003094C3D44A: Send Message: Select SoftKeys Message(0x0110) 20 bytes length
 SCCP: packet hex dump:
 00000000 - 14 00 00 00 0B 00 00 00  10 01 00 00 00 00 00 00  - ................
 00000010 - 00 00 00 00 00 00 00 00  FF FF FF FF              - ............
    -- Processing Event 0x7fc1f000ca70 of Type Feature Changed via 0 callback:0x7fc1cb515667
    -- Processing Event 0x7fc1f000ca70 of Type Feature Changed via 1 callback:0x7fc1cb51f6bf
    -- Processing Event 0x7fc1f000ca70 of Type Feature Changed via 2 callback:0x7fc1cb4ac068
    -- SEP003094C3D44A: Received Feature Change Event: dnd(5)
    -- SEP003094C3D44A: >> Send message Clear Prompt Status Message
[2020-01-24 11:09:03] NOTICE[11631]: sccp_session.c:1048 sccp_session_send2: SEP003094C3D44A: Send Message: Clear Prompt Status Message(0x0113) 12 bytes length
 SCCP: packet hex dump:
 00000000 - 0C 00 00 00 0B 00 00 00  13 01 00 00 00 00 00 00  - ................
 00000010 - 00 00 00 00                                       - ....
    -- SEP003094C3D44A: Clear the status prompt on line 0 and callid 0
    -- SEP003094C3D44A: ( sccp_device.c:2131:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '9' for line 0 (0)
    -- SEP003094C3D44A: >> Send message Display Dynamic Prompt Status Message
[2020-01-24 11:09:03] NOTICE[11631]: sccp_session.c:1048 sccp_session_send2: SEP003094C3D44A: Send Message: Display Dynamic Prompt Status Message(0x0145) 20 bytes length
 SCCP: packet hex dump:
 00000000 - 14 00 00 00 0B 00 00 00  45 01 00 00 00 00 00 00  - ........E.......
 00000010 - 00 00 00 00 00 00 00 00  1E 39 00 00              - .........9..
    -- SEP003094C3D44A: Display prompt on line 0, callid 0, timeout 0
    -- Processing Event 0x7fc1f000ca70 of Type Feature Changed via 3 callback:0x7fc1cb540cf6
    -- SEP003094C3D44A: StorageBackend got Feature Change Event: dnd(5)
    -- SEP003094C3D44A: change dnd to reject

<<<Press of DND to cycle from reject back to "Your current options">>>

    -- SEP003094C3D44A: >> Got message SoftKey Event Message (0x26)
[2020-01-24 11:09:06] NOTICE[11704]: sccp_actions.c:123 check_session_message_device: SEP003094C3D44A: SCCP Handle Message: SoftKey Event Message(0x0026) 24 bytes length
 SCCP: packet hex dump:
 00000000 - 18 00 00 00 00 00 00 00  26 00 00 00 16 00 00 00  - ........&.......
 00000010 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  - ................
    -- SEP003094C3D44A: Got Softkey: DND (63) line=0 callid=0
    -- SEP003094C3D44A: Feature dnd Change Event Scheduled
    -- SEP003094C3D44A: >> Send message Clear Prompt Status Message
[2020-01-24 11:09:06] NOTICE[11704]: sccp_session.c:1048 sccp_session_send2: SEP003094C3D44A: Send Message: Clear Prompt Status Message(0x0113) 12 bytes length
 SCCP: packet hex dump:
[... 13 lines cut ...]
    -- SEP003094C3D44A: Received Feature Change Event: dnd(5)
       > SEP003094C3D44A: clear message stack 4
    -- SEP003094C3D44A: >> Send message Clear Prompt Status Message
[... cut ...]
[2020-01-24 11:09:06] NOTICE[11632]: sccp_session.c:1048 sccp_session_send2: SEP003094C3D44A: Send Message: Select SoftKeys Message(0x0110) 20 bytes length
 SCCP: packet hex dump:
 00000000 - 14 00 00 00 0B 00 00 00  10 01 00 00 00 00 00 00  - ................
 00000010 - 00 00 00 00 00 00 00 00  FF FF FF FF              - ............
    -- Processing Event 0x7fc1f000ceb0 of Type Feature Changed via 3 callback:0x7fc1cb540cf6
    -- SEP003094C3D44A: StorageBackend got Feature Change Event: dnd(5)
    -- SEP003094C3D44A: change dnd to off

dkgroot, 2020-01-24:
Thanks for the debug output, analyzing...

dkgroot, 2020-01-24:
```
[2020-01-24 11:09:03] NOTICE[11704]: sccp_session.c:1048 sccp_session_send2: SEP003094C3D44A: Send Message: Display Dynamic Prompt Status Message(0x0145) 20 bytes length
 SCCP: packet hex dump:
 00000000 - 14 00 00 00 0B 00 00 00  45 01 00 00 00 00 00 00  - ........E.......
 00000010 - 00 00 00 00 00 00 00 00  80 13 00 00              - ............
```
`80 13 00 00` translates to: "YOUR_CURRENT_OPTIONS" label on the status bar
And
```
    -- SEP003094C3D44A: >> Send message Display Dynamic Prompt Status Message
[2020-01-24 11:09:03] NOTICE[11631]: sccp_session.c:1048 sccp_session_send2: SEP003094C3D44A: Send Message: Display Dynamic Prompt Status Message(0x0145) 20 bytes length
 SCCP: packet hex dump:
 00000000 - 14 00 00 00 0B 00 00 00  45 01 00 00 00 00 00 00  - ........E.......
 00000010 - 00 00 00 00 00 00 00 00  1E 39 00 00              - .........9..
```
`1E 39 00 00` Translates to: DO_NOT_DISTURB_IS_ACTIVE

So logically we are sending the right thing, it might just be that this is not mapped correctly in the 7960 languages file.

Do you have `dnd=reject` set at global level or at device level ? `global dnd=reject` was deprecated some time ago, in favor of dndFeature=on/off. You can set dnd=reject at device level and/or line level now.

Stevehans, 2020-01-31:
Checking this now.

Stevehans, 2020-01-31:
Can confirm the dnd settings are correct, dndFeature=on at the global level and dnd=reject at both the device and line level.

Stevehans, 2020-01-31:
Adding the debug in now in [general]

dkgroot, 2020-01-31:
Ok i don't know how to proceed from here. If you know your previous approximate checkout date, i would be able to guide you through the `git bisect` process. We can also just guess (let's say 01-01-2014 or something). It normally would take between 7-9 compiles before finding the causing commit, that will show the first time the issue popped up.

Stevehans, 2020-01-31:
I'll go back a few revisions and see what I can see.

Strange that nobody else with 7960's is seeing it?

I'll also grab a tcpdump and see if there's any other information there.

dkgroot, 2020-01-31:
To get the revision at 01-01-2014 you would do
```
git checkout 'develop@{2014-01-01 00:00:00}'
```

dkgroot, 2020-01-31:
The tcpdump will have the same binary data you showed me with "sccp debug core,message" before, so there is not much we can find out from there.
There could be an issue with the locale files, not containing a translation for the "1E 39" status bar message. That would also cause an undefined. So if you are using locale files for your language, please check the locale files to see if "do not Disturb is Active" is in there.

Stevehans, 2020-01-31:
I don't believe we have any locale files

dkgroot, 2020-01-31:
Maybe we you can make a couple of jumps (by year) to see when the issue was there, that would be our starting point. If you like i can explain the bisect process (pretty simple).

Stevehans, 2020-01-31:
Ok, I'll pick this up Monday and figure out which change caused the issue.

BTW do you happen to have a set of English locale files I could drop on my tftp server?

dkgroot, 2020-01-31:
Yes i do, you can find them at https://github.com/dkgroot/provision_sccp

Stevehans, 2020-02-03:
Locale was the issue, I was previously running without locale files, adding the locale files has resolved this issue.

I'm guessing that, as you said, the message doesn't exist in the default load so we now must have the locale files.

Thanks.
Steve

closed 2020-02-03 by Stevehans

dkgroot, 2020-02-03:
Hi Steve,

Thanks a lot for checking out the locale files and closing the issue.

Would you mind checking our Wiki and see if you can fit in a Q-A message about the issue somewhere (for example: https://github.com/chan-sccp/chan-sccp/wiki/Troubleshooting-Software or maybe a new page about Language / Locale etc).