chan-sccp/chan-sccp #503: chan_sccp cause system hangs randomly

hsynynrdg, 2020-02-13:
Hello there,

We have a production Asterisk installed with hybrid, some of IP phones and some of SCCP (Cisco 7925G) phones on system. Any time of the day asterisk crashes and restarting himself.
Here is some informations;

1. updated chan_sccp to latest branch
2. configured with  ./configure --with-asterisk=/opt/asterisk/ --with-astmoddir=/usr/lib64/asterisk/modules/
3. make -j2 && make install && make reload

**chan-sccp version:** : Loaded chan_sccp.so => (Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 8ba339a (built by 'root' on '2020-02-13 13:20:30 UTC')

**asterisk version:** : Asterisk 13.28.0 built by root 

---------
[Feb 13 14:59:09] WARNING[108117][C-00000c07] channel.c: Hard hangup called by thread LWP 108117 on SCCP/01000-00000258, while blocked by thread LWP 127584 in procedure ast_waitfor_nandfds!  Expect a failure
[Feb 13 14:59:09] ERROR[127584][C-00001190] lock.c: astobj2.c line 470 (internal_ao2_ref): Error destroying mutex &obj_mutex->mutex.lock: Device or resource busy
[Feb 13 14:59:09] ERROR[127584][C-00001190] astobj2.c: FRACK!, Failed assertion bad magic number 0x1fd5 for object 0x7f55e0000098 (0)
[Feb 13 14:59:09] ERROR[127584][C-00001190] : Got 20 backtrace records
# 0: /opt/asterisk/sbin/asterisk(__ast_assert_failed+0x84) [0x6065ff]
# 1: /opt/asterisk/sbin/asterisk() [0x45afa7]
# 2: /opt/asterisk/sbin/asterisk() [0x45b61f]
# 3: /opt/asterisk/sbin/asterisk(__ao2_ref+0x2e) [0x45ba82]
# 4: /opt/asterisk/sbin/asterisk() [0x5dc7f5]
# 5: /opt/asterisk/sbin/asterisk(ast_datastore_free+0x4a) [0x4f5367]
# 6: /opt/asterisk/sbin/asterisk() [0x5dd2ef]
# 7: /opt/asterisk/sbin/asterisk(ast_channel_publish_dial_forward+0x137) [0x5da2ec]
# 8: /opt/asterisk/sbin/asterisk(ast_channel_publish_dial+0x41) [0x5da3d2]
# 9: /usr/lib64/asterisk/modules/app_dial.so(+0x7233) [0x7f55c31af233]
#10: /usr/lib64/asterisk/modules/app_dial.so(+0xd908) [0x7f55c31b5908]
#11: /usr/lib64/asterisk/modules/app_dial.so(+0xff78) [0x7f55c31b7f78]
#12: /opt/asterisk/sbin/asterisk(pbx_exec+0x11c) [0x58798e]
#13: /opt/asterisk/sbin/asterisk() [0x574b3d]
#14: /opt/asterisk/sbin/asterisk(ast_spawn_extension+0x64) [0x578268]
#15: /opt/asterisk/sbin/asterisk() [0x578f43]
#16: /opt/asterisk/sbin/asterisk() [0x57a71c]
#17: /opt/asterisk/sbin/asterisk() [0x6035a1]
#18: /usr/lib64/libpthread.so.0(+0x7e65) [0x7f564131ae65]
#19: /usr/lib64/libc.so.6(clone+0x6d) [0x7f56406bc88d]
-------

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: No
- Problem can be reliably reproduced, doesn't happen randomly: No
- Platform:
  - VMware Platform
  - 64 Bit
  - Centos
  - 7
- Direct Contact information: huseyin.yanardag@ado.com.tr

dkgroot, 2020-02-13:
@hsynynrdg Hi Huseyin,

Is there any chance you could switch to Asterisk-13.31.0 ? The issue might have already been resolved.
Do you get a core dump and if so, would you might create a gdb backtrace from it (See: [Reporting-Bugs](https://github.com/chan-sccp/chan-sccp/wiki/Reporting-Bugs) Page on our wiki).

hsynynrdg, 2020-02-13:
Hello @dkgroot , first of all thank you for your fast response. I have compiled asterisk with DONT_OPTIMIZE and DEBUG_THREADS enabled and also compiled chan_sccp with —disable-optimize parameter. But i dont know where to get core dump never did before :( would you lead me to how can i do? Should i do when the error occurs?

Best regard,

dkgroot, 2020-02-13:
Hi Huseyin,
Ok very nice,
It depends a little on your OS/system/setup. Check asterisk.conf and set dumpcore=yes. Make sure your /etc/defaults/asterisk or /etc/sysconf/asterisk has DUMPCORE=yes. And restart asterisk.
After that you can check if `ps ax | grep asterisk` shows asterisk running with '-g'... the '-g' means that it should dump a core file.
Once we are sure it does that, we can figure out where it is putting the core dump file (likely places /tmp or /var/log/asterisk)

dkgroot, 2020-02-13:
@hsynynrdg We also have a [chat](https://gitter.im/chan-sccp/chan-sccp) that might make it easier to talk and figure out how to help you. BTW what distribution are you running ?

dkgroot, 2020-02-13:
Hi Huseyin,

I have just committed a potential fix to the hard-hangup issue you reported above into a separate branch. This is not really tested yet.

**So please only run this, when you guys have some downtime (like evening or weekend) or on a dedicated test server. If you are on a Virtual Machine, you can clone your current production server.**

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/503
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

_We should still continue with collecting the data requested earlier, but i did some work in advance to see if i could come up with a potential fix._

hsynynrdg, 2020-02-15:
Hello @dkgroot thank you again,
How can i simulate issue on test server? Because its only happening randomly. I really dont know what is causing to trigger that issue.
We did chat with you and updated Asterisk to 13.30 and will check on Monday for any issue. Afterwards can you lead me to how to trigger this issue on test side?

Best regards,

dkgroot, 2020-02-16:
@hsynynrdg 
Hi Huseyin,

To be honest, I don't know exactly what triggers the issue (yet), that's why we were hoping to get some more logging by setting `sccp debug core, channel, pbx, indicate` when the issue happens again. I am not sure if switching to asterisk-13.30 helps, but similar issues were reported before and then stopped more recently.
The backtrace we got from the coredump is not directly helpfull, because the issue was caused in a different thread, meaning the lead up to the problem is not shown in the backtrace. This is unfortunate. I am hoping that by having more debug data, we can find out the state causing the issue and then fix it from there.

I did create a new branch 'issues/503', but not based on any evidence or knowledge about a way to trigger the issue. I went over the the asterisk source code (and comments) and re-implemented the hangup procedures based on that. The asterisk code is a little unclear about which hangup method should be used in what situation (there is 5 of them). I know which one caused the crash, but i don't know why it was called in this situation.

Once you have run into the issue again using 13.30 + sccp debug statements, we:
- might be/should be able to determine what happened
- should be able to check if the newly proposed implementation in 'issues/503' would handle this better
- could try out 'issues/503' branch to see if it would stop the problem from happening

Sorry for the lengthy (evasive) answer. But at the moment we don't have anything yet. We must wait until monday and see if the issue still occurs.

Regards,
Diederik

hsynynrdg, 2020-02-18:
Hello Diederik,
I am at office now. Sorry for late response. On monday its happened again. Now its just only 
Asterisk 13.30 and sccp debug statements. Didn't tried new branch yet.
At now which files should i upload to you or maybe if its available to check with remote session?

Best regards,

hsynynrdg, 2020-02-18:
Now i have compiled with issues/503. Its late night here. Tried with some minor calls i think its working. Will try on morning with all circumtances. But anyway we got some debug logs when you lead me to enable sccp debug statements. When you available i can show you maybe its give you an idea.

Best regards,

dkgroot, 2020-02-18:
You can upload the last part of the debug logs until the crash here. So i can have a look, and then see if the new issues/503 version should have a positive effect.

hsynynrdg, 2020-02-18:
Here is the last call before crash/hangs asterisk.

[full18022020.txt](https://github.com/chan-sccp/chan-sccp/files/4220889/full18022020.txt)

dkgroot, 2020-02-22:
Reworked sccp_channel_answer
    
Now running in a separate thread and using sccp_channel_syncOpenReceiveChannel to wait for the rtp channel to be opened successfully before continuing. The new channel_answer thread will be joined during hangup, so that we cleanup afterwards (in case of huntgroup being answered at the same time).

Updated pickup/gpickup/auto_answer accordingly, to copy with new sccp_channel_answer implementation.

dkgroot, 2020-02-24:
Currently running on site: f11d965 
Current call count 13.250

dkgroot, 2020-02-26:
@hsynynrdg 
Hi Huseyin,

I have completely re-engineered my solution and come up with a much better one. I have committed it to a separate branch 'issues/503_1' to split from the previous branch. If you have a test server available, i would like you to start testing during the week. Including:
- [ ] simple call answer
- [ ] multiple phones answering the same call
- [ ] call gpickup with modeanswer
- [ ] call pickup with modeanswer
- [ ] call gpickup without modeanswer
- [ ] call pickup without modeanswer
- [ ] call transfer
- [ ] call barge
- [ ] conference

And anything else you can think of / that you guys use.

It contains a major overhaul of some of the call handling, so testing is required before putting in it production.

I am please that the f11d965 revision has giving us some more time to put the finishing touches on this one.

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/503_1
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

dkgroot, 2020-02-28:
I have now replicated the asterisk-16 specific changes to the other asterisk version 11->15

dkgroot, 2020-02-28:
@hsynynrdg 
Hi Huseyin,

I have completely re-engineered my solution and come up with a much better one. I have committed it to a separate branch 'issues/503_1' to split from the previous branch. If you have a test server available, i would like you to start testing during the week. Including:
- [x] simple call answer
- [x] multiple phones answering the same call
- [X] call gpickup with modeanswer
- [ ] call pickup with modeanswer
- [ ] call gpickup without modeanswer
- [ ] call pickup without modeanswer
- [x] call transfer
- [x] call barge
- [x] call barge on shared line
- [x] conference
- [x] video calls
- [x] video calls with gpickup

And anything else you can think of / that you guys use.

It contains a major overhaul of some of the call handling, so testing is required before putting in it production.

I am please that the f11d965 revision has giving us some more time to put the finishing touches on this one.

***Git***
_Checkout this temporary branch_
```
git pull
git fetch -a
git branch
git checkout issues/503_1
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

dkgroot, 2020-02-29:
@hsynynrdg
Before testing / debugging, please set sccp.conf
`debug = core, channel, pbx, rtp, socket, feature`

dkgroot, 2020-02-29:
Change report so far
```
 NEWS                         |   7 ++
 src/Makefile.am              |  29 ++++----
 src/Makefile.in              |  29 ++++----
 src/pbx_impl/ast/ast.c       |  72 +++++++++++++------
 src/pbx_impl/ast/define.h    |  24 +++++++
 src/pbx_impl/ast106/ast106.c |  30 ++++++--
 src/pbx_impl/ast108/ast108.c |  31 ++++++--
 src/pbx_impl/ast110/ast110.c |  31 ++++++--
 src/pbx_impl/ast111/ast111.c | 117 ++++++++++++++++++++-----------
 src/pbx_impl/ast112/ast112.c | 123 ++++++++++++++++++++------------
 src/pbx_impl/ast113/ast113.c | 134 ++++++++++++++++++++---------------
 src/pbx_impl/ast114/ast114.c | 135 ++++++++++++++++++++---------------
 src/pbx_impl/ast115/ast115.c | 137 ++++++++++++++++++++----------------
 src/pbx_impl/ast116/ast116.c | 137 ++++++++++++++++++++----------------
 src/sccp_actions.c           |  62 ++++++++--------
 src/sccp_channel.c           | 730 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++---------------------------------------------------------------------------------------------
 src/sccp_channel.h           |   7 +-
 src/sccp_cli.c               | 149 +++++++++++++++++++++++++++++++++------
 src/sccp_device.c            |   4 +-
 src/sccp_enum.in             |  20 +++++-
 src/sccp_feature.c           | 173 +++++++++++++++++++++------------------------
 src/sccp_indicate.c          |  45 ++++++------
 src/sccp_pbx.c               | 114 +++++++++++++++---------------
 src/sccp_protocol.c          | 395 +++++++++++++++++++++++++++++++++++++++++++++++++++++++-----------------------------------------------
 src/sccp_protocol.h          |  24 +++----
 src/sccp_rtp.c               |  36 +++++++++-
 src/sccp_rtp.h               |  10 ++-
 src/sccp_session.c           | 146 ++++++++++++++++++++++++++------------
 src/sccp_session.h           |   4 +-
 src/sccp_softkeys.c          |   4 +-
 30 files changed, 1736 insertions(+), 1223 deletions(-)
```

sdstorts, 2020-03-05:
Here are some debugs for the one way audio issue.   https://0x0.st/iTZv.txt

dkgroot, 2020-03-07:
@sdstorts @matthoskins1980_gitlab 
Created a patch for the even order in issues/503_1. Can you give this a go in your test environment
If/when simple call works, please go through the list provided above to run the other tests mentioned.
(please set: `sccp debug core, channel, pbx, rtp, codec, feature` to make sure we get enough detail incase of a problem). There is a slight chance of locking issues atm (deadlock), so please do not run this in production just yet :-))

sdstorts, 2020-03-09:
I am seeing an issue with multiple phones answering the same call.  The call is placed from an SCCP phone and I pick it up at the same time on the dialed extension(both SCCP phones).  One of the phones will grab the call, however, there is only one way audio(the phone making the call can hear but not the other way around).  The other phone that answered the call shows that it has the call as well(with no audio) and it also continues to ring.  When I hang that phone up, the call is disconnected on all phones.
https://0x0.st/iT7R.txt

dkgroot, 2020-03-09:
@sdstorts / @matthoskins1980_gitlab 

Ooh this is a sharedline you are calling, is that right ? That seems to be messing everything up. Can you confirm this is a sharedline ?
I was testing Dial(SCCP/123&SCCP/432); like in the original issue report. I was only aiming to fix that issue first.

Both devices sharing the line, is completely messing up the rtp traffic as shown in
```
    -- SEPD0574CF752AD: Got OpenChannel ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'10.160.223.20:30370', Type:Indirect RTP, PassThruPartyId:4294967288, CallID:7
    -- SEPD0574CF752AD: Getting the active channel on device.
    -- SCCP: (setPhoneRTPAddress) Update PBX to send RTP/UDP media to '10.160.223.20:30370' (new remote) (NAT: no)
    -- SEP649EF3A70FC6: Tell PBX   to send RTP/UDP media from 10.160.220.13:19928 to 10.160.223.20:30370 (NoNat)
```
Where one device is using the rtp port from another :-)
Sorry about that. I thought i had taken that into consideration

I guess that will be the bit of testing i will have to do next. Thanks for reporting this, i will make some time tomorrow and fix it (should not be to hard).

dkgroot, 2020-03-10:
@sdstorts 
Patch committed for the shared line issue, should be a lot cleaner now in behaviour. Let me know how the testing goes

commit 537b2c4d (2020-02-26): Implement split sccp_channel_answer via rtp.audio.reception.callback
Fix double answer causing:
 - segfault
 - stuck pbxchannels
Fix slow openreceive channel response by making sure events occur in right order (-1 second delay)

Fixes: #503
commit 8387020b (2020-03-21): Merge branch 'issues/503_1' into develop

Fixes: #503

Thanks: @sdstorts, @hboetes, @hsynynrdg, @matthoskins1980_gitlab
closed 2020-03-21 by dkgroot