chan-sccp/chan-sccp #504: https://asterisk/admin/config.php?display=sccp_phone does not show DN after bulk import

hboetes, 2020-02-25:
After a bulk import the https://asterisk/admin/config.php?display=sccp_phone list does not show the Display name (DN).

After opening the entry and saving it again —without having made any changes, the DN is show.

Am I missing something or is this a problem?

dkgroot, 2020-02-27:
@hboetes Hi Han,
I don't really know what you are talking about to be honest. Is this related to FreePBX | sccp_manager maybe ? In that case the issue should be reported to https://github.com/PhantomVl/sccp_manager which is where the sccp_manager plugin is maintained.
the chan-sccp project is only involved in the chan-sccp asterisk module, which is separate. I personally don't use FreePBX so i have 0 knowledge about it.
Sorry to be referring you without being able to help.

hboetes, 2020-02-27:
Ah, sorry for the confusion. I'll report it over there.

closed 2020-02-27 by hboetes

dkgroot, 2020-02-27:
No problem. And thanks for closing the issue :-)