chan-sccp/chan-sccp #505: Inconsistent DND output between DB and CLI output

hboetes, 2020-03-05:
In the sccp_manager I disabled DND for line 315,  only to find the CLI output say it was in reject mode:
```
# asterisk -rx "sccp show line 315"|grep dnd
DND Action                                     : Reject
```
The database states it is off though.
```
MariaDB [asterisk]> select dnd from sccpline where name = '315';
+------+
| dnd  |
+------+
| off  |
+------+
```
```
# asterisk -rx "sccp show line 315"|grep dnd
DND Action                                     : Reject
```

dkgroot, 2020-03-05:
I just ran through the `dnd` permutations in sccp.conf:line and that all worked out correctly.
on my system i tried:
```
[98031]
...
dnd = off
dnd = reject
dnd = silent
dnd = somewrongvalue
```

To see why this is not parsed correctly over there, would require:
sccp.conf
```
[general]
debug=core, config
```
and `sccp reload line 315` to see what we get from the database and how this is handled.

Some DND backgroup information, that might be helpfull:
Help line provided by the source in sccp_config_entries.hh:325 states
> "allow setting dnd action for this line. Valid values are 'off', 'reject' (busy signal), 'silent' (ringer = silent) or 'user' (not used at the moment). . The value 'on' has been made obsolete in favor of 'reject'\n"

However:
- device->dndFeature controls the availability of the dnd to the user (will they see a 'dnd' softkey yes/no)
- device->dnd and line->dnd controls the state of dnd off the phone / line:
  - 'off' means we use tri-state sequence through all three modes.  (to be honest 'cycle' would haven been a better name, in retrospect)
  - 'reject' means we use simple toggling between 'off' and 'reject'
  - 'busy' means we use simple toggling between 'off' and 'busy'

To see if dnd is actually active on the phone, you can check `astdb` which stores temporary states.
`database show SCCP/SEP381C1AB6B976`

dkgroot, 2020-03-06:
Can you show me what `realtime load sccpline name 315` returns, that is the actual value returned to chan-sccp when we ask for it. I need to know if mysql is not returning an enum value instead of the enum string for example.

hboetes, 2020-03-06:
Most certainly!
```asterix*CLI> realtime load sccpline name 315
                   Column Name  Column Value                  
          --------------------  --------------------          
                            id  641                           
                           pin                                
                         label  firstname lastname        
                   description  firstname lastname <315>    
                       context  from-internal                 
                 incominglimit  2                             
                      transfer  on                            
                       mailbox                                
                         vmnum  *97                           
                      cid_name  firstname lastname          
                       cid_num  315                           
                      disallow  all                           
                         allow                                
                       trnsfvm                                
     secondary_dialtone_digits  9                             
       secondary_dialtone_tone  0x22                          
                    musicclass  default                       
                      language                                
                   accountcode  315                           
                    echocancel  on                            
            silencesuppression  off                           
                     callgroup                                
                   pickupgroup                                
                namedcallgroup                                
              namedpickupgroup                                
                   adhocNumber                                
                        meetme                                
                     meetmenum                                
                    meetmeopts                                
                      regexten                                
               directed_pickup                                
       directed_pickup_context  from-internal                 
             pickup_modeanswer                                
                      amaflags                                
                           dnd  off                           
                     videomode  auto                          
                        setvar                                
                          name  315                           
```

dkgroot, 2020-03-07:
OK that looks fine, hmmm. Time to check the config parse then: Please set
sccp.conf
```
[general]
debug=core, config
```
and run `sccp reload line 315` to see how that get's parsed.

dkgroot, 2020-04-07:
@hboetes Was this resolved in the meantime ?

hboetes, 2020-04-07:
Bah, I locked myself out of github. :-(

hboetes, 2020-04-14:
Looks a lot better now. :-)
```
root@asterix ~ # asterisk -rx "sccp show line 315"|grep dnd
DND Action                                     : Disabled / Cycle
root@asterix ~ # echo 'select dnd from sccpline where name = '315';' |mysql asterisk
dnd
off
```

closed 2020-04-14 by hboetes