chan-sccp/chan-sccp #506: Group pickups fail if I call with a SIP phone to an SCCP phone.

hboetes, 2020-03-05:
Group pickups fail if I call with a SIP phone to an SCCP phone.

The 192.168.254 is the range at home. The 10.10.20 is the companies voice VLAN

```
[2020-03-05 16:44:01] NOTICE[9474][C-00000070]: sccp_feature.c:434 sccp_feat_grouppickup: SEP1C17D341C00D: (gpickup) get channel: <null>.
[2020-03-05 16:44:01] NOTICE[9474][C-00000070]: sccp_feature.c:446 sccp_feat_grouppickup: SCCP/107-00000280: executing_gpickup
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
    -- SCCP/107-00000280 answered PJSIP/146-00000069
       > 0x7fc0a412ead0 -- Strict RTP learning after remote address set to: 192.168.254.101:47880
    -- Channel SCCP/107-00000280 joined 'simple_bridge' basic-bridge <ce1df316-7fa1-41f0-952d-4bbd21ce58f8>
    -- Channel PJSIP/146-00000069 joined 'simple_bridge' basic-bridge <ce1df316-7fa1-41f0-952d-4bbd21ce58f8>
       > 0x7fc0600033a0 -- Strict RTP learning after remote address set to: 10.10.20.163:17120
    -- Channel PJSIP/146-00000069 left 'simple_bridge' basic-bridge <ce1df316-7fa1-41f0-952d-4bbd21ce58f8>
    -- Channel SCCP/107-00000280 left 'simple_bridge' basic-bridge <ce1df316-7fa1-41f0-952d-4bbd21ce58f8>
  == Spawn extension (macro-dial-one, s, 56) exited non-zero on 'PJSIP/146-00000069' in macro 'dial-one'
  == Spawn extension (macro-exten-vm, s, 26) exited non-zero on 'PJSIP/146-00000069' in macro 'exten-vm'
  == Spawn extension (ext-local, 117, 2) exited non-zero on 'PJSIP/146-00000069'
    -- Executing [h@ext-local:1] Macro("PJSIP/146-00000069", "hangupcall,") in new stack
    -- Executing [s@macro-hangupcall:1] GotoIf("PJSIP/146-00000069", "1?theend") in new stack
    -- Goto (macro-hangupcall,s,3)
    -- Executing [s@macro-hangupcall:3] ExecIf("PJSIP/146-00000069", "0?Set(CDR(recordingfile)=)") in new stack
    -- Executing [s@macro-hangupcall:4] NoOp("PJSIP/146-00000069", "SCCP/107-00000280 montior file= ") in new stack
    -- Executing [s@macro-hangupcall:5] GotoIf("PJSIP/146-00000069", "1?skipagi") in new stack
    -- Goto (macro-hangupcall,s,7)
    -- Executing [s@macro-hangupcall:7] Hangup("PJSIP/146-00000069", "") in new stack
  == Spawn extension (macro-hangupcall, s, 7) exited non-zero on 'PJSIP/146-00000069' in macro 'hangupcall'
  == Spawn extension (ext-local, h, 1) exited non-zero on 'PJSIP/146-00000069'
asterix*CLI> quit
```

hboetes, 2020-03-05:
I also tried with a sip phone connected to our WIFI LAN, so no NAT, same problem.

dkgroot, 2020-03-05:
@hboetes I will try to replicate the circumstances tomorrow, to see what is going on, I will test this on both 'develop' and 'issues/503_1' (as call answering has changed). Note: I cannot reproduce your dialplan, but logically that should not be involved.

dkgroot, 2020-03-05:
Added a bit of debug output, so we can see what is being tried
```
    -- 0003: Allocated asterisk channel SCCP/98041-00000003
[Mar  5 22:36:14] NOTICE[4732]: sccp_feature.c:458 sccp_feat_grouppickup: SEP0023043403F9: (gpickup) retrieving channel: SCCP/98041-00000003 (98041@internal) (pickupgroup:'0', namedpickupgroups:'sales').
[Mar  5 22:36:14] NOTICE[4732]: sccp_feature.c:463 sccp_feat_grouppickup: SEP0023043403F9: (gpickup) target channel found: SCCP/98031-00000002 (callgroup:'2', namedcallgroups:'sales').
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_CONNECTED_LINE: Indicate connected line has changed'
```
There might be a context or callgroup/pickupgroup mismatch between pjsip and sccp in your dialplan. Meaning that pjsip calls are running in a different context then the sccp calls. causing a pickup to fail. It should however have reported this back to the user and logged the incident in the messages log.

hboetes, 2020-03-06:
It's almost a shame that the bug has appeared to have gone since changed a few apparently unrelated settings...
We'll keep it open for now. And do a few more tests and check the console output.

dkgroot, 2020-03-07:
Thats a good idea, i am little shocked it just went away as well. Maybe the wipping out of astdb has something to do with it (don't know).

dkgroot, 2020-04-04:
@hboetes Closing the issue for now. Hope that is ok

closed 2020-04-04 by dkgroot

hboetes, 2020-04-04:
Yes, I couldn't really test it because we now all work from home. But my last info is it's working so let's go with that. Otherwise I'll open up a new one.

dkgroot, 2020-04-04:
You can reopen this issue if/when it becomes necessary
Hope you are recovering well btw !