chan-sccp/chan-sccp #507: Cisco 8945 Record/Monitor Softkey Issue

tensi0n519, 2020-03-08:
Continuation of an issue that started on https://github.com/PhantomVl/sccp_manager/issues

dkgroot, 2020-03-08:
@tensi0n519 

I did some more testing. If i change the lamp color to amber/orange for this button, the text also changes to 'amber'. So clearly this is some debug info accidentally send out the display, during the development of this firmware. which nobody has noticed or reported.

I have found a way to fix it.
Instead of using colorfull dynamic feature buttons (allowing the control over the blinking, color and icon of the button). I have changed monitoring back to to plain old on/off button for monitor on the 89xx series. I can control the lavel of the feature button now, so we are back in business.

Do note: the button works differently, between pressing it onhook and oncall, and it is now difficult to tell because the light only goes on/off.

Committing this change to `issues/507` branch for you to test.
***Git***
_Checkout this temporary branch_
```
cd /usr/src
git fetch origin
git branch issues/507 origin/issues/507
git checkout issues/507
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
git checkout develop
```

dkgroot, 2020-03-08:
(Sorry my commit message closed the issue, but you first need to test it of course) :-)

tensi0n519, 2020-03-08:
Great work! The 8945 line button label now makes sense. The 7975 works as well. No issues there. Only suggestion is on the 8945...The "Conversaion is being recorded" is cut off. Maybe you know this already? See the screen shots below.

[image]
[image]

tensi0n519, 2020-03-08:
@dkgroot 
Now that I've tested it, do you want me to move back to development?

dkgroot, 2020-03-08:
@tension, cannot do anything about the cut of, the screen on the 8945 is split in a left side and a right side, so longer mesages get truncated. I only sent the label 'Recording" but this is what that turns into :-(
On the 79xx series that looks fine (of course).
I guess with the popup you know what it says,.. We could instead send a string but then we dont have translation
So lets keep this. if there are complains or you have an idea let me know
Merging this for now.

dkgroot, 2020-03-08:
Pushed to develop, you can pull again to get the new version

dkgroot, 2020-03-08:
Actually now that i think about it, we are getting tripple feed back about the recording being done on the 8945, that is a bit much. We should drop the label extension there.

tensi0n519, 2020-03-08:
Ok should I wait to pull then?

dkgroot, 2020-03-08:
Ok committed patch

dkgroot, 2020-03-08:
@tensi0n519 
Ok I committed patch 41ab5b. Go ahead.
New version provides a cleaner quieter display
Thanks for testing and the feedback !

Happy recording !
If you do have some time and you could hookup the 89xx to the CME you have, that would be perfect. There are quite a couple of things i would like to figure out about these things (including what is and isnt' broken :-))

tensi0n519, 2020-03-08:
Applied the update and all is working well. Good work on your end. Tomorrow I should have some time and I'll fire up cme and see if the 8945 will register. Either way I will let you know. Would you like me to pm you or report back here?

dkgroot, 2020-03-08:
You can switch to the https://gitter.im/chan-sccp/chan-sccp and talk there, would be better. And thanks for helping with these. I have been searching for a while for someone with a CME/CUCM and a 8945, which seems to be rare.
Note: you might have to upload the latest (cop) firmware to the cme first. it contains load information for the cme, which it inserts into it's database, to gain knowledge about these phones.

tensi0n519, 2020-03-08:
Sounds good. Looking forward to working with you more! Talk to you soon.

dkgroot, 2020-03-13:
Ran a test on CME + 8945 and added the `LiveRcd` softkey to the `connected` softkeyset
![LiveRcd](
https://user-content.gitter-static.net/ddbb4c2a96dd181b5ec0110ee644c5b96a220941/68747470733a2f2f66696c65732e6769747465722e696d2f74656e7369306e3531392f666342352f7468756d622f494d475f343437332e6a7067)

So we need to fix chan-sccp and add this button

tension, 2020-03-13:
？？？

dkgroot, 2020-03-13:
Does the link not work ?

tension, 2020-03-13:
Did you contact the wrong person?

dkgroot, 2020-03-13:
I am really sorry !

commit cffff605 (2020-03-08): Fix for record/monitor feature button on 8945

Apparently the phones firmware contains a bug, displaying the color of the light as the label.
Setting the label in the dynamic text does not have any influence over the outcome on the display.
Switching the 89xx series to old feature_monitor button style and writing extra information after
the label, to show state.

Fixes: #507
commit 325bfa25 (2020-03-08): Fix for record/monitor feature button on 8945

Apparently the phones firmware contains a bug, displaying the color of the light as the label.
Setting the label in the dynamic text does not have any influence over the outcome on the display.
Switching the 89xx series to old feature_monitor button style and writing extra information after
the label, to show state.

Fixes: #507