chan-sccp/chan-sccp #508: lots of logfile warnings.

hboetes, 2020-03-10:
Regularly a blurb of this appears in the logfiles. Can you make it go away?

```
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] WARNING[29408][C-0000006b]: chan_sip.c:23013 func_header_read: This function can only be used on SIP channels.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
[2020-03-10 11:43:52] ERROR[29408][C-0000006b]: res_pjsip_header_funcs.c:454 func_read_header: This function requires a PJSIP channel.
```

dkgroot, 2020-03-10:
Does not look like a chan-sccp issue, from the logs provided

dkgroot, 2020-03-10:
Something is calling a sip related channel function, but the log is not clear what that is, the message or command coming before this is missing.

closed 2020-03-10 by dkgroot