chan-sccp/chan-sccp #509: Unable to make calls with Cisco 9737

Zw1d, 2020-04-03:
Cisco 7937 allows to make a call only once after registering (sometimes not even once), but does receive calls without issues. 

**Reproduction Steps:**

1. Make a call using Cisco 7937
2. If it worked, try again
3. Call fails

**chan-sccp version:** v4.3.2_DEV

**asterisk version:** 16.6.1

**Expected behavior:**

Regular calling and hanging up from Cisco 7937

**Observed behavior:**
Below you can see logs from two identical actions, successful and failed one.
Extension was dialed from cisco 7937, call was answered by SCCP device (cisco 7962), rest is as follows:

Successful call that I have managed to capture, this one worked fine:
```
 -- SCCP/4001-00000031 Internal Gosub(func-apply-sipheaders,s,1) start
    -- Executing [s@func-apply-sipheaders:1] NoOp("SCCP/4001-00000031", "Applying SIP Headers to channel SCCP/4001-00000031") in new stack
    -- Executing [s@func-apply-sipheaders:2] Set("SCCP/4001-00000031", "TECH=SCCP") in new stack
    -- Executing [s@func-apply-sipheaders:3] Set("SCCP/4001-00000031", "SIPHEADERKEYS=") in new stack
    -- Executing [s@func-apply-sipheaders:4] While("SCCP/4001-00000031", "0") in new stack
    -- Jumping to priority 12
    -- Executing [s@func-apply-sipheaders:13] Return("SCCP/4001-00000031", "") in new stack
  == Spawn extension (from-internal, 4001, 1) exited non-zero on 'SCCP/4001-00000031'
    -- SCCP/4001-00000031 Internal Gosub(func-apply-sipheaders,s,1) complete GOSUB_RETVAL=
    -- Called SCCP/4001
    -- SCCP/4001-00000031 is ringing
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
       > 0x73e14018 -- Strict RTP learning after remote address set to: 10.4.48.243:26730
    -- SCCP/4001-00000031 answered SCCP/4005-00000030
    -- Channel SCCP/4001-00000031 joined 'simple_bridge' basic-bridge <c3e59cd8-115c-4a30-b0b7-9eb19c2b4ac5>
    -- Channel SCCP/4005-00000030 joined 'simple_bridge' basic-bridge <c3e59cd8-115c-4a30-b0b7-9eb19c2b4ac5>
       > 0x73e14018 -- Strict RTP switching to RTP target address 10.4.48.243:26730 as source
       > 0x73dabda0 -- Strict RTP learning after remote address set to: 10.4.48.238:21602
       > 0x73dabda0 -- Strict RTP switching to RTP target address 10.4.48.238:21602 as source
       > 0x73e14018 -- Strict RTP learning complete - Locking on source address 10.4.48.243:26730
       > 0x73dabda0 -- Strict RTP learning complete - Locking on source address 10.4.48.238:21602
    -- Channel SCCP/4005-00000030 left 'simple_bridge' basic-bridge <c3e59cd8-115c-4a30-b0b7-9eb19c2b4ac5>
  == Spawn extension (macro-dial-one, s, 55) exited non-zero on 'SCCP/4005-00000030' in macro 'dial-one'
  == Spawn extension (macro-exten-vm, s, 14) exited non-zero on 'SCCP/4005-00000030' in macro 'exten-vm'
  == Spawn extension (from-internal, 4001, 2) exited non-zero on 'SCCP/4005-00000030'
```

And this is a failed attempt. Target device picked up the call, then just diconnected after second or two, but cisco 7937 didn't notice the pick up, nor the disconnect and kept on calling for couple seconds more then just hung up. Also there's significant lag when using pick up and hang up buttons. **Tested with multiple units.**

```
-- Executing [s@func-apply-sipheaders:1] NoOp("SCCP/4001-00000033", "Applying SIP Headers to channel SCCP/4001-00000033") in new stack
    -- Executing [s@func-apply-sipheaders:2] Set("SCCP/4001-00000033", "TECH=SCCP") in new stack
    -- Executing [s@func-apply-sipheaders:3] Set("SCCP/4001-00000033", "SIPHEADERKEYS=") in new stack
    -- Executing [s@func-apply-sipheaders:4] While("SCCP/4001-00000033", "0") in new stack
    -- Jumping to priority 12
    -- Executing [s@func-apply-sipheaders:13] Return("SCCP/4001-00000033", "") in new stack
  == Spawn extension (from-internal, 4001, 1) exited non-zero on 'SCCP/4001-00000033'
    -- SCCP/4001-00000033 Internal Gosub(func-apply-sipheaders,s,1) complete GOSUB_RETVAL=
    -- Called SCCP/4001
    -- SCCP/4001-00000033 is ringing
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
       > 0x73e77ba0 -- Strict RTP learning after remote address set to: 10.4.48.243:30106
    -- SCCP/4001-00000033 answered SCCP/4005-00000032
[2020-04-02 17:04:36] WARNING[1775][C-00000009]: sccp_session.c:165 sccp_session_waitForPendingRequests: SEP0004F2EB4449: waitForPendingRequests timed out!
  == Spawn extension (macro-dial-one, s, 55) exited non-zero on 'SCCP/4005-00000032' in macro 'dial-one'
  == Spawn extension (macro-exten-vm, s, 14) exited non-zero on 'SCCP/4005-00000032' in macro 'exten-vm'
  == Spawn extension (from-internal, 4001, 2) exited non-zero on 'SCCP/4005-00000032'
    -- Executing [h@from-internal:1] Macro("SCCP/4005-00000032", "hangupcall") in new stack
    -- Executing [s@macro-hangupcall:1] GotoIf("SCCP/4005-00000032", "1?theend") in new stack
    -- Goto (macro-hangupcall,s,3)
    -- Executing [s@macro-hangupcall:3] ExecIf("SCCP/4005-00000032", "0?Set(CDR(recordingfile)=)") in new stack
    -- Executing [s@macro-hangupcall:4] NoOp("SCCP/4005-00000032", "SCCP/4001-00000033 montior file= ") in new stack
    -- Executing [s@macro-hangupcall:5] GotoIf("SCCP/4005-00000032", "1?skipagi") in new stack
    -- Goto (macro-hangupcall,s,7)
    -- Executing [s@macro-hangupcall:7] Hangup("SCCP/4005-00000032", "") in new stack
  == Spawn extension (macro-hangupcall, s, 7) exited non-zero on 'SCCP/4005-00000032' in macro 'hangupcall'
  == Spawn extension (from-internal, h, 1) exited non-zero on 'SCCP/4005-00000032'
[2020-04-02 17:04:41] WARNING[1775][C-00000009]: sccp_session.c:165 sccp_session_waitForPendingRequests: SEP0004F2EB4449: waitForPendingRequests timed out!
```

dkgroot, 2020-04-03:
@Zw1d 
Thanks for reporting the issue.

Is this a recent regression (maybe since merging issue #503), if it is would mind going a `git checkout 7d2ea9` to see if this older version has the same issue. 

Please set `sccp debug core, channel, rtp, indicate, pbx, codec` So we can see what is going on with such a call (in either case).

Zw1d, 2020-04-03:
@dkgroot 
First of all, thank You for responding so quickly.
I have reinstalled the version you suggested and unfortunately the problem persists. But I might have narrowed it down to that particular hangup button (both physical and softkey), which seems to act up. First pick up and hang up works just fine, second pick up is good, but hangup lags and executes after console displays
`[2020-04-03 14:41:14] WARNING[4977]: sccp_session.c:165 sccp_session_waitForPendingRequests: SEP0004F2EB452A: waitForPendingRequests timed out!`

I set the debug you asked for, then just picked up, and hung up:

```
raspbx*CLI> sccp debug core, channel, rtp, indicate, pbx, codec
SCCP new debug status: (0 -> 530501) core,rtp,channel,indicate,pbx,codec
    -- SEP0004F2EB452A: Getting the active channel on device.
    -- SEP0004F2EB452A: No active channel on device.
    -- SEP0004F2EB452A: Taken Offhook
    -- SCCP: Looking for channel by state '22' on device: SEP0004F2EB452A
    -- SEP0004F2EB452A: line: '4009'
    -- SEP0004F2EB452A: Could not find active channel with state RINGING(22) on device
SEP0004F2EB452A: (getEmptyChannel) on line:4009, maybe_c:
    -- SEP0004F2EB452A: Getting the active channel on device.
    -- SEP0004F2EB452A: No active channel on device.
    -- SEP0004F2EB452A: Set the active channel 3 on device
    -- 49: New channel number: 3 on line 4009
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 4009
    -- SCCP:             cid_num: 4009
    -- SCCP:            cid_name: Bez nazwy
    -- SCCP:         accountcode: 4009
    -- SCCP:               exten:
    -- SCCP:             context: from-internal
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/4009-00000003
    -- SCCP: combined audio caps: [g722/64k, ulaw/64k, alaw/64k, g729a, g729, ilbc]
    -- SCCP: reduced audio prefs: [alaw/64k, ulaw/64k]
    -- SCCP: combined video caps: []
    -- SCCP: reduced video prefs: []
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/4009-00000003
    -- allocPBXChannel: using nativeformats=(alaw) fmt=alaw
    -- SCCP/4009-00000003: rtp server instance created at 10.4.48.253:16632
  == SCCP/4009-00000003: update rtpmap: format:CISCO-DTMF, payload:101, mime:audio, rate:0
    -- SCCP/4009-00000003: (sccp_rtp_createRtpServer) updated phone Audio RTP destination to : 10.4.48.253:16632, family:IPv4, mapped: False
    -- 49: Allocated asterisk channel SCCP/4009-00000003
    -- SEP0004F2EB452A: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/4009-00000003, lineInstance:1 (previous channelstate:DOWN)
    -- SEP0004F2EB452A: Finish to indicate channel state:OFFHOOK on call:SCCP/4009-00000003, lineInstance:1. New channel state:OFFHOOK
    -- SCCP/4009-00000003: schedule digittimeout 16
    -- SEP0004F2EB452A: is Onhook (buttonIndex: 1, callid: 3)
    -- SCCP/4009-00000003: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:3) (hangup_id:-1) (cfwd_noanswer_id:-1)
  == SEP0004F2EB452A: Ending call SCCP/4009-00000003 (state:OFFHOOK)
    -- SEP0004F2EB452A: Sending hangupRequest to Call SCCP/4009-00000003 (state: OFFHOOK)
SCCP/4009-00000003: (RequestQueueHangup):
 - pbx_channel: SCCP/4009-00000003
 - channel_flags:
   - ZOMBIE: NO
   - BLOCKING: NO
 - pbx_channel_hangup_locked: NO
 - runningPbxThread: NO
 - pbx_channel_is_bridged: NO
 - pbx_channel_pbx: NO
SCCP/4009-00000003: (RequestQueueHangup): Hard Hangup
    -- SCCP/4009-00000003: Asked to hangup channel.
[2020-04-03 14:45:53] WARNING[4977]: sccp_session.c:165 sccp_session_waitForPendingRequests: SEP0004F2EB452A: waitForPendingRequests timed out!
  == SCCP/4009-00000003: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on device SEP0004F2EB452A
       > SEP0004F2EB452A: Stopping PBX audio rtp transmission on channel SCCP/4009-00000003
    -- SCCP/4009-00000003: Current pbxchannelstate OFFHOOK(10)
    -- SEP0004F2EB452A: Indicate SCCP new state:ONHOOK, current channel state:OFFHOOK on call:SCCP/4009-00000003, lineInstance:1 (previous channelstate:DOWN)
    -- SEP0004F2EB452A: Finish to indicate channel state:ONHOOK on call:SCCP/4009-00000003, lineInstance:1. New channel state:DOWN
    -- SEP0004F2EB452A: Device is Requesting Connections Statistics And Clear
    -- SCCP: Cleaning channel SCCP/4009-00000003
  == SCCP/4009-00000003: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on device SEP0004F2EB452A
    -- SEP0004F2EB452A: Set the active channel 0 on device
    -- SEP0004F2EB452A: Looking for selected channel (3)
    -- Destroying channel SCCP/4009-00000003
  == SCCP/4009-00000003: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on device SCCP
    -- SCCP: destroying PBX rtp server on channel SCCP/4009-00000003
raspbx*CLI>
```

dkgroot, 2020-04-03:
@Zw1d I will have a good look at your looks tomorrow, I think your might have already partially solved it by finding the: "waitForPendingRequests" message. Give me a little bit of time to figure out the cause and possible solution.

Zw1d, 2020-04-04:
@dkgroot unfortunately I don't have access to the phones on the weekend. Unless there's a way to simulate it remotely? I have access to the network.

dkgroot, 2020-04-04:
You can control the phone from remote via the asterisk CLI
```
asterisk -rvvvv
CLI> sccp debug none
CLI> sccp debug core, channel, indicate, pbx, rtp, codec, session, action
CLI> sccp call SEP0004F2EB452A 4009
CLI> sccp onhook SCCP/4009-xxxxxxxx
```
For the last line use the `tab` key to complete the channel name

Zw1d, 2020-04-06:
Okay, here we go:
successful call: https://pastebin.com/792VcKeJ
unsuccessful call: https://pastebin.com/x16qaSDd

dkgroot, 2020-04-06:
@Zw1d 
I am sorry i said "session" where i meant "socket" in the debug string. Can you try again and set
```
sccp debug core, channel, indicate, pbx, rtp, codec, socket, action
```
Sorry about that.

Zw1d, 2020-04-06:
All good,
I've restarted the phone from console, but dialed manually, I hope that doesn't matter.
successful call: https://pastebin.com/Na1BCaYq
unsuccessful call: https://pastebin.com/8T6QSeyN

Since we restarted the device, first call is fine.

dkgroot, 2020-04-06:
Nope manual dial is perfect
Looking into the issue now

Zw1d, 2020-04-06:
@dkgroot I did `git checkout 7d2ea9` like you suggested, should I get back to latest?

dkgroot, 2020-04-06:
Sorry for the confusion. I am normally just looking at the latest source, trying to match it up to the debug log. Having another look in a bit.

dkgroot, 2020-04-06:
Found the cause:
`grep -e Request -e Response successful.call`
Shows
```
    -- SEP7C95F3C8138C: Device is Requesting Connections Statistics And Clear
    -- SEP7C95F3C8138C: Request 'Connection Statistics Request' to device Pending
    -- SEP0004F2EB452A: Device is Requesting Connections Statistics And Clear
    -- SEP0004F2EB452A: Request 'Connection Statistics Request' to device Pending
    -- SEP7C95F3C8138C: Response 'Connection Statistics Response' Received
```
Looks like the SEP0004F2EB452A never sends back a Statistics Response. Causing our pendingRequest counter to increase and never go down.
Fixes:
1. Reset the pendingRequest counter if we timeout while waiting for it
2. Add an exception for this phone type.

dkgroot, 2020-04-06:
***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/XXX origin/issues/509
git checkout issues/509
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

Zw1d, 2020-04-06:
Yup, now it works as expected!

dkgroot, 2020-04-06:
Ok merging the branch back into develop. So you can switch back to develop in a minute or two.

Thank you for reporting the issue and being a patient with me while fixing it..

closed 2020-04-06 by dkgroot

Zw1d, 2020-04-06:
@dkgroot thank You for such a fast fix!
Best regards