chan-sccp/chan-sccp #510: Possible memory leak when closing session

mpaland, 2020-04-06:
I'm running multiple asterisk 16 environments with pjsip. On the one which is running pjsip and chan sccp I can observe increasing process memory usage.
Have a look at this graph:
[image]
In October the router crashed, after the downtime in Nov asterisk was upgraded to v16 and chan sccp to its actual version.
Right from this moment, issue #492 came up. I haven't had enough time for its investigation right now, so I commented the according phone (SEP001DE5EA99B8) out in the config.
When memory falls down, asterisk was restarted.
What I can see in the trace is this periodical event:

```
[Apr  6 11:39:10] NOTICE[30483]: sccp_session.c:676 sccp_session_device_thread: SCCP: Closing session because connection timed out after 120 seconds (ip-address: 192.168.2.1:2000).
  == SCCP: Stopping Session Thread
    -- SCCP: Exiting sccp_socket device thread
    -- SCCP: Session without a device attached !
    -- SCCP: cleanup session
    -- SCCP: Destroy Session 192.168.2.153
    -- SCCP: Shutdown socket 31
    -- SCCP: Closing socket 31
    -- SCCP: Accepted Client Connection from 192.168.2.153:36022
    -- SCCP: Connected on server via 192.168.2.1:2000
       > 192.168.2.1:2000: keepalive:62, keepaliveinterval:60
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
    -- SCCP: Device 'SEP001DE5EA99B8' not found in realtime table 'sccpdevice'
[Apr  6 11:39:10] NOTICE[30775]: sccp_actions.c:740 handle_SPCPTokenReq: SEP001DE5EA99B8: Rejecting device: not found
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
    -- SCCP: Device 'SEP001DE5EA99B8' not found in realtime table 'sccpdevice'
[Apr  6 11:40:10] NOTICE[30775]: sccp_actions.c:740 handle_SPCPTokenReq: SEP001DE5EA99B8: Rejecting device: not found
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
    -- SCCP: Device 'SEP001DE5EA99B8' not found in realtime table 'sccpdevice'
[Apr  6 11:41:10] NOTICE[30775]: sccp_actions.c:740 handle_SPCPTokenReq: SEP001DE5EA99B8: Rejecting device: not found
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
    -- SCCP: Device 'SEP001DE5EA99B8' not found in realtime table 'sccpdevice'
[Apr  6 11:42:10] NOTICE[30775]: sccp_actions.c:740 handle_SPCPTokenReq: SEP001DE5EA99B8: Rejecting device: not found
[Apr  6 11:44:10] NOTICE[30775]: sccp_session.c:676 sccp_session_device_thread: SCCP: Closing session because connection timed out after 120 seconds (ip-address: 192.168.2.1:2000).
  == SCCP: Stopping Session Thread
    -- SCCP: Exiting sccp_socket device thread
    -- SCCP: Session without a device attached !
    -- SCCP: cleanup session
    -- SCCP: Destroy Session 192.168.2.153
    -- SCCP: Shutdown socket 30
    -- SCCP: Closing socket 30
    -- SCCP: Accepted Client Connection from 192.168.2.153:44531
    -- SCCP: Connected on server via 192.168.2.1:2000
       > 192.168.2.1:2000: keepalive:62, keepaliveinterval:60
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
    -- SCCP: Device 'SEP001DE5EA99B8' not found in realtime table 'sccpdevice'
[Apr  6 11:44:10] NOTICE[31047]: sccp_actions.c:740 handle_SPCPTokenReq: SEP001DE5EA99B8: Rejecting device: not found
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
    -- SCCP: Device 'SEP001DE5EA99B8' not found in realtime table 'sccpdevice'
[Apr  6 11:45:10] NOTICE[31047]: sccp_actions.c:740 handle_SPCPTokenReq: SEP001DE5EA99B8: Rejecting device: not found
  == SEP001DE5EA99B8: is requesting a token, Instance: 16777216, Type: Cisco SPA 524SG (80001)
    -- SCCP: Device 'SEP001DE5EA99B8' not found in realtime table 'sccpdevice'
[Apr  6 11:46:10] NOTICE[31047]: sccp_actions.c:740 handle_SPCPTokenReq: SEP001DE5EA99B8: Rejecting device: not found
```
Is the memory of the destroyed session really freed?
This is not really a big problem and normally all phones are registered, but just for cleanness.

dkgroot, 2020-04-06:
This is a problem that needs to be fixed. It can lead to serious problems on smaller machine. Thank for monitoring this and telling me about it. I was planning to run some valgrind anyway, so we will have to figure out where the leak is occuring.

dkgroot, 2020-04-06:
I found a refcount issue in realtime lines causing device memory leak.
trying to trace it down

dkgroot, 2020-04-06:
***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/XXX origin/issues/510
git checkout issues/510
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

dkgroot, 2020-04-07:
I ran several tests and don't see a reason why not to merge it right now. So i am going to go ahead and do so.
@mpaland Thank you Marco for reporting this and keeping a close eye on memory consumption, much appreciated.

mpaland, 2020-04-07:
Again, thanks a lot.
I'll compile the latest develop version and have an eye on it the next days/weeks and let you know here if it works well.

mpaland, 2020-04-07:
I applied latest https://github.com/chan-sccp/chan-sccp/commit/abfda3e719377d61745e233b3480658375ef6243 , but just after 3 hours I can still observe increasing memory.
Too early for a trend, but there still seems to be something left. 😞

dkgroot, 2020-04-08:
Are you sure this leak is in chan-sccp ? Ok i will have another look. I used the memory debug in asterisk to find the previous leaks, by loaded/unloading the module. Increase over loaded time, is a little more difficult to find, but i will try.

mpaland, 2020-04-08:
Here is the log of last night, there weren' any incoming or outgoing calls, but memory is increasing constantly.
[image]
Tonight I'll unload chan-sccp to verify if this is really the cause.

Meanwhile, you could try this without any calls:
1. Physically disconnect one registered phone
2. Comment out in the config file another physically connected phone
3. Wait a few hours

dkgroot, 2020-04-08:
I have tried closing the port on my switch + resetting the device + disconnecting the cable + physical powercycle of the phone, but that seemed all ok. I have not tryied commenting out a device (i take it you reloaded the config after commenting (right?)).

dkgroot, 2020-04-08:
I copied my sccp.conf to sccp.conf.withoutSEPExxxxx
`sccp reload file sccp.conf.withoutSEPExxxxx`
waited untill device tried to reconnect
`sccp reload file sccp.conf`
to switch back to the original
result:
```
--- summary.prev	2020-04-08 15:27:09.830859630 +0200
+++ summary.log	2020-04-08 15:28:19.294858314 +0200
@@ -1,7 +1,8 @@
       1536 bytes in          2 allocations in file chan_sccp.c
       1792 bytes in          2 allocations in file sccp_actions.c
       2688 bytes in          4 allocations in file sccp_callinfo.c
-     15724 bytes in        521 allocations in file sccp_config.c
+        53 bytes in          1 allocations in file sccp_cli.c
+     15700 bytes in        520 allocations in file sccp_config.c
       1824 bytes in         12 allocations in file sccp_device.c
       1440 bytes in         10 allocations in file sccp_event.c
        216 bytes in          4 allocations in file sccp_featureParkingLot.c
--- allocs.prev	2020-04-08 15:27:10.046859626 +0200
+++ allocs.log	2020-04-08 15:28:19.566858309 +0200
@@ -3,11 +3,11 @@
         48 bytes allocated by boolean_t sccp_prePBXLoad(void)() line   174 of chan_sccp.c
        896 bytes allocated by btnlist *sccp_make_button_template(sccp() line  1030 of sccp_actions.c
        896 bytes allocated by btnlist *sccp_make_button_template(sccp() line  1030 of sccp_actions.c
+        53 bytes allocated by int sccp_cli_reload(int, int, char **)() line  3037 of sccp_cli.c
        672 bytes allocated by sccp_callinfo_t *const callinfo_Constru() line    96 of sccp_callinfo.c
        672 bytes allocated by sccp_callinfo_t *const callinfo_Constru() line    96 of sccp_callinfo.c
        672 bytes allocated by sccp_callinfo_t *const callinfo_Constru() line    96 of sccp_callinfo.c
        672 bytes allocated by sccp_callinfo_t *const callinfo_Constru() line    96 of sccp_callinfo.c
-        10 bytes allocated by sccp_config_file_status_t sccp_config_g() line  2797 of sccp_config.c
         10 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
         10 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
         10 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
@@ -64,8 +64,8 @@
         15 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
         15 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
         15 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
-        15 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
         18 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
+         1 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
         22 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
         26 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
         26 bytes allocated by sccp_configurationchange_t sccp_config_() line   487 of sccp_config.c
```
So sccp reload is causing another small memory leak. Looking into it.

dkgroot, 2020-04-08:
Note: the memory leak I have been able to find since the initial patch is really tiny. For them to show up in your graph would take a long long time i would assume. Not sure what the current resident memory of your asterisk process is, but i would guess something like >1Mb. This might suggest the issue is in one of the other asterisk modules. Can you have a look at the asterisk changelog to see if any memory leaks were reported/fixed since the revision you are running ?

mpaland, 2020-04-08:
These actual tests were made with 16.9.0, the lastest LTS one.
I'm using the very same compiled version on 4 similar systems, 3 are stable concerning process memory. The only one with the problem has chan-sccp loaded, therefore my report.
I'm doing some overnight tests with the unloaded driver the next days. to track down the problem.

Yes, I restarted asterisk every time the configuration changed.

Last night I had a problem with my VPN tunnel, so one phone was not reachable, therefore the idea with the disconnected phone.

dkgroot, 2020-04-08:
Created another issues/510 branch for further testing.

***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/XXX origin/issues/510
git checkout issues/510
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

dkgroot, 2020-04-08:
> These actual tests were made with 16.9.0, the lastest LTS one.
> I'm using the very same compiled version on 4 similar systems, 3 are stable concerning process memory. The only one with the problem has chan-sccp loaded, therefore my report.

Ok, that is good to know.

> I'm doing some overnight tests with the unloaded driver the next days. to track down the problem.
> 
> Yes, I restarted asterisk every time the configuration changed.

If you could run asterisk with DEBUG_MALLOC enabled, that would make finding the cause a lot easier i think.
> Last night I had a problem with my VPN tunnel, so one phone was not reachable, therefore the idea with the disconnected phone.

dkgroot, 2021-01-01:
@mpaland Can you check with the latest develop version to see if the issue is already resolved, I managed to find a couple of small potential leaks a couple of weeks/days ago. If the issue is already resolved, please close the issue report. If not we need to spend a little more work on reproducibility.

mpaland, 2021-01-03:
Test with 066a14a is in progress.

mpaland, 2021-01-03:
While installing this update, I rechecked the memory graph. Have a look:
[image]
I installed ad4e0deae676ef97525eada867a8ba6a7dc2220a on Nov 14th and as you can see it improved the memory issue significantly! The zigzags represent asterisk restarts, mostly together with an update.
I observe this the next week for the actual version, but I think, we can close this issue then.
Really good work, Diederik!

dkgroot, 2021-01-03:
Thanks Marco,
That really does confirm the issue looks to be resolved.

mpaland, 2021-01-05:
Compared to older versions, it's way better, but a small memory leak problem is still present.
Have a look, the pink area is increasing slowly:
[image]
On machines with a similar setup, but without chan-sccp, the pink memory line is absolutely flat.

dkgroot, 2021-01-05:
Hi Marco,
Ok i will try to figure out where the leak might be. I might need a little help in form of configure options used and of a sccp.conf if i cannot find it with mine. Thanks for keeping an eye on this !

mpaland, 2021-01-05:
Compiling and setting up a testversion for the production system with DEBUG_MALLOC to identify the cause.

dkgroot, 2021-01-05:
Hi Marco,
I think i found one if i do:
```
sccp reload force
sccp restart SEPxxxxxxx
```

[dkgroot:/data … rces/chan-sccp/src] develop(+1/-1)+* 😊> ./collect_memleak 
```
--- mem_summary	2021-01-05 20:10:24.699835510 +0100
+++ mem_summary_new	2021-01-05 20:10:38.671545744 +0100
@@ -3,7 +3,7 @@
       2016 bytes in          3 allocations in file sccp_callinfo.c
      16203 bytes in        551 allocations in file sccp_config.c
       1976 bytes in         13 allocations in file sccp_device.c
-       728 bytes in          5 allocations in file sccp_devstate.c
+       872 bytes in          6 allocations in file sccp_devstate.c
       1440 bytes in         10 allocations in file sccp_event.c
       1928 bytes in          8 allocations in file sccp_hint.c
         88 bytes in          1 allocations in file sccp_line.c
--- mem_allocations	2021-01-05 20:10:24.915831030 +0100
+++ mem_allocations_new	2021-01-05 20:10:38.891541182 +0100
@@ -701,6 +701,7 @@
        144 bytes allocated by SubscribingDevice_t *addSubscriber(devi() line   155 of sccp_devstate.c
        144 bytes allocated by SubscribingDevice_t *addSubscriber(devi() line   155 of sccp_devstate.c
        144 bytes allocated by SubscribingDevice_t *addSubscriber(devi() line   155 of sccp_devstate.c
+       144 bytes allocated by SubscribingDevice_t *addSubscriber(devi() line   155 of sccp_devstate.c
        152 bytes allocated by deviceState_t *createDeviceStateHandler() line   296 of sccp_devstate.c
        144 bytes allocated by void sccp_event_module_start(void)() line   135 of sccp_event.c
        144 bytes allocated by void sccp_event_module_start(void)() line   135 of sccp_event.c
```

mpaland, 2021-01-05:
This system is mostly idle, no restarts/reloads etc for days/weeks. Just a few external calls.

dkgroot, 2021-01-05:
You might be able to use the same script (or a version that collects multiple captures, 1 per hour and diffs against previous).

commit 409a3c7d (2020-04-07): Fix sccp_mwi subscription memory leak
Fix refcount issue leading to memory leak

Fixes: #510
commit abfda3e7 (2020-04-07): Merge branch 'issues/510' into develop

Fixes: #510