chan-sccp/chan-sccp #511: Outbound Calls do not play ringing tone to caller

Jed-Giblin, 2020-04-10:
Using a 7962, when an outbound call is placed, the caller doesn't hear the ringing tone from the phone. The call connects fine and voice quality is not an issue.  

**Reproduction Steps:**

1. Place an outbound call via a SIP trunk

**chan-sccp version:** develop @ c8af12d4f

**asterisk version:** 13.18.3~dfsg-1ubuntu4

**Expected behavior:** 

Ringing sound can be heard by caller

**Observed behavior:**

No sound is heard by caller.

**Screenshots**

N/A

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Unknown
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - Intel Xeon E3-1225
  - 64 Bit
  - Little Endian
  - Ubuntu
  - 18.04
- Direct Contact information: Here, gitter
- Log Files Will collect and attach shortly
- Comments:

Also using Freepbx 14

dkgroot, 2020-04-10:
@Jed-Giblin / @rolakron / @
Turned out the be a firewall RTP/UDP forwarding issue in this case. 

Internal calls where producing simulated ringing as normal. External Calls never received a 180 RINGING from the provider but immediately went from DIALING to 183 Progress, indicating we should start receiving RTP packet with remote ringing sounds (inband). These packets were however being blocked.

dkgroot, 2020-04-11:
The issues seem to be cause by firewall blocking rtp flow. In the 509 branch that was merged a while back, i had removed some code, that would automatically punch a hole through the firewall from the inside, to allow traffic back in through that same port. I guess some people started relying on this behavior even for other channel drivers like chan_sip and chan_pjsip. So i decided to re-add the hole punching behavior in branch "issues/511_1" for testing.

The new implementation is slightly modified to close the start rtp flow as soon as the hole is open, so that we are not actually transmitting audio during PROGRESS/RING-OUT.

Please give it a go on your test system to see if this works for you.

During testing you can set:
```
sccp debug core, channel, rtp, indicate, pbx
core set verbose 5
core set debug 1
rtp set debug on
```
For a single call, to see the hole punch in action. (First packet is send during progress (183) and then media transmission is terminated again, until the call is answered).

Jed-Giblin, 2020-04-12:
I would love to test this, but I don't have any phones for my test system right now, and it will probably be a while before I can get any with everything going on right now

dkgroot, 2020-04-12:
@Jed-Giblin No problem, i think @rolakron might be able to help

dkgroot, 2020-05-12:
Finally made some headway with all of this in a new branch:

Looks like the ringing issue has been resolved (AFAIK):

Note: Some FreePBX outbound trunks might require "R" to be added to the Dial_options, which can be done per trunk or globally (settings->advanced settings->outbound_trunk_dial_options), so that simulated ringing will be performed by asterisk when the remote end of the trunk does not send inband audio and does not signal RINGING(180) or PROGRESS(183), this should be rare though.

***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/511_3 origin/issues/511_3
git checkout issues/511_3
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

dkgroot, 2020-05-13:
@Jed-Giblin Don't worry about it. Already sent a message to rolakron :-) Thanks !

commit 1b593aa5 (2020-05-29): Simple implementation of earlyrtp + holepunching
Replicate changes to other asterisk versions

Fixes: #511

Caveat: DirectRTP + Nat is currently broken (after the call_answer patches), and needs work.
commit 3fcda7f9 (2020-05-29): Merge branch 'issues/511_3' into develop

Fixes: #511
closed 2020-05-29 by dkgroot