chan-sccp/chan-sccp #512: Creating a custom phone book Cisco ip phone

clone-hash, 2020-04-13:
Made the  custom phone book work on my PC following the Wiki,   but would like to make this work whit my USB drive   on my Router, did check to see if the address  works whit a browser and it is Ok , the phone seems to be able to navigate to the USB drive because when y hit the directories key on the phone y don't see the Request Prompt looking for a connection ,it would seem that the phone is finding the path to the USB  drive folder label  Directory . XML, but it doesn't populate the list which should be after  Placed Calls in the  directories list,   this is the command lines ( file:\192.168.0.1\usbkek512\directory.xml ) that i'm using to reach the directory  on my USB drive  and when y test this line whit a browser it works  fine.

Can this be done using a Router  instead of running a HTTPserver on my PC 

Thank you in advance for any help

dkgroot, 2020-04-14:
I would say no. A file:// URI is always local to the machine on which the URI is entered (https://en.wikipedia.org/wiki/File_URI_scheme).  So the phone has no way to reach it over the network. I don't see how a browser is able to do this, but it might actually trying to help and "fix" the uri you have given. 

However if your router is also running the tftp server to provision the phones you can use it to servea static xml file to the phone using a TFTP:// path instead. You have not told us anything about the router, so i don't know what it is capable of.

clone-hash, 2020-04-15:
Hello Diederik , and thank you for your help,
I forgot to mention the router detail, so here it is , it is an  Asus RT-AC68U  Wifi router  loaded whit  Asuswrt-Merlin vers.384.15 ,  the USB drive that is attached to  one of the Asus port is shared to everyone on my Lan network whit no user, password , so like i mention if i browse the URL address of the router whit the path to the USB drive and the files it works, then when y  put the same path for the URL in the XML config  for the directory in the phone it should work to,  and it seams to partially work since the phone does not prompt the message that is not finding the Directory.xml , it is just not finding the info for the Phone-book entry .so is there  a problem in the Syntax for the phone to parse the phone-book  page.

This is the URL in the phone XMl Sep Config for the directories URL  file://192.168.0.1/usbkek512/directory.xml/<file://192.168.0.1/usbkek512/phonebook-1.xml</>
This is the Phone-book  URL  file://192.168.0.1/usbkek512/phonebook-1.xml/<file://192.168.0.1/usbkek512/phonebook-1.xml</>

My setup for the phone doesn't include a  Running TFTP server, y only start  TFTP64 Server on my PC to load the phone, once it is loaded the phone runs standalone whit my Sip Provider and it works very well

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-green-avg-v1.png]<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>     Virus-free. www.avg.com<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

dkgroot, 2020-04-16:
@clone-hash I don't know how you are sharing the content of the usb stick to all PC's but i am guessing SMB (or samba). Cisco phones don't know how to access a SMB drive and do therefor have not way to participate in this sharing. Simple answer this will not work.

You can re-share the file via your TFTP64 to the phone or setup a webserver on your PC to do the same. The Cisco Phones are embedded devices and limited in their abilities.

closed 2021-01-01 by dkgroot