chan-sccp/chan-sccp #513: Error compiling on Centos 6.6

bwhiteford, 2020-04-16:
Greetings chan-sccp community!

We've been a loyal user of this wonderful module for years for our business phone system, and decided today to run an upgrade on packages on our PBX that were very old, including Asterisk. I expected to need to recompile chan-sccp, but we're running into a compile error.

 **This is the output of configure**:

```
Operating System : Linux
          Kernel : 2.6.32-642.6.2.el6.i686
    Architecture : i686
Asterisk Version : 11300 / 113
  Asterisk Repos : TRUNK
      Build user : root
         Version : 4.3.2_develop_rabfda3e
Compiler Version : 4.4.7
              CC : gcc
             CPP : gcc -E
         HOST_CC : gcc
          CFLAGS : -O0 -Wall -I. -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -std=gnu99 -fgnu89-inline -Wno-ignored-qualifiers  -isystem /usr/include -DHAVE_ASTERISK -g3 -ggdb3 -pthread
        CPPFLAGS : -U_FORTIFY_SOURCE  -isystem /usr/include -DHAVE_ASTERISK  -g3 -ggdb3
         LDFLAGS :  -lpthread -lc       -L/usr/lib
        PBX_TYPE : Asterisk
      PBX_PREFIX : /usr
         PBX_ETC : /etc/asterisk
         PBX_LIB : /usr/lib
      PBX_VARLIB : /var/lib/asterisk
     PBX_DATADIR : /var/lib/asterisk
     PBX_INCLUDE : /usr/include/asterisk
Module Directory : /usr/lib/asterisk/modules
```

**And here is where make fails:**
`
In file included from sccp_xml.h:16,
                 from sccp_xml.c:13:
forward_declarations.h:6: error: redefinition of typedef ‘sccp_group_t’
forward_declarations.h:6: note: previous declaration of ‘sccp_group_t’ was here
forward_declarations.h:9: error: redefinition of typedef ‘sccp_session_t’
forward_declarations.h:9: note: previous declaration of ‘sccp_session_t’ was here
forward_declarations.h:10: error: redefinition of typedef ‘sccp_linedevice_t’
forward_declarations.h:10: note: previous declaration of ‘sccp_linedevice_t’ was here
forward_declarations.h:11: error: redefinition of typedef ‘sccp_device_t’
forward_declarations.h:11: note: previous declaration of ‘sccp_device_t’ was here
forward_declarations.h:12: error: redefinition of typedef ‘sccp_line_t’
forward_declarations.h:12: note: previous declaration of ‘sccp_line_t’ was here
forward_declarations.h:13: error: redefinition of typedef ‘sccp_channel_t’
forward_declarations.h:13: note: previous declaration of ‘sccp_channel_t’ was here
forward_declarations.h:14: error: redefinition of typedef ‘sccp_speed_t’
forward_declarations.h:14: note: previous declaration of ‘sccp_speed_t’ was here
forward_declarations.h:15: error: redefinition of typedef ‘sccp_service_t’
forward_declarations.h:15: note: previous declaration of ‘sccp_service_t’ was here
forward_declarations.h:16: error: redefinition of typedef ‘sccp_addon_t’
forward_declarations.h:16: note: previous declaration of ‘sccp_addon_t’ was here
forward_declarations.h:17: error: redefinition of typedef ‘sccp_hint_t’
forward_declarations.h:17: note: previous declaration of ‘sccp_hint_t’ was here
forward_declarations.h:18: error: redefinition of typedef ‘sccp_hostname_t’
forward_declarations.h:18: note: previous declaration of ‘sccp_hostname_t’ was here
forward_declarations.h:19: error: redefinition of typedef ‘sccp_rtp_t’
forward_declarations.h:19: note: previous declaration of ‘sccp_rtp_t’ was here
forward_declarations.h:20: error: redefinition of typedef ‘sccp_header_t’
forward_declarations.h:20: note: previous declaration of ‘sccp_header_t’ was here
forward_declarations.h:21: error: redefinition of typedef ‘sccp_msg_t’
forward_declarations.h:21: note: previous declaration of ‘sccp_msg_t’ was here
forward_declarations.h:42: error: redefinition of typedef ‘sccp_devstate_specifier_t’
forward_declarations.h:42: note: previous declaration of ‘sccp_devstate_specifier_t’ was here
forward_declarations.h:44: error: redefinition of typedef ‘sccp_featureConfiguration_t’
forward_declarations.h:44: note: previous declaration of ‘sccp_featureConfiguration_t’ was here
forward_declarations.h:45: error: redefinition of typedef ‘sccp_selectedchannel_t’
forward_declarations.h:45: note: previous declaration of ‘sccp_selectedchannel_t’ was here
forward_declarations.h:46: error: redefinition of typedef ‘sccp_ast_channel_name_t’
forward_declarations.h:46: note: previous declaration of ‘sccp_ast_channel_name_t’ was here
forward_declarations.h:47: error: redefinition of typedef ‘sccp_buttonconfig_t’
forward_declarations.h:47: note: previous declaration of ‘sccp_buttonconfig_t’ was here
forward_declarations.h:48: error: redefinition of typedef ‘sccp_hotline_t’
forward_declarations.h:48: note: previous declaration of ‘sccp_hotline_t’ was here
forward_declarations.h:49: error: redefinition of typedef ‘sccp_callinfo_t’
forward_declarations.h:49: note: previous declaration of ‘sccp_callinfo_t’ was here
forward_declarations.h:50: error: redefinition of typedef ‘sccp_call_statistics_t’
forward_declarations.h:50: note: previous declaration of ‘sccp_call_statistics_t’ was here
forward_declarations.h:51: error: redefinition of typedef ‘sccp_softKeySetConfiguration_t’
forward_declarations.h:51: note: previous declaration of ‘sccp_softKeySetConfiguration_t’ was here
forward_declarations.h:52: error: redefinition of typedef ‘sccp_mailbox_t’
forward_declarations.h:52: note: previous declaration of ‘sccp_mailbox_t’ was here
forward_declarations.h:53: error: redefinition of typedef ‘sccp_subscription_id_t’
forward_declarations.h:53: note: previous declaration of ‘sccp_subscription_id_t’ was here
forward_declarations.h:54: error: redefinition of typedef ‘sccp_conference_t’
forward_declarations.h:54: note: previous declaration of ‘sccp_conference_t’ was here
forward_declarations.h:55: error: redefinition of typedef ‘sccp_private_channel_data_t’
forward_declarations.h:55: note: previous declaration of ‘sccp_private_channel_data_t’ was here
forward_declarations.h:56: error: redefinition of typedef ‘sccp_private_device_data_t’
forward_declarations.h:56: note: previous declaration of ‘sccp_private_device_data_t’ was here
forward_declarations.h:57: error: redefinition of typedef ‘sccp_cfwd_information_t’
forward_declarations.h:57: note: previous declaration of ‘sccp_cfwd_information_t’ was here
forward_declarations.h:58: error: redefinition of typedef ‘sccp_buttonconfig_list_t’
forward_declarations.h:58: note: previous declaration of ‘sccp_buttonconfig_list_t’ was here
forward_declarations.h:59: error: redefinition of typedef ‘sccp_threadpool_t’
forward_declarations.h:59: note: previous declaration of ‘sccp_threadpool_t’ was here
forward_declarations.h:60: error: redefinition of typedef ‘xmlDoc’
forward_declarations.h:60: note: previous declaration of ‘xmlDoc’ was here
forward_declarations.h:61: error: redefinition of typedef ‘xmlNode’
forward_declarations.h:61: note: previous declaration of ‘xmlNode’ was here
forward_declarations.h:65: error: redefinition of typedef ‘boolean_t’
forward_declarations.h:65: note: previous declaration of ‘boolean_t’ was here
forward_declarations.h:89: error: redefinition of typedef ‘sk_func’
forward_declarations.h:89: note: previous declaration of ‘sk_func’ was here
forward_declarations.h:90: error: redefinition of typedef ‘sccp_sched_cb’
forward_declarations.h:90: note: previous declaration of ‘sccp_sched_cb’ was here
cc1: warning: unrecognized command line option "-Wno-format-truncation"
make[3]: *** [libsccp_la-sccp_xml.lo] Error 1
make[3]: Leaving directory `/root/Downloads/chan-sccp/src'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/root/Downloads/chan-sccp/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/root/Downloads/chan-sccp/src'
make: *** [all-recursive] Error 1
`

Are you guys able to lend me a hand? Thank you so much!

dkgroot, 2020-04-16:
Look like there is an issue with this old compiler and the sourcecode. But first things first, which version of chan-sccp is this and where did you get it from.

bwhiteford, 2020-04-16:
Wow Diederik, thank you for the amazing support! That fix was all we needed. I apologize for not trying the new compiler first. I saw the current compiler version met the build requirements, so I thought I'd give it a try. I will use a newer version in the future. Everything's working great now!

Much appreciated,
Brooks

dkgroot, 2020-04-16:
Ok thanks for the update !! Closing this issue for now

closed 2020-04-16 by dkgroot