chan-sccp/chan-sccp #516: CID Issue

justinkoby, 2020-04-20:
Yep, another CID issue here. I've been trying to go through the issue 487 to see if that resolves my issue, but it doesn't seem to be helping me figure mine out.

I am running FreePBX 15 with Chan-sccp 4.3.2.

When Outbound CID has any value in it, that value is displayed for internal calls (SCCP --> SCCP as well as SCCP --> SIP).

Here's the log of sccp debug core,channel,pbx,callinfo & core set debug 1 & core set verbose 5
https://pastebin.com/3ZCp2802

Is this going to be like 487 where it's a freepbx issue? To me, it looks like the CID is populated before I even begin dialing.

justinkoby, 2020-04-23:
Any thoughts on this?  I'm still reading through the logs and trying to make sense of it. 

On this line:

`Executing [s@macro-user-callerid:2] Set("SCCP/109-00000078", "AMPUSER=9999999999") in new stack`

Is the AMPUSER=9999999999 how the pbx looks up the caller's information in the AMP database?

justinkoby, 2020-04-23:
OMG. Using this page:
https://pbxinaflash.com/community/threads/different-callerids-for-different-outbound-routes-with-sccp.18883/

I tried changing the CID_NUM value in the sccpline table in the database to be the internal CID for the phone. It appears to work when I make that change.

Internal to Internal shows the extension number & name of person calling.

Internal to External shows the external CID number that is programmed as the Outbound CID for the extension.

If I make a change in the extension page after doing the manual change, that overwrites the CID_NUM in the sccpline table with the value entered in the Outbound CID.

dkgroot, 2020-04-23:
@justinkoby Does that mean you figure out a way to fix the problem, and we just need to update/fix sccp_manager to reflect this finding ?

I cannot fully follow what you are changing in the database and what the change in sccp_manager/Freepbx would have to be/look like, to be honest.

justinkoby, 2020-04-23:
I may have figured out a way to fix it.

Ok. So, here's what I observe with some basic/dummy data.

Initially extension 109 is in the database as
ID: 200
CID_NUM: 99999999
Name: 109

If I run an update sccpline set CID_NUM=109 where ID=200 sql command it changes the database to read:
ID: 200
CID_NUM: 109
Name: 109

Everything _appears_ to work as it should.

Then, I go into the extension manager and make a change to the extension 109 (ie changing the displayname). When I look in the sccptable, I see the following:
ID: 212
CID_NUM: 99999999
Name: 109

It looks like it deletes the previous entry and readds it to the database with the incorrect CID_NUM.

dkgroot, 2020-04-23:
Thanks for explaining what you found and thanks for reporting the issue to sccp_manager !

It does clear everything up for me (I am assuming you are using "Name" it refers to the "cid_name" column in the database, right ?).

Apparently sccp_manager is doing three things not quite right/wrong in this case:
-  it should never use a delete/create cycle, merely to update a value or record, instead it should use an appropriate sql "UPDATE" statement, and build up the statement so it cover's only the value that have been changed in the form;
-  it should not be filling any fields with default string values just to get around a "NOT NULL" requirement, these requirements are there for a reason;
- And it should be setting cid_name & cid_num with valid values so that devices know what cid to use during channel creation, without depending/hoping that FreePBX will take care of it. The later will not work correctly when using shared lines or line subscription id's.

justinkoby, 2020-04-23:
No, there is a separate CID_NAME column that I'm not including in my description here. In SCCPLINE, "NAME" is the extension number. Not sure why its called that, but that's whats populated in the field.

dkgroot, 2020-04-23:
Ok, thanks for clearing that up. Yes name might/could be a little confusing, but in chan_sccp we call lines the thing you attach to a phone and the extension the entry in the dialplan to dial that line. I guess the names depends a little on your perspective.