chan-sccp/chan-sccp #519: Some calls from queue have to be picked up twice

Zw1d, 2020-05-18:
At first I wanted to make it clear that I think this is quite likely NOT chan-sccp fault! However I wanted to ask advice on how I could diagnose it. Any hints would be much appreciated!

**Reproduction Steps:**

1. Call external number
2. I don't know, get lucky I guess

**chan-sccp version:** Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 33da090

**asterisk version:** Asterisk 16.6.1

**Expected behavior:**

Call get connected / picked up on the first try.

**Observed behavior:**
This happens maybe in 1 out of 5 calls.
Incoming calls get placed in queue with 5 sccp agents (all local). 
Extensions ring, then one gets picked up, but caller doesn't get connected.
The caller that had this issue said he didn't hear even the queue music (like he wasn't "fully" placed in queue, since the queue did ring it's agents?), just silence, then got disconnected (I assume he gets disconnected when the agent picks up). He called again, got connected without issues.

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [No]
- Problem can be reliably reproduced, doesn't happen randomly: [No]
- Platform:
  - ARMv7 Processor rev 4 (v7l)
  - RasPBX
  - Raspbian GNU/Linux 10

dkgroot, 2020-05-18:
@Zw1d 

We did change the 'call answer' logic quite substantially recently. So to confirm if the issue is caused by this somewhat recent change, can you see 8faf32 if has the same effect (`git checkout 8faf32`) ? We have done quite a bit of testing with this new "call answer" logic, but in retrospect we might have skipped over the use of "Local Channels" and "optimizing out". If we can establish that the issue started after march 8th (revision: 8faf32), we would know what to focus our attention on during testing.

For me to reproduce the issue, are you using local channels in the queue and are they set to "optimize" out or not. To ask the question a different way, can you show me a queue definition to replicate your current situation.

Zw1d, 2020-05-18:
Since the issue can be tested only on my production system I will start with the latest version and report back.
But do you really think this could be sccp-chan module fault? Why isn't caller hearing music from the queue?
What do you mean by 
> are you using local channels in the queue and are they set to "optimize" out or not
? 
Channels are local to that queue.

dkgroot, 2020-05-18:
It would help if you show me the queue definition on the CLI (`queue show [queuename]`), so i can try an reproduce the issue. Also setting the debug level as requested should make it possible to analyze what is going on, and potentially point the finger at the case, it being chan_sccp or not.

> are you using local channels in the queue and are they set to "optimize" out or not

See: https://wiki.asterisk.org/wiki/display/AST/Local+Channel

Zw1d, 2020-05-18:
Okay here's my queue:
```
o*CLI> queue show 4099
4099 has 0 calls (max unlimited) in 'ringall' strategy (23s holdtime, 35s talktime), W:0, C:2, A:0, SL:100.0%, SL2:100.0% within 60s
   Members:
      Room1 (Local/4105@from-queue/n from hint:4105@ext-local) (ringinuse enabled) (Not in use) has taken 1 calls (last was 52 secs ago)
      Person1 (Local/4004@from-queue/n from hint:4004@ext-local) (ringinuse enabled) (Not in use) has taken no calls yet
      Person2 (Local/4001@from-queue/n from hint:4001@ext-local) (ringinuse enabled) (Not in use) has taken no calls yet
      Person3 (Local/4002@from-queue/n from hint:4002@ext-local) (ringinuse enabled) (Not in use) has taken 1 calls (last was 3 secs ago)
      Person4 (Local/4003@from-queue/n from hint:4003@ext-local) (ringinuse enabled) (Not in use) has taken no calls yet
   No Callers

o*CLI>
```

Yeah I use FreePBX, asterisk wasn't simple enough on the first glance to dive right in, but I'm slowly getting there.

dkgroot, 2020-05-18:
Ok so we are using "Local" channels which is fine. And that means i should run a couple of tests, to see if i can reproduce the issue on `vanilla` asterisk you are seeing. Being able to reproduce the problem makes it much easier to fix any potential issue.

- Note: If i cannot reproduce the issue on vanilla asterisk, it doesn't mean the problem does not exist, or/and is not produced by chan_sccp.
- Testing with the 8faf32 revision, would still help (I guess we have to wait for some downtime).
- Also it would help if you send the CLI debug output from your current system (when it is and isn't happening) so we can find the difference (You should be able to do that at this moment). As we are using Local channels, the dialplan output is also important, so please set `core set debug 1` and `core set verbose 5` before running both tests.

Zw1d, 2020-05-18:
All right, I will try to do that. So far I can confirm that the issue just occured on:
`Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 81f140a`
I will try to catch logs from it.

dkgroot, 2020-05-18:
I just set up a Queue on my vanilla asterisk and on a remote FreePBX machine and ran some tests with both (running latest chan_sccp). So far i have not seen what you are reporting. You mentioned "luck" which is not great when trying to reproduce the issue, that might indicate we have a timing issue on our hands (different hardware will react very differently in that scenario).

Zw1d, 2020-05-18:
Okay, I will change the debug. I think I will just hunt for that buggy call and try to pin down the logs.

Zw1d, 2020-06-01:
A week ago I have changed register string in my primary sip provider. I removed (I think) timeout ~90 from the end of the string. So changed from:
1234@host.com:password:1234@0.0.0.0~90
to:
1234@host.com:password:1234@0.0.0.0
And since then haven't experienced that particular issue, so hopefully that was it.

closed 2020-06-01 by Zw1d

dkgroot, 2020-06-01:
@Zw1d Thanks for reporting back and closing the issue. If the issue does pop up do open it up again.