chan-sccp/chan-sccp #520: Paging/Intercom dialing causes phones to briefly ring

Terrabytee, 2020-05-20:
I do not know when exactly this started, but for some reason, whenever I try to page my phones, they briefly ring before the auto-answer kicks in on the Cisco handsets I have (one 7945G and two 7965Gs). This did not used to happen.

**Reproduction Steps:**

1. Follow instructions as per this page: https://github.com/chan-sccp/chan-sccp/wiki/FreePBX_Intercom-HowTo
2. Dial paging extension as set in FreePBX

**chan-sccp version:** Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 81f140aM (built by 'root' on '2020-05-20 19:29:22 UTC')

**asterisk version:** 13.32.0

**Expected behavior:**

The handsets should pick up automatically with no ringing

**Observed behavior:**

The handsets ring very briefly before automatically answering

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [**Yes**/No]
- Problem can be reliably reproduced, doesn't happen randomly: [**Yes**/No]
- Platform: FreePBX 14.0.13.28 in a VMware environment
- Direct Contact information: [You can contact me here]
- Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): https://pastebin.com/hFpdjxn2

dkgroot, 2020-05-21:
@Terrabytee This must have happened after the change in the "call-answer" project a few weeks ago. We had a problem in the way calls were answered, and there might be a slight delay in the "auto-answer" system because of that. 
If you don't want the ringing to occur, you can set:
`Set(ALERT_INFO=Off);`
or use:
`Dial(SCCP/98041/ringer=off,aa=2w);`

If that does not help, we may have to investigate. If that solves the problem, we might want to update the wiki page mentioned.

Terrabytee, 2020-05-21:
Thanks for the response! Just for clarity, which file would I insert those into?

dkgroot, 2020-05-21:
I don't what solution you are running, if you are running the simple one at the top, the change would go into: /etc/asterisk/extensions_custom.conf`

For the multicast solution i would have to have a look

Terrabytee, 2020-05-22:
Running FreePBX 14 right now, but I do have the ability to access and make changes to files in /etc/asterisk and execute them.

I do not use multicast paging at the moment, just the macro 2-way option (2b in the wiki page), for intercom communication as well.

I added all three of those options into my extensions_custom.conf file and restarted Asterisk, but those didn't work. Just to make sure I am doing this right, I will include the contents of my extensions_custom.conf file, again just to make sure I am doing this correctly.

`[macro-2way-paging]`

`exten => s,1(check),ChanIsAvail(SCCP/${ARG1},sj)`
`exten => s,n,Dial(SCCP/${ARG1}/aa=2wb,5,A(none))`
`exten => s,n(end),Busy(20)`
`exten => s,n,Macro(hangupcall,)`
`exten => s,check+101,Busy(20)`
`exten => s,n,Macro(hangupcall,)`
`Set(ALERT_INFO=Off);`

Thanks again!

dkgroot, 2020-05-23:
Doing it that way would never be able to work. The new dialplan you created doesn't even compile and should generate errors.

I falsely assumed that if you were editing dialplan logic, you would know what you are doing :-) sorry i was not clear enough.

You need to do something like this
```
[macro-2way-paging]
exten => s,1(check),ChanIsAvail(SCCP/${ARG1},sj)
exten => s,n,Set(ALERT_INFO=Off)
exten => s,n,Dial(SCCP/${ARG1}/aa=2wb,5,A(none))
exten => s,n(end),Busy(20)
exten => s,n,Macro(hangupcall,)
exten => s,check+101,Busy(20)
exten => s,n,Macro(hangupcall,)
```
or
```
[macro-2way-paging]
exten => s,1(check),ChanIsAvail(SCCP/${ARG1},sj)
exten => s,n,Dial(SCCP/${ARG1}/aa=2wb/ringer=silent,5,A(none))
exten => s,n(end),Busy(20)
exten => s,n,Macro(hangupcall,)
exten => s,check+101,Busy(20)
exten => s,n,Macro(hangupcall,)
```

Terrabytee, 2020-05-26:
Hey, sorry about that confusion!

Anyways, I tried the first one above and that did the trick! :) Thanks again!

closed 2020-05-26 by Terrabytee

dkgroot, 2020-05-26:
Ok thanks for checking and testing. I will update the wiki, to include the optional line.