chan-sccp/chan-sccp #522: Segmentaion Fault on Chan_SCCP

muradbrotecs, 2020-06-22:
Segmentaion Fault on Chan_SCCP

_asterisk[18001]: segfault at xxx ip xxxxxxxxxx sp xxxxxxxxxx error 4 in chan_sccp.so[7f678e700000+10a000]_

**chan-sccp version:** Release: 4.2.3 STABLE - 6728M 

**asterisk version: 13.17.2

Kernel / Linux version
3.10.0-327.10.1.el7.x86_64 #1 SMP  x86_64 x86_64 x86_64 GNU/Linux

OS CentOS Linux release 7.2.1511 (Core)

Any help will be much appreciated

dkgroot, 2020-06-23:
@muradbrotecs Hi Murad

I recognise this revision "4.2.3 STABLE - 6728M" as a pretty ancient svn based revision number. Please checkout the current develop branch from git instead and test with that.

Note: if you are running into a segfault like this, we normally need a backtrace (check out wiki: https://github.com/chan-sccp/chan-sccp/wiki/Reporting-Bugs to see how to produce these). But before doing that, checkout the latest `develop` branch from git first and see if that fixes the issue.

Regard,

Diederik

dkgroot, 2020-07-07:
@muradbrotecs ?

dkgroot, 2020-08-29:
@muradbrotecs Any progress ?

closed 2021-01-01 by dkgroot