chan-sccp/chan-sccp #523: Implement call auto select fix

abradfordcms, 2020-06-24:
If a phone does not have the following within it's .cnf.xml file, call transfer attempts wind up facing issue with the "transfer" softkey disappearing

<vendorConfig>
<autoSelectLineEnable>0</autoSelectLineEnable>
<autoCallSelect>0</autoCallSelect>
...
</vendorConfig>

**Reproduction Steps:**

1. Have an existing caller (internal or external) on an active call
2. Attempt transfer to internal or external number
3. Observe that the call that is "selected" on the screen as the transfer "to" target changes from the party that has been dialed out to that you intend to transfer to. Call selector moves to the pre-existing call leg, which changes the softkey state. 

**chan-sccp version:** All versions

**asterisk version:** All Asterisk versions - observed on both Asterisk 11 and 16 

**Expected behavior:** Expected that the call selector stays on the call leg that has been nailed up after the initial transfer softkey is pressed to begin the "consult" with the second party.

dkgroot, 2020-06-24:
We should set/add two variables to the calls involved in the transfer and use them as marker. We can point them to the linkedid of the call destination

In the first step of transfer process we set the transferee and transferer on the first call. 
TRANSFER=SRC
TRANSFER_LINK=linkedid
TRANSFERER=SCCP/1234

Then put it on hold and create the second channel
The second channel will be marked with
TRANSFER=DST
TRANSFER_LINK=linkedid or the source call (fransferee)
TRANSFERER=SCCP/1234

During transfer completion, we match know exactly which parts are involved and how. When transfer completion is accidently pressed on the source we have to search a little for the destination, but should not be to difficult to implement.

dkgroot, 2020-06-24:
Without this new implementation we still require the user to set
```xml
<vendorConfig>
<autoSelectLineEnable>0</autoSelectLineEnable>
<autoCallSelect>0</autoCallSelect>
...
</vendorConfig>
```
in their cnf.xml