chan-sccp/chan-sccp #524: Rollover on Line Buttons

danielw0581, 2020-06-29:
Running FreePBX 15 with the latest Chan-SCCP module and SCCP Manager which works very well. However, is it possible to have 2 line buttons for the default extension of a phone on that same phone and have the second call ring the second button? So have ext 101 on buttons 1 and 2 and if the first button is in use have the second call ring on button 2?

Also, is there a way to have the SCCP Manager module signed so it doesn't say "you have unsigned modules" on the dashboard.

You guys have done great work on this!

Thanks for all of your hard work!

dkgroot, 2020-06-29:
Hi @danielw0581

> Running FreePBX 15 with the latest Chan-SCCP module and SCCP Manager which works very well. However, is it possible to have 2 line buttons for the default extension of a phone on that same phone and [...]

> Also, is there a way to have the SCCP Manager module signed so it doesn't say "you have unsigned modules" on the dashboard.

danielw0581, 2020-07-10:
Hi @dkgroot ,

Thanks for your quick reply!

Its been a long time since I have used a Cisco system. IIRC the lines would act as call appearances. Put two of the same line key in the phone and you can make/receive two calls. If the first line key was in use the second call would ring on the second key. If line/call appearance 1 was on hold and the phone was on hook, when you picked up the handset it would auto select line/call appearance 2. I don't believe you could transfer calls between them but you could conference calls between them. If there were no call appearance buttons available then an outgoing call could not be made and incoming calls to that line would be sent to voicemail or ring busy.

I appreciate you looking into whether this would be feasible.

And again, thank you for all you have done with this project!

Sincerely,
Daniel

dkgroot, 2020-07-10:
@danielw0581 

Sounds very practical on devices with no or a very small display. Less usefull on the phones with bigger (or even smaller) displays.

The rollover line button was never really a consideration in the current implementation and the amount of work to retrofit it in, looks to be a somewhat major overhaul. Also not sure how this would fit in with our current shared line implementation (and how to assign incoming calls in that scenario).

I am not saying it's impossible, but would require considerable effort with only little payoff (usefulness on display limited devices).