chan-sccp/chan-sccp #526: Random end call

bknv, 2020-08-05:
Time to time

, i have problem with phone cisco (7911,7912,7940,7945 and more). Call end random

Find the cause or what is causing the problem - I could not

About my servers:
1)
Virtual Machine on VMware ESXi, 6.5.0, 10884925
Version 6.5
FreePBX 14.0.13.23
Asterisk 13.27.1 built by mockbuild @ jenkins7 on a x86_64 running Linux on 2019-09-06 21:11:59 UTC
Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 81f140a (built by 'root' on '2020-05-27 18:18:45 UTC')

2)
Virtual Machine on VMware ESXi, 6.5.0, 10884925
Version 6.0
FreePBX 14.0.13.28
Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 81f140a (built by 'root' on '2020-05-27 18:10:51 UTC')
Asterisk 13.32.0 built by mockbuild @ jenkins7 on a x86_64 running Linux on 2020-03-20 19:19:15 UTC

I want show you example with problem call and corrected. In attach i send next files:  _full.txt
[full.txt](https://github.com/chan-sccp/chan-sccp/files/5029147/full.txt)
_ - log file and _dump.zip_
[dump.zip](https://github.com/chan-sccp/chan-sccp/files/5029168/dump.zip)
 - tcpdump 

Problem call from 2841 to 2840:
Start call - 15:48:41
Answer - 15:48:44
Break(started problem) - 15:48:57
Hangup - 15:49:00

Corrected call from 2841 to 2840:
Start call - 15:49:11
Answer - 15:49:14
Hangup - 15:49:44

P.S. I hope I was able to explain correctly

dkgroot, 2020-08-05:
@bknv
Thanks for reporting the issue. I understand the issue reported.

When i look at the full.txt log, i see INVALIDNUMBER in line 1014. Which seems to be the reason for the failed call. We need to figure out where this is coming from.

bknv, 2020-08-12:
Tommorow i updated second server to version 8f9abc2 and will watch for problem

dkgroot, 2020-08-29:
@bknv Have you had any chance to collect some debug log for this issue ?

bknv, 2020-08-31:
At the moment, I have not yet managed to catch this error after updating
Users also did not handle a similar problem
I will definitely report the results

dkgroot, 2020-08-31:
We should close the issue report 1-2 weeks from now, if the issue does not occur anymore.

bknv, 2020-08-31:
I agree with you

bknv, 2020-09-04:
Today user reported me about our problem.
[log.txt](https://github.com/chan-sccp/chan-sccp/files/5173787/log.txt)
Start call 10:42:29
Answer 10:42:32
Break(started problem) 10:42:48

dkgroot, 2020-09-04:
Am i right to be checking/tracing this device: SEPA4563041D7FC and call:SCCP/2812-000023F2 just to make sure before i start looking into it.

dkgroot, 2020-09-04:
Looking at this part of the log:
```
[2020-09-04 10:42:33] NOTICE[12370][C-0000012a] sccp_session.c:        > SEPA4563041D7FC: set poll timeout 54 for session 65
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_actions.c:     -- SEPA4563041D7FC: >> Got message Accessory Status Message (0x49)
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_actions.c: SEPA4563041D7FC: SCCP Handle Message: Accessory Status Message(0x0049) 20 bytes length
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_session.c: SEPA4563041D7FC: Sending Message: Clear Notify Message(0x0115) 4 bytes length
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: Clear the display notify message
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: >> Send message Clear Prompt Status Message
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_session.c: SEPA4563041D7FC: Sending Message: Clear Prompt Status Message(0x0113) 12 bytes length
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: Clear the status prompt on line 0 and callid 0
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: ( sccp_device.c:2133:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '' for line 0 (0)
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: >> Send message Display Dynamic Prompt Status Message
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_session.c: SEPA4563041D7FC: Sending Message: Display Dynamic Prompt Status Message(0x0145) 20 bytes length
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_protocol.c:     -- SEPA4563041D7FC: Display prompt on line 0, callid 0, timeout 0
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_softkeys.c:     -- SEPA4563041D7FC: softkey 'Redial' on ONHOOK to on
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_softkeys.c:        > SEPA4563041D7FC: found softkey 'Redial' at 1
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_softkeys.c:     -- SEPA4563041D7FC: softkey 'Confrn' on ONHOOK to on
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_softkeys.c:     -- SEPA4563041D7FC: softkey 'Join' on ONHOOK to off
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_softkeys.c:     -- SEPA4563041D7FC: softkey 'ConfList' on ONHOOK to on
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_softkeys.c:     -- SEPA4563041D7FC: softkey 'Record' on ONHOOK to off
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: Set softkeyset to ONHOOK(0) on line 0  and call 0
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: validKeyMask 4294967295
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: >> Send message Select SoftKeys Message
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_session.c: SEPA4563041D7FC: Sending Message: Select SoftKeys Message(0x0110) 20 bytes length
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: Finish DisplayPrompt
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_device.c:     -- SEPA4563041D7FC: Accessory 'Handset' is 'On Hook'
```
We can see that the device/phone is sending us :
```
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_actions.c:     -- SEPA4563041D7FC: >> Got message Accessory Status Message (0x49)
[2020-09-04 10:42:48] NOTICE[12370][C-0000012a] sccp_actions.c: SEPA4563041D7FC: SCCP Handle Message: Accessory Status Message(0x0049) 20 bytes length
```
Which translates to `Accessory 'Handset' is 'On Hook'`. And it is coming from the phone the user is using to dial out. The question is did the call break before 10:42:48 (potentialially caused by RTP/audio issue) and did he/she hangup to cancel the connection. Or was the `Accessory 'Handset' is 'On Hook'` message received from the phone an error (problem in the phone firmware ?).

You would have to check ask the user, to make sure if they hungup the call, or it was spontaneously hungup for them. If it was the audio that cut out and everything was still working (except for them hearing anything) we have very different issue to research.

bknv, 2020-09-04:
10:42:32 both user joined channel and started RTP. Talk among themselves
after 00:00:12:341 disappear outbound RTP from SCCP/2812
further user SCCP/2817 does not hear abonent SCCP/2812 but there is an outgoing RTP from SCCP/2817

to say for sure whether user 2812 put down the phone at this time (10:42:48) or not - I cannot

sorry if i do something wrong

[image]

dkgroot, 2020-09-04:
@bknv Is there a NAT firewall of some kind between the asterisk server and one/both of these phones, that might explain the loss to rtp traffic ?

bknv, 2020-09-07:
NAT - no, these both users locate in one network.
Firewall - yes, but i think if the problem would be with it, both users did not hear each other initially

dkgroot, 2020-09-07:
So there is a firewall between asterisk and/or both phones ? If so the firewall may simply be timing out, causing the interruption in RTP flow. From the information provided it does not look like chan-sccp is actively doing anything to interrupt the call (if it did, it would have logged something, it's pretty chatty as you may have noticed). I think it might be time to look at other potential reasons. As the problem is (seems to be) difficult to reproduce i am not sure how much help i can offer. If it were easier to reproduce i would start with a packet capture.
I am not sure where the nice graph above came from, if that is from an active packet capture, there might be some information in there.

bknv, 2020-09-07:
I can offer the next version:
I can switch off all iptables on server, enable tcpdump and send it when the problem will appear.
What you think about it?

dkgroot, 2020-09-07:
If your server is outward facing (ie: exposed to the internet), then removing all firewallrules might not be such a good idea. If not, and it's safe to do so, then i guess why not.

bknv, 2020-09-24:
Hello @dkgroot . 
I disable iptables and problem repeat. Hears what i found:

In 09:44:34 between user 2867(sccp) and 3128(sip) created channel
[2020-09-24 09:44:34] VERBOSE[5107][C-000001b3] bridge_channel.c: Channel SIP/3128-000042d6 joined 'simple_bridge' basic-bridge <ac901a40-ce7e-4bb3-8c83-31740d1a1afc>
[2020-09-24 09:44:34] VERBOSE[5032][C-000001b3] bridge_channel.c: Channel SCCP/2867-00004516 joined 'simple_bridge' basic-bridge <ac901a40-ce7e-4bb3-8c83-31740d1a1afc>

and both user talk between themselves

in 09:44:36 i see state:INVVALIDNUMBER and next 2867 SCCP device stop all media
[2020-09-24 09:44:36] NOTICE[2014][C-0000003f] sccp_indicate.c:     -- SEPF07F06B8DA87: Indicate SCCP new state:INVALIDNUMBER, current channel state:CONNECTED on call:SCCP/2867-00004516, lineInstance:1 (previous channelstate:RINGOUT)
[2020-09-24 09:44:36] NOTICE[2014][C-0000003f] sccp_channel.c:   == SCCP/2867-00004516: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on device SEPF07F06B8DA87

and 09:44:37 channel left 2867
[2020-09-24 09:44:37] VERBOSE[5032][C-000001b3] bridge_channel.c: Channel SCCP/2867-00004516 left 'simple_bridge' basic-bridge <ac901a40-ce7e-4bb3-8c83-31740d1a1afc>

I think that the closure of the rtp happened for a reason state:INVVALIDNUMBER. What you think about it?
[log.txt](https://github.com/chan-sccp/chan-sccp/files/5274771/log.txt)

dkgroot, 2020-09-26:
@bknv I think there might be some issue with one of the schedulers not being cleared, producing the INVALIDNUMBER indication on timeout.

bknv, 2020-09-29:
Yesterday i switched to issues/526
I will write when the problem arises

bknv, 2020-10-02:
Sending log

[log.txt](https://github.com/chan-sccp/chan-sccp/files/5318293/log.txt)

dkgroot, 2020-10-03:
@bknv Thanks for the log. I have created a patch that should fix the issue in the current "issues/526" branch. Can you pull it and see if that solves the issue.

bknv, 2020-10-05:
Okey. Today i updating

dkgroot, 2020-10-17:
@bknv Any new results ? Does the new branch fix the issue ?

bknv, 2020-10-19:
Hello, after update (14 day last) we are not see problem. 
In the end of week i suggest to close issue if we will not get problem.

dkgroot, 2020-10-19:
@bknv In that case we should merge the changes. When changes are merged, the issue is automatically closed by the merge. I wlll wait until the end of the week, but it looks like we are pretty confident that the two small changes took care of the issue.

bknv, 2020-10-23:
It's all right. I don't see problem any more

dkgroot, 2020-10-23:
ok mergnig the change ! Thanks for reporting and testing !

closed 2020-11-12 by dkgroot