chan-sccp/chan-sccp #527: Can't answer incoming call from SIP peer

CRCinAU, 2020-08-11:
When my folks call me from their Cisco SIP phone connected to the same asterisk server, my SCCP phone rings, but when trying to answer, does not answer and continues ringing.

```
voip*CLI> core show version 
Asterisk 16.12.0 built by root @ voip on a x86_64 running Linux on 2020-07-16 17:25:12 UTC
voip*CLI> sccp show version 
Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 47557d5 (built by 'root' on '2020-08-02 19:57:22 UTC')
voip*CLI> 
```

Attached is the asterisk log with requested debug levels enabled.

There are two SCCP phones in a shared line config, IDs replaced with SEPxxxxxxxxxxxx and SEPyyyyyyyyyyyy.

[messages.txt](https://github.com/chan-sccp/chan-sccp/files/5055284/messages.txt)

dkgroot, 2020-08-12:
Hi Steven,
The first thing i notice in the debug output, is that we are talking about huntgroup, not a simple call.
```
Dial("SIP/105-0000000d", "SCCP/101&SIP/102&SIP/103,30,c")
```
Also we seem to be dealing with a shared line as well.

Line 91: `[2020-08-11 06:56:03] ERROR[24215][C-00000003] sccp_channel.c: SCCP/101-0000000F: (sccp_channel_openReceiveChannel) Channel already hanging up` Seems to be the start of the issues.

***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/527 origin/issues/527
git checkout issues/527
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

Please show me the new debug out with the same settings and preferably a similar/same call.

CRCinAU, 2020-08-13:
Added log with the same endpoints.

[messages-issues-527.txt](https://github.com/chan-sccp/chan-sccp/files/5067378/messages-issues-527.txt)

dkgroot, 2020-08-13:
As suspected it was the answered_elsewhere check preventing the rtp setup. Pushed an update, please pull latest and retest

CRCinAU, 2020-08-13:
Have done so - will report back tomorrow when I can get my folks to ring that phone...

CRCinAU, 2020-08-14:
Can confirm - this seems to answer calls fine now in that scenario.

dkgroot, 2020-08-14:
Have not merged the change yet, need to check if there would be a regression when/if merged.

dkgroot, 2020-08-15:
Thanks for reporting the issue.
Branch has been merged

commit 2a3eaf3f (2020-08-15): Fixed indentation sccp_indicate
Changed sccp_channel_answer to set callstate directly instead of calling sccp_indicate with OFFHOOK state.
Remove answered_elsewhere check from rtp setup

Fixes: #527
closed 2020-08-15 by dkgroot