chan-sccp/chan-sccp #528: Parking doesn't play announcement and end the call

drtmk, 2020-08-20:
When parking a call by pressing soft key button, the parked extension number is not announced through audio, and the call doesn't end on the phone on which soft key button was pressed. [**Demo**](https://drive.google.com/file/d/1L7nHGessS-HhtqUi1Bbnodx0aCDlYTlB/view?usp=sharing)

chan-sccp version: 4.3.2 develop 8f9abc2
asterisk version: 16.12.0 x86_64
phone model: 79xx

Expected behavior:
After pressing [Park] the phone should play the extension number of parking space and end the call.

dkgroot, 2020-08-27:
@drtmk I have been trying to get the announcement to work (again) but so far no success.. The changes in asterisk seem to make this pretty hard with the current code. I can make a simple change to at least perform the hangup and we rely only on the display of the parking location. Maybe that would be a good first step. I would like to figure out a way to get the announcement of the parkingslot again though.
Would adding a hangup already help ?

drtmk, 2020-08-27:
I think hangup with text message is enough. Audio announcement is good for old analog phones.

dkgroot, 2020-08-28:
Ok going with that. Having a look at the code to make that happen.Thanks for the feedback

dkgroot, 2020-08-29:
@drtmk I managed to get parking + announcements to work (I think)
Can you help me out to check if this is working as one would expect. I have not been able to get to the phones to test parking on the physical phone. I have done all the testing from the CLI using `sccp set channel SCCP/1234-000001 park` (new CLI command to enable parking).

***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/528 origin/issues/528
git checkout issues/528
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

drtmk, 2020-08-31:
I'm getting error during compilation:

```
ast116.c: In function ‘parking_event_cb’:
ast116.c:1489:124: error: ‘struct ast_channel_snapshot’ has no member named ‘base’; did you mean ‘name’?
    pbx_log(LOG_NOTICE, "%s Parked call:%s to parkingspace:%d@%s\n", ast_channel_name(parker_chan), parked_payload->parkee->base->uniqueid, parked_payload->parkingspace, parked_payload->parkinglot);

```

dkgroot, 2020-08-31:
Pushed a fix for the compile error.

drtmk, 2020-08-31:
Audio announcement works and the call is being hang up. I can continue it by dialing parked extension (from any phone). This simple scenario works fine. But there are other issues.

Parking extension is now random number from the pool from res_parking.conf. Before it was first free extension (I think).
Asterisk always shows this error when pressing [Park], even if it's the first Parking since asterisk restart.

```
    -- Parking 'SCCP/100-00000001' in 'default' at space 720
    -- Channel SCCP/100-00000001 joined 'holding_bridge' parking-bridge <61585ffd-beef-4f22-8a0e-f52076cf59de>
       > Bridge 58296c2b-f920-4f99-8076-34374892cd7f: switching from native_rtp technology to simple_bridge
[Aug 31 13:56:55] ERROR[981]: ast116.c:1522 parking_event_cb: SCCP/120-00000002:Park event:1935832403 not handled
[Aug 31 13:56:55] NOTICE[981]: ast116.c:1489 parking_event_cb: SCCP/120-00000002 Parked call to parkingspace:720@default
    -- Started music on hold, class 'default', on channel 'SCCP/100-00000001'
```
 
When the call is parked twice from 2 phones, one after another, both phones show parking number as a message above soft keys. The message should be displayed only on the phone that made the parking.
When I make a call from A to B (pressing Park on B), I continue the call from phone C (by dialing parked extension). Then I park this call again on phone C, and both phones show the message about parking space - phone B and C, although B has nothing to do with the call anymore. And the message on phone B stays there forewer.

And one more. When I press Park on the phone that was already Parked (I have that soft key available) and is now playing music on hold, Asterisk shuts down.

```
    -- Parking 'SCCP/100-0000000B' in 'default' at space 703
[Aug 31 14:07:05] ERROR[999]: ast116.c:1522 parking_event_cb: SCCP/120-0000000C:Park event:1935832403 not handled
[Aug 31 14:07:05] NOTICE[999]: ast116.c:1489 parking_event_cb: SCCP/120-0000000C Parked call to parkingspace:703@default
    -- Channel SCCP/100-0000000B joined 'holding_bridge' parking-bridge <61585ffd-beef-4f22-8a0e-f52076cf59de>
       > Bridge 1f44181b-57f5-4d76-8879-f1cd6481c3d4: switching from native_rtp technology to simple_bridge
    -- <SCCP/120-0000000C> Playing 'digits/7.g722' (language 'en')
    -- Started music on hold, class 'default', on channel 'SCCP/100-0000000B'
    -- <SCCP/120-0000000C> Playing 'digits/0.g722' (language 'en')
    -- <SCCP/120-0000000C> Playing 'digits/3.g722' (language 'en')
    -- Channel SCCP/120-0000000C left 'simple_bridge' basic-bridge <1f44181b-57f5-4d76-8879-f1cd6481c3d4>
    -- QualityStats: MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;ICR=0.0000;CCR=0.0017;ICRmx=0.0035;CS=2;SCS=0;MLQKvr=0.95
    -- SEPD0C282425243: Call Statistics:
       [
         Last Call        : CallID: 12 Packets sent: 411 rcvd: 403 lost: 0 jitter: 21 latency: 0
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0000;CCR=0.0017;ICRmx=0.0035|CS=2;SCS=0
         Mean Statistics  : #Calls: 6 Packets sent: 348 rcvd: 342 lost: 0 jitter: 11 latency: 0
         Mean Quality     : MLQK=0.7406;MLQKav=0.7406;MLQKmn=0.7406;MLQKmx=4.4438;MLQKvr=0.95|ICR=0.0006;CCR=0.0013;ICRmx=0.0035|CS=0;SCS=0
       ]
asterisk*CLI>
Disconnected from Asterisk server
Asterisk cleanly ending (0).
Executing last minute cleanups
Asterisk ending (0).
root@asterisk:/etc/asterisk#
```

Dynamic speed dials are not properly updated when parking and resuming a call.

dkgroot, 2020-08-31:
> Audio announcement works and the call is being hang up. I can continue it by dialing parked extension (from any phone). This simple scenario works fine. But there are other issues.

Thanks for checking / testing !!

> Parking extension is now random number from the pool from res_parking.conf. Before it was first free extension (I think).
Asterisk always shows this error when pressing [Park], even if it's the first Parking since asterisk restart.

The parking order being random can be controlled with the "findslot" entry in features.conf.
The error you mentioned should be fixed/removed (Current park code is a little extra verbose).

> When the call is parked twice from 2 phones, one after another, both phones show parking number as a message above soft keys. The message should be displayed only on the phone that made the parking.

I will check this out tomorrow. Should not be to hard to correct (i think).

> Dynamic speed dials are not properly updated when parking and resuming a call.

I have not paid any attention to the dynamic speeddials in regard to parking. I will have a look at how much work that would.

The thing you overlooked, is that parked calls now return to the parker after timeout, which is something that didn't use to work (most of the time).

All in all the results are not bad. Thanks a lot for being rigorous in you testing / checking.

drtmk, 2020-08-31:
> The thing you overlooked, is that parked calls now return to the parker after timeout, which is something that didn't use to work (most of the time).

I did test that. The call was returning to the phone that made the parking.

dkgroot, 2020-08-31:
> I did test that. The call was returning to the phone that made the parking.

This was not the case in the past (i think), so we made some progress :-)

Also do you like the random slot assignement, or would you like "first" or "next" instead. I think i might have hardcoded the "random" case at the moment.

dkgroot, 2020-09-01:
Fix: Use regular (non random) parking slot selection
Fix: Remove error report about event not handled in parking_event_cb
Fix: Do not attempt to park a channel that only has one bridge channel connected to a dialplan application (ie: parking a parked channel) -> return failure to user
Fix: Only show the parking slot message on the phone once (will require a little more work)

drtmk, 2020-09-01:
> Fix: Use regular (non random) parking slot selection

Works as expected. First free extension is selected from the pool.

> Fix: Remove error report about event not handled in parking_event_cb

Fixed. There are no errors/warnings during parking.

> Fix: Do not attempt to park a channel that only has one bridge channel connected to a dialplan application (ie: parking a parked channel) -> return failure to user

Now you cannot park already parked call. Soft key button is available but pressing it causes failure in parking and asterisk shows warning. Music continues to play. I guess that is fine.
```
[Sep  1 08:27:32] WARNING[733][C-00000001]: ast116.c:1586 sccp_astwrap_park: SCCP/100-0000002C: Remote channel is missing, giving up (connected to application?)
[Sep  1 08:27:36] WARNING[733][C-00000001]: ast116.c:1586 sccp_astwrap_park: SCCP/100-0000002C: Remote channel is missing, giving up (connected to application?)
[Sep  1 08:27:37] WARNING[733][C-00000001]: ast116.c:1586 sccp_astwrap_park: SCCP/100-0000002C: Remote channel is missing, giving up (connected to application?
```
> Fix: Only show the parking slot message on the phone once (will require a little more work)

Fixed.

I also tested two simultaneous parked calls. They were working fine. After given time calls were returning to the parker.

So except speed dials, everything works. Good job.

dkgroot, 2020-09-01:
Thanks for checking everything.

I will still have to work on connected_line / callerid information displayed and the dynamic speeddial information connected to that. I am not sure if we should attempt to display the parkingslot to the parked caller (we don't want to tell incoming external callers to much about our dialplan). Also the dynamic speeddial is not that important, most of time you will not be parking people from inside the company, normally parking is used for incoming external calls (for which you would not have a dynamic speeddial).

dkgroot, 2020-09-01:
Added a called party name update to show park:[parkinglot] on the party being parked, and forcing an update off the indication including remote ones (aka: dynamic speeddials)
Have not had time to fixup the callerid update when the parked call is resumed yet (Not sure how/when/where to do that).

drtmk, 2020-09-02:
Right now, when you press [park] on a phone that answered a call the calling party gets "To: park:default" on his call plane. Also speed dial of that line shows: "park:default <=> Name". And it stays there when call returns.
[Video](https://drive.google.com/file/d/164ZY6NKL-jr63cr5NP8WHP01Z2T2FTDI/view?usp=sharing)

When you press [park] on a phone that initiated a call, parked phone doesn't change a call plane and speed dial is not updated.
[Video](https://drive.google.com/file/d/1PtBLewNwBAelPEnLYdCXN27m7QXgjLPs/view?usp=sharing)

dkgroot, 2020-09-02:
> Right now, when you press [park] on a phone that answered a call the calling party gets "To: park:default" on his call plane. Also speed dial of that line shows: "park:default <=> Name". And it stays there when call returns.

True and i am not happy about the last part. Also i should have been using the Park Label (Parkuj) instead of "park:"

> When you press [park] on a phone that initiated a call, parked phone doesn't change a call plane and speed dial is not updated.

That should be easy to fix (I hope).

To be honest i am not sure we should be going down this road though. Asterisk has nothing in place in the call-parking code to update caller id's etc. So there is no expectancy for any of this to be happening. I am not 100% what would be the best way to solve this, And maybe we should just not do any callerid update. The chance of anyone parking a local channel (with a dynamic speeddial pointing at it) are very very slim, and would most likely be a mistake. I will have another look to see what can be done savely, but maybe we should just forget about updating the callerid information for parked call.s alltogether.

drtmk, 2020-09-02:
I think the caller shouldn't see his call was parked. It should look like he's on hold.
The speed dial that shows his line on other phones, could just show that he's busy (red light + line name without party).
So before parking speed dial could display: [Bob <=> Alice], and after parking just [Alice]. That would mean that Alice line is busy but not connected with anyone.

dkgroot, 2020-09-02:
Sounds really easy i know, but it really isn't. I am looking for a way, but haven't found it yet

dkgroot, 2020-09-02:
Pushed some new code to handle callparking and dynamic speeddials a little nicer.

drtmk, 2020-09-03:
Right now when you park a call, the phone that was parked does not update a call plane and looks like his on hold. (i think that's ok). When you dial parked extension, call planes on both phones show correct names. The one that was parked updates his call plane with correct new name.

After parking a call, dynamic speed dial shows 'Parkuj <=> Name', which is really nice. When a call returns and is in calling state, the speed dial doesn't blink orange but stays red. I think it should blink because the line is making a call (return call).

And I found one case when speed dial doesn't update party name:
Phone A calls C, C anwers the call and parks it. Phone B that has A speed dial shows 'Parkuj <=> A'.
When a call returns to C, speed dial is not updated. It doesn't update party name during calling state nor after answering a call. Still shows 'Parkuj <=> A'. But it does update when you call a parked extension.

When you change the initial direction of the call (C -> A, A answers, C parks) speed dial works fine.

dkgroot, 2020-09-03:
> Right now when you park a call, the phone that was parked does not update a call plane and looks like his on hold. (i think that's ok). When you dial parked extension, call planes on both phones show correct names. The one that was parked updates his call plane with correct new name.

Ok nice !

> After parking a call, dynamic speed dial shows 'Parkuj <=> Name', which is really nice. When a call returns and is in calling state, the speed dial doesn't blink orange but stays red. I think it should blink because the line is making a call (return call).

It's not the phone making it a call, it's the asterisk parking application, calling the other phone. The parked call has no idea what is happening (on it's behalf).

> Phone A calls C, C anwers the call and parks it. 

I will have another look at this one, a little later on.

We are getting closer though :-)

dkgroot, 2020-09-09:
50d9b14 should take care of updateing the remote hinted speeddial monitoring a parked call, upon auto-return.

drtmk, 2020-09-09:
I don't have time to check this today.

drtmk, 2020-09-10:
Parking and dialing parked extension works ok.

Problems starts to happen when nobody calls parked extension and call returns.

Alice calls Bob, Bob answers a call, Alice parks it. So far everything is ok: speed dials, call plates.
When call returns, Bob is calling Alice. Alice's phone has incoming connection from Bob (that's ok). Speed dials with Alice's extension also works ok.
The problem is that Bob's phone looks like he already established a call - the call plate shows: `From Alice` (because Alice made that call in the first place). It would be better to update it to`To Alice` because Bob is now calling back to Alice. And Bob's phone shows time duration of the call, and it's increasing.
The speed dials with Bob's extension configured also show that Bob is already connected with Alice (`Alice <=> Bob` and red lamp).
It should display: `Alice <- Bob` with blinking orange.
So as I sad, Bob's phone behaves like he's already connected with Alice, but he plays outgoing call sound (beep,3  second slice, beep).

When you change the direction of parking there's a different problem.

Alice calls Bob, Bob answers and parks it. Call plates, speed dials are ok. After a while Alice calls back.
Alice's phone looks like he's making a new call. Call plates shows `To Bob`, at the botton there's a message `Establishing a call`. It plays outgoing call sound. Speed dials with Alice's extension also are showing outgoing connection (`Bob <- Alice` with orange light). This is actually how Bob's phone should behave in the first scenario.
When I answer the call on Bob's phone, Alice's phone is still in ringing state (and also Alice's speed dials are still blinking), and plays outgoing call sound, but what's suprising audio between 2 phones flows in both directions.
Bob's phone and Bob's speed dials are ok, they show the call is established.

dkgroot, 2020-09-10:
We will have to wait a little bit for #533 to finish, before going back to this one, The bigger/major change for #533, could cause some sideeffects for parking and dynamic speeddials.

I don't understand why the first one is not working for you (maybe because i was running in experimental mode), i checked it several times. But i will come back to this when #533 is closed.

dkgroot, 2020-09-10:
Rebased on top of last develop branch (to include the changes made for #533)
Resting cases.

dkgroot, 2020-09-10:
Fix: continued ringing during ParkedCallTimeout calling back
Fix: change the call direction when handling ParkedCallTimeout
Fix: Update indication and dynamic speeddial when handling ParkedCallTimeout

Hopefully we are making a little progress here

drtmk, 2020-09-11:
The first case is almost perfect (Alice calls Bob, Bob answers a call, Alice parks it.)
One thing is left to fix: When Bob makes a return call, his phone and speed dials with his extension show that he's making a call to himself. (`To Bob` and `Bob <- Bob`). He's actually makes a call to Alice. Alice's phone behaves ok.

The second case (Alice calls Bob, Bob answers and parks it.) is almost 100% fixed.
I think there's something wrong with Alice's phone when it plays outgoing call sound. It sounds to me like it's generated twice. I'm not sure.

dkgroot, 2020-09-11:
I am a little afraid that i might be making to many changes, to fix this.

drtmk, 2020-09-11:
> 1. What do you mean "makes a return call" . Do mean when he times out and start calling alice back ?

Yes.

dkgroot, 2020-09-11:
Confirmed case 1. Checking what's going on.

dkgroot, 2020-09-11:
Confirmed both
Fixed both
Minor patch required.

drtmk, 2020-09-11:
First one is fixed.
The second one is hard to tell. It sounds to me like there are still 2 outgoing sounds. But they're aligned so I can only distinguish it because beep tone is a little louder and it starts and ends more harder? But it's no big deal.

As for issue #533 branch 4.3.2 issues/528 - a78dea3 resolves the problem with lamp going from orange to red. The second problem (Invalid speed dial direction after answering Group Pickup) still persists but I guess it's not that branch that supposed to resolve it.

dkgroot, 2020-09-11:
I stopped producing a second ringout tone, when we already have inband audio. So the double ringout tones should not be happening. I will add a little debug, to check and make sure this is not the case. Others will start reporting the issue pretty fast if we would be producing double ring sounds, as it is more pronounced on some phones/system then others.

Regarding #533 the changes made to the hint system earlier would/could have some side effect. #533 was already rolled into develop and the #528 was rebased on top of those changes so it is all accumulated. The "RED" problem was the bigger one for you (if i remember correctly), so i am happy that one is also resolved.

dkgroot, 2020-09-15:
@drtmk Have you had a chance to check the double ring again ? I think it should be fixed as well. I would like to close this issue and merge it into develop.

drtmk, 2020-09-16:
I'm not sure if it's one or two calling tones. But if you fixed it we can assume it's one.

closed 2020-09-16 by dkgroot

dkgroot, 2020-09-16:
@drtmk Thank you very much for your patience and testing work !

commit eb9d8c83 (2020-09-16): Squashed commit of the following:

Enh: Added a cli command to park a call: `sccp set channel SCCP/xxxx-xxxxx park`
Fix: Use regular (non random) parking slot selection
Fix: Remove error report about event not handled in parking_event_cb
Fix: Do not attempt to park a channel that only has one bridge channel connected to a dialplan application (ie: parking a parked channel) -> return failure to user
Fix: Update called party name to "park:xxxxx" on the channel being parked
Fix: update remote hinted speeddial when parked call is auto-returned
Fix: continued ringing during ParkedCallTimeout callin [... cut]