chan-sccp/chan-sccp #529: Missing party information in dynamic speed dials

drtmk, 2020-08-27:
Sometimes dynamic speed dials don't show party information during calling or connected state. This is a random problem and was observed on 7942 phones.

In all 3 of the following cases, Alice (100) is calling Bob (120) by pressing Redial soft key button. Bob has Alice extension configured as dynamic speed dial.

The first case shows how the phone should behave. Bob answers the call and speed dial shows that Alice is connected with Bob.
[image]

Debug log: [ok.txt](https://github.com/chan-sccp/chan-sccp/files/5135316/ok.txt)

In the second case, after answering the call, the monitored line does not show who Alice called.
[image]

Debug log: [connected error.txt](https://github.com/chan-sccp/chan-sccp/files/5135388/connected.error.txt)

The 3rd case shows another problem. Sometimes during calling phase, the speed dial doesn't show where monitored line is calling.
[image]

Debug log: [calling error.txt](https://github.com/chan-sccp/chan-sccp/files/5135399/calling.error.txt)

**chan-sccp version:**  Release: 4.3.2 develop - 8f9abc2
**asterisk version:** Asterisk 16.12.0 built by root @ asterisk on a x86_64 running Linux on 2020-08-14 17:38:40 UTC

dkgroot, 2020-08-27:
@drtmk I have been trying to get the announcement to work (again) but so far no success.. The changes in asterisk seem to make this pretty hard with the current code. I can make a simple change to at least perform the hangup and we rely only on the display of the parking location. Maybe that would be a good first step. I would like to figure out a way to get the announcement of the parkingslot again though.
Would adding a hangup already help ?

dkgroot, 2020-08-27:
Thank you for the issue report and the clear description and log files. It will take me a little longer than usual to have a look and find out how to fix this, because of spine/back problems i am experiencing at the moment (typing this while lying down).

dkgroot, 2020-08-27:
Checking the first two log files, it looks like we might have a race condition:
ok.txt:
```
       > 100 (hint_updateLineState) Update Line Channel State: RINGOUT(20)
       > 100: get speeddial party: 'Bob <120>' (calledParty)
       > 100 (hint_updateLineStateForSingleChannel) partyName: Bob, partyNumber: 120
       > 100 (hint_updateLineStateForSingleChannel) Set singleLineState to CONNECTED(30)
       > SCCP: (sccp_hint_notifyLineStateUpdate) matched lineName:SCCP/100 to dialplan:SCCP/100
    -- SCCP: (sccp_hint_notifyLineStateUpdate) Notify asterisk to set state to sccp channelstate 'CONNECTED' (30) on line 'SCCP/100'
    -- SCCP: (sccp_hint_notifyLineStateUpdate) => asterisk: 'Device is ringing' (6) => 'Device is in use' (2) on line SCCP/100
    -- SCCP: (sccp_hint_notifySubscribersViaPbx) Notify asterisk to set state to sccp channelstate 'CONNECTED' (30) => asterisk: 'Device is in use' (2) on channel SCCP/100
    -- SCCP: (sccp_hint_notifyLineStateUpdate) Notified asterisk to set state to sccp channelstate 'CONNECTED' (30) => asterisk: 'Device is in use' (2) on channel SCCP/100
  == 100 (hint_devstate_cb) Got new hint event SCCP/100, state: 1 (InUse), cidname: Bob, cidnum: 120
    -- 100 (hint_notifySubscribers) notify 1 subscriber(s) of SCCP/100's state CONNECTED
       > SEPD0C282425243 (hint_notifySubscribers) notify subscriber SEPD0C282425243 of SCCP/100's state CONNECTED (30), devicetype:Cisco 7942
       > 100 (hint_notifySubscribers) notify device: SEPD0C282425243@1, displayMessage:Bob <=> Alice, state: CONNECTED ->  In-use
    -- Got new hint event SCCP/100, cidname: , cidnum:
    -- SEPD0C28243AE44: Finish to indicate channel state:CONNECTED on call:SCCP/100-00000017, lineInstance:1. New channel state:CONNECTED
    -- SCCP/100-00000017: (sccp_pbx_remote_answer) Switching to STATE_UP
    -- 100 (getLinestate) state:CONNECTED, party:120/Bob, calltype:Outbound
       > SCCP: (devicestate) sccp_hint returned state:CONNECTED for '100'
       > SCCP: (devicestate) PBX requests state for '100' - state In use
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SCCP/100-00000017: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
```
connected.error.txt:
```
       > 100 (hint_updateLineState) Update Line Channel State: RINGOUT(20)
       > 100: get speeddial party: 'Bob <120>' (calledParty)
       > 100 (hint_updateLineStateForSingleChannel) partyName: Bob, partyNumber: 120
       > 100 (hint_updateLineStateForSingleChannel) Set singleLineState to CONNECTED(30)
       > SCCP: (sccp_hint_notifyLineStateUpdate) matched lineName:SCCP/100 to dialplan:SCCP/100
    -- SCCP: (sccp_hint_notifyLineStateUpdate) Notify asterisk to set state to sccp channelstate 'CONNECTED' (30) on line 'SCCP/100'
    -- SCCP: (sccp_hint_notifyLineStateUpdate) => asterisk: 'Device is ringing' (6) => 'Device is in use' (2) on line SCCP/100
    -- SCCP: (sccp_hint_notifySubscribersViaPbx) Notify asterisk to set state to sccp channelstate 'CONNECTED' (30) => asterisk: 'Device is in use' (2) on channel SCCP/100
    -- Got new hint event SCCP/100, cidname: , cidnum:
  == 100 (hint_devstate_cb) Got new hint event SCCP/100, state: 1 (InUse), cidname: , cidnum:
    -- 100 (hint_notifySubscribers) notify 1 subscriber(s) of SCCP/100's state CONNECTED
       > SEPD0C282425243 (hint_notifySubscribers) notify subscriber SEPD0C282425243 of SCCP/100's state CONNECTED (30), devicetype:Cisco 7942
       > 100 (hint_notifySubscribers) notify device: SEPD0C282425243@1, displayMessage:Alice, state: CONNECTED ->  In-use
    -- SEPD0C28243AE44: Finish to indicate channel state:CONNECTED on call:SCCP/100-00000037, lineInstance:1. New channel state:CONNECTED
    -- SCCP/100-00000037: (sccp_pbx_remote_answer) Switching to STATE_UP
    -- 100 (getLinestate) state:CONNECTED, party:120/Bob, calltype:Outbound
       > SCCP: (devicestate) sccp_hint returned state:CONNECTED for '100'
       > SCCP: (devicestate) PBX requests state for '100' - state In use
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SCCP/100-00000037: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
 SEPD0C28243AE44: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/100-00000037 (rtp_instance:yes, reception.state:1/G.722 64k, transmission.state:0/G.722 64k)
  == SEPD0C28243AE44: (pbx_indicate) finish: send indication (res:-1)
    -- SCCP: (sccp_hint_notifyLineStateUpdate) Notified asterisk to set state to sccp channelstate 'CONNECTED' (30) => asterisk: 'Device is in use' (2) on channel SCCP/100
```

causing a difference:
```
(ok )  == 100 (hint_devstate_cb) Got new hint event SCCP/100, state: 1 (InUse), cidname: Bob, cidnum: 120
(err)  == 100 (hint_devstate_cb) Got new hint event SCCP/100, state: 1 (InUse), cidname: , cidnum:
```
I need to figure out how to prevent the race and make sure the events occur synchronously.

dkgroot, 2020-08-27:
@drtmk I assume you have `--enable-distributed-devicestate` in your `configure` statement when compiling chan-sccp. Which seems to have this side effect. Can you reconfigure (and recompile) chan-sccp to see if that stops the issue for occuring ? We still need to fix the problem, but it would be good to know, that this is the part that's causing the race.

drtmk, 2020-08-27:
I will try this tomorrow.

dkgroot, 2020-08-27:
Created a fix to prevent "enable-distrubuted-devicestate" causing empty cidnum/cidname if a race occurs. This should remedy the problem in general.

***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/529 origin/issues/529
git checkout issues/529
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

dkgroot, 2020-08-27:
BTW: Thanks again for putting in the time to report the issue and providing the debug log files, images etc.. It really helps !

drtmk, 2020-08-28:
By eliminating `--enable-distributed-devicestate` from `configuration`, I wasn't able to reproduce above behavior. Messages in dynamic speed dials are working fine. I don't know what this option actually do.

And when it comes to temporary branch you have created, I'm not sure I'm doing this right. I made a copy of `chan_sccp` directory do `chan_sccp2`. But git doesn't download any files.

```
root@asterisk:/usr/src/chan_sccp2# git fetch origin
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 10 (delta 8), reused 10 (delta 8), pack-reused 0
Unpacking objects: 100% (10/10), done.
From https://github.com/chan-sccp/chan-sccp
   8f9abc2d..2a3eaf3f  develop    -> origin/develop
 * [new branch]        issues/529 -> origin/issues/529
root@asterisk:/usr/src/chan_sccp2# git branch issues/529 origin/issues/529
Branch 'issues/529' set up to track remote branch 'issues/529' from 'origin'.
root@asterisk:/usr/src/chan_sccp2# git checkout issues/529
Switched to branch 'issues/529'
Your branch is up to date with 'origin/issues/529'.
root@asterisk:/usr/src/chan_sccp2# git pull
Already up to date.
```

dkgroot, 2020-08-28:
@drtmk
1. Ok so you confirmed that --enable-distributed-devicestate on asterisk-16 was causing the issue. Good
2. distributed-devicestate is only required when you have a cluster of asterisk machines and would like share the devicestate across multiple machines (not used by most people)
3. Looking at the git output, it looks like it worked just fine. Just run ./configure and make on that new directory and check if there are not negative side effects (there should not be any)
Thanks for checking/testing and reporting back !

drtmk, 2020-08-28:
I've tried `issues/529` branch with `--enable-distributed-devicestate` and without it. Both are running fine. No issues with missing parties in speed dials.

dkgroot, 2020-08-28:
Thanks for checking ! Nice job. I have merged the change back to the `develop` branch.

commit 76863b24 (2020-08-28): Only set distributed cidNum/cidName if the contain valid information. Preventing empty cidname/cidnum when the distributed event races ahead of the devstate change.

Fixes: #529
closed 2020-08-28 by dkgroot