chan-sccp/chan-sccp #530: incominglimit

dkgroot, 2020-09-02:
Calculation for line->incominglimit comparison is incorrect. Needs adjusting for both single and shared line. See: #476

dkgroot, 2020-09-02:
@drtmk

***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/530 origin/issues/530
git checkout issues/530
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

dkgroot, 2020-09-02:
Note: Currently it does not add calls that are on HOLD to the number of active calls, meaning that an incominglimit=1 + a single call on hold would allow another incoming call to be processed.

drtmk, 2020-09-03:
It looks like this.

When A establishes a call with B, you cannot make a call to them because you get busy tone (as it should be).
You can still call a phone that pressed [Hold] soft key. But you cannot call the other phone, which plays music on hold.
You can invoke 2 incoming calls on a single line (single phone) and answer both of them.

dkgroot, 2020-09-03:
Not sure what that last part means, to be honest

drtmk, 2020-09-03:
You can dial from two phones (two lines) into one line (one phone). It will show 2 incoming calls.
After one incoming call, you shouldn't be able to call that line.

dkgroot, 2020-09-03:
Ooh i guess i did not take into account calls that are not established yet. Will have a look, to see where to do that.
Thanks for being way better at testing then i am :-)

dkgroot, 2020-09-04:
Pushed an update

drtmk, 2020-09-04:
> You can dial from two phones (two lines) into one line (one phone). It will show 2 incoming calls.
> During one incoming call, you shouldn't be able to call that line.

This is fixed.

> Note: Currently it does not add calls that are on HOLD to the number of active calls, meaning that an incominglimit=1 + a single call on hold would allow another incoming call to be processed.

This is fixed too.

Everything looks good. I also made tests with `incominglimit=2` and that looked fine.
I haven't tested shared line.

dkgroot, 2020-09-04:
Sounds like we can merge it. I don't think incominglimit is used a lot by people using shared lines tbh.

commit 8a2e0548 (2020-09-04): Added sccp_getCallCount(ld).
Fix: check l->incominglimit against the number of ongoing calls per line per device (to account for shared lines)
Fix: callwaiting playing callwaiting tone to the active_channel instead of the new incoming one.
When calculating CallCount deal with shared and single lines independently

Fixes: #530
closed 2020-09-04 by dkgroot