chan-sccp/chan-sccp #531: Unknown Number while originate call

DL44227, 2020-09-03:
When I originate a call via callfiles (/var/spool/asterisk/outgoing) the given Caller ID name and number are not displayed on the Cisco 7970 phone. However the same callfile used to originate a call with a SIP phone causes it to display the caller ID correctly.

**Reproduction Steps:**

1. Create a call file, e.g. /tmp/callfile:

Channel: SCCP/290
Callerid: "CIDName" <206>
MaxRetries: 0
WaitTime: 30
Context: secure
Extension: 206
Priority: 1

2. Move the call file to /var/spool/asterisk/outgoing. The Cisco phone 290 is signalling an incoming call, displaying "Von Unbekannte Nummer" (from unknown number).

3. Create a second call file, e.g. /tmp/callfile2:

Channel: SIP/290
Callerid: "CIDName" <206>
MaxRetries: 0
WaitTime: 30
Context: secure
Extension: 206
Priority: 1

4. Move the second call file to /var/spool/asterisk/outgoing. The SIP phone 290 is signaling an incoming call, displaying "CIDName 206" (as expected).

**chan-sccp version:** (include revision and/or branch if relevant)

Skinny Client Control Protocol (SCCP). Release: 4.3.2 TRUNK - unknown (built by 'root' on '2020-09-01 15:06:48 UTC')

(develop Branch downloaded and compiled on 2020-09-01)

**asterisk version:** (include revision and/or branch if relevant)

Asterisk 13.18.3~dfsg-1ubuntu4 built by nobody @ buildd.debian.org on a unknown running Linux on 2018-02-28 06:44:47 UTC

(from Ubuntu 18.04 repositories)

**Expected behavior:**

The Cisco 7970 phone should display the Caller ID name and number on asterisk originated calls.

**Observed behavior:**

The Cisco 7970 phone is displaying "Von Unbekannte Nummer" (from unknown number). On the other side SIP phone are displaying the Caller ID correctly.
[Describe observed behavior here]

**Screenshots**

none

**Addition / Optional Information:**

- Problem was not present on (very old) version of asterisk and chan_sccp. After migrating to Asterisk-13 and the current chan_sccp the originated calls causes the phones to not display the Caller ID.

- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - Intel(R) Xeon(R) CPU L3426
  - 64 Bit
  - [Big/Small Endian]
  - Ubuntu 
  - 18.04

dkgroot, 2020-09-06:
Ran a test on vanilla asterisk-17:
```
Channel: SCCP/98031
Callerid: "callfile" <98031>
MaxRetries: 0
WaitTime: 30
Context: internal
Extension: 98041
Priority: 1
```
Which produces
```
moviebox*CLI> core show channel SCCP/98031-00000004 
 -- General --
           Name: SCCP/98031-00000004
           Type: SCCP
       UniqueID: 1599403847.5
       LinkedID: 1599403847.5
      Caller ID: 98031
 Caller ID Name: callfile
Connected Line ID: 98041
Connected Line ID Name: PHONE4
Eff. Connected Line ID: 98041
Eff. Connected Line ID Name: PHONE4
...
 --   PBX   --
        Context: internal
      Extension: 98041
       Priority: 4
...
  CDR Variables:
level 1: clid="callfile" <98031>
level 1: src=98031
level 1: dst=98041
level 1: dcontext=internal
level 1: channel=SCCP/98031-00000004
level 1: dstchannel=SCCP/98041-00000005
...
```
```
moviebox*CLI> core show channel SCCP/98041-00000005 
 -- General --
           Name: SCCP/98041-00000005
           Type: SCCP
       UniqueID: 1599403850.6
       LinkedID: 1599403847.5
      Caller ID: 98041
 Caller ID Name: PHONE4
Connected Line ID: 98031
Connected Line ID Name: callfile
Eff. Connected Line ID: 98031
Eff. Connected Line ID Name: callfile
...
 --   PBX   --
        Context: internal
      Extension: 
       Priority: 1
     Call Group: 0
   Pickup Group: 0
...
  CDR Variables:
level 1: clid="PHONE4" <98041>
level 1: src=98041
level 1: dcontext=internal
...
```
So on asterisk-17 it seems to produce exactly what you expected. I will run another test on asterisk-13 to make sure this is also the case. I am not sure if you are using FreePBX or not, but i do know that FreePBX does quite a bit of manipulation of the callerid in it's dialplan.

dkgroot, 2020-09-06:
Ran the same test on latest asterisk-13 revision with the same result. Seems to work just fine.

DL44227, 2020-09-07:
My environment seems to differ from yours. I have tested it on again fresh compiled asterisk-13.36.0/chan_sccp_develop (2020-09-01) and also on asterisk-16.13.0/chan_sccp_develop (2020-09-01). In both cases the phones are showing "Von Unbekannte Nummer". My phones are Cisco 7970 (and 7975) with firmware version SCCP70.9-4-2SR13-1S (SCCP75.9-4-2SR2-2S).

Maybe I should create a log. Can you tell me, which debug items I should activate? (sccp debug ...)

dkgroot, 2020-09-07:
@DL44227
I guess that must be the case.
Just set `debug=core,channel,pbx,callinfo` and all the information should hopefully be there. Do make sure to set `core set verbose 5` and `core set debug 1` so we also get the dialplan entries that are being executed.

DL44227, 2020-09-07:
This is the CLI log after setting the above debug levels:

```
    -- Attempting call on SCCP/206 for 251@secure:1 (Retry 1)
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(slin), lineName=206, options=
[Sep  7 14:26:28] NOTICE[5652]: ast116.c:1676 sccp_astwrap_request: Could not match audio codec, Falling back to ULAW
 SCCP: callinfo constructor: 0x7f2160003d30
    -- 206: New channel number: 4 on line 206
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 206
    -- SCCP/206-00000004: (sccp_callinfo_setter) : CallInfo:
        |- calledParty: App206 <206>, valid
        |- presentation: CallerId Presentation Allowed

    -- SCCP:             cid_num: 206
    -- SCCP:            cid_name: App206
    -- SCCP:         accountcode: 206
    -- SCCP:               exten: 
    -- SCCP:             context: secure
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/206-00000004
    -- SCCP: combined audio caps: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
    -- SCCP: reduced audio prefs: [g722/64k, g722/56k, g722/48k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
    -- SCCP: combined video caps: []
    -- SCCP: reduced video prefs: []
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/206-00000004
    -- 206: Allocated asterisk channel SCCP/206-00000004
    -- SCCP: Asterisk request to call SCCP/206-00000004 (dest:206, timeout: 0)
    -- 206: Asterisk request to call SCCP/206-00000004
    -- SCCP/206: Incoming calls:0, Incoming calls limit: 1
    -- SCCP: (sccp_pbx_call) asterisk callerid=' <>'
    -- SCCP/206-00000004: SCCP/206-00000004 : CallInfo:
        |- calledParty: App206 <206>, valid
        |- presentation: CallerId Presentation Allowed

    -- SEP2401C7824D35: Getting the active channel on device.
    -- SEP2401C7824D35: No active channel on device.
    -- SEP2401C7824D35: Ringing Line: 206 on device:SEP2401C7824D35 using channel:SCCP/206-00000004, ringermode:Outside
    -- SEP2401C7824D35: Found matching ld. Aux parameter = 
    -- SEP2401C7824D35: Normal ring occurred.
    -- SEP2401C7824D35: Finish to indicate channel state:RINGING on call:SCCP/206-00000004, lineInstance:1. New channel state:RINGING
    -- SCCP/206-00000004: (sccp_pbx_call) Returning: 0
    -- Called 206
    -- SCCP/206-00000004 is ringing
```

dkgroot, 2020-09-07:
@DL44227 I am missing the rest. The callerid is used when calling the destination (ie: when the calling party has lifted the receiver to start the outbound call).

DL44227, 2020-09-07:
I had to rewrite the call file because I am using the test environment (Asterisk-16) now. The content of the call file is:

```
Channel: SCCP/206
Callerid: "CIDName" <206>
MaxRetries: 0
WaitTime: 30
Context: secure
Extension: 251
Priority: 1
```

Here the full log:

```    -- Attempting call on SCCP/206 for 251@secure:1 (Retry 1)
    -- SCCP: Asterisk asked us to create a channel with type=SCCP, format=(slin), lineName=206, options=
[Sep  7 17:07:41] NOTICE[8749]: ast116.c:1676 sccp_astwrap_request: Could not match audio codec, Falling back to ULAW
 SCCP: callinfo constructor: 0x7f217000c160
    -- 206: New channel number: 7 on line 206
    -- SCCP: (pbx_channel_allocate) try to allocate Inbound channel on line: 206
    -- SCCP/206-00000007: (sccp_callinfo_setter) : CallInfo:
        |- calledParty: App206 <206>, valid
        |- presentation: CallerId Presentation Allowed

    -- SCCP:             cid_num: 206
    -- SCCP:            cid_name: App206
    -- SCCP:         accountcode: 206
    -- SCCP:               exten: 
    -- SCCP:             context: secure
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/206-00000007
    -- SCCP: combined audio caps: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
    -- SCCP: reduced audio prefs: [g722/64k, g722/56k, g722/48k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
    -- SCCP: combined video caps: []
    -- SCCP: reduced video prefs: []
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/206-00000007
    -- 206: Allocated asterisk channel SCCP/206-00000007
    -- SCCP: Asterisk request to call SCCP/206-00000007 (dest:206, timeout: 0)
    -- 206: Asterisk request to call SCCP/206-00000007
    -- SCCP/206: Incoming calls:0, Incoming calls limit: 1
    -- SCCP: (sccp_pbx_call) asterisk callerid=' <>'
    -- SCCP/206-00000007: SCCP/206-00000007 : CallInfo:
        |- calledParty: App206 <206>, valid
        |- presentation: CallerId Presentation Allowed

    -- SEP2401C7824D35: Getting the active channel on device.
    -- SEP2401C7824D35: No active channel on device.
    -- SEP2401C7824D35: Ringing Line: 206 on device:SEP2401C7824D35 using channel:SCCP/206-00000007, ringermode:Outside
    -- SEP2401C7824D35: Found matching ld. Aux parameter = 
    -- SEP2401C7824D35: Normal ring occurred.
    -- SEP2401C7824D35: Finish to indicate channel state:RINGING on call:SCCP/206-00000007, lineInstance:1. New channel state:RINGING
    -- SCCP/206-00000007: (sccp_pbx_call) Returning: 0
    -- Called 206
    -- SCCP/206-00000007 is ringing
    -- SEP2401C7824D35: Getting the active channel on device.
    -- SEP2401C7824D35: No active channel on device.
    -- SEP2401C7824D35: Set the active channel 7 on device
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
    -- SEP2401C7824D35 - SCCP/206-00000007: (recalculateAudioCodecformat) 
        channel capabilities: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
        channel preferences [g722/64k, g722/56k, g722/48k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
        device preferences: [g722/64k, g722/56k, g722/48k, alaw/64k, ulaw/64k, g729, g729a, g729b, g729ab]
        remote caps: []
        Format:g722/64k
    -- SCCP/206-00000007, OpenReceiveChannel with format G.722 64k, payload 6, echocancel: YES, passthrupartyid: 4294967288, pbx_channel_name: SCCP/206-00000007
    -- SCCP: No externip/externhost set in sccp.conf.
When you are running your PBX on a seperate host behind a NAT-TING Firewall you need to set externip/externhost.
    -- SEP2401C7824D35: Getting the active channel on device.
       > 0x7f219400be00 -- Strict RTP learning after remote address set to: 192.168.100.63:18932
    -- SEP2401C7824D35: send callInfo on SCCP/206-00000007 with lineInstance: 1
0x7f217000c160: (sccp_callinfo_send) ci has not changed since last send. Skipped sending
    -- SEP2401C7824D35: (channel_answer_completion) Answering Call: SCCP/206-00000007 (state:OffHook)
    -- SCCP: No externip/externhost set in sccp.conf.
When you are running your PBX on a seperate host behind a NAT-TING Firewall you need to set externip/externhost.
    -- SEP2401C7824D35: Finish to indicate channel state:CONNECTED on call:SCCP/206-00000007, lineInstance:1. New channel state:CONNECTED
    -- SEP2401C7824D35: (channel_answer_completion) Answered channel SCCP/206-00000007
    -- SCCP/206-00000007 answered
    -- Executing [251@secure:1] VoiceMailMain("SCCP/206-00000007", "") in new stack
    -- <SCCP/206-00000007> Playing 'vm-login.gsm' (language 'de')
    -- SEP2401C7824D35: Getting the active channel on device.
       > 0x7f219400be00 -- Strict RTP switching to RTP target address 192.168.100.63:18932 as source
```

dkgroot, 2020-09-07:
Can you make the callfiles destination a Dial() command to some other SCCP phone, instead of doing VoiceMailMain. And you will see the callerid of the calling phone will have been set to "CIDName". Or print the channel information using `core show channel SCCP/206-0000000x` during the call.

DL44227, 2020-09-07:
Here is the channel information while dialing the first phone: 

```
s841*CLI> core show channel SCCP/206-00000008
 -- General --
           Name: SCCP/206-00000008
           Type: SCCP
       UniqueID: 1599493238.10
       LinkedID: 1599493238.10
      Caller ID: 206
 Caller ID Name: CIDName
Connected Line ID: 206
Connected Line ID Name: CIDName
Eff. Connected Line ID: 206
Eff. Connected Line ID Name: CIDName
    DNID Digits: (N/A)
       Language: de
          State: Ringing (5)
  NativeFormats: (g722)
    WriteFormat: g722
     ReadFormat: g722
 WriteTranscode: No 
  ReadTranscode: No 
 Time to Hangup: 0
   Elapsed Time: 0h0m10s
      Bridge ID: (Not bridged)
 --   PBX   --
        Context: secure
      Extension: 206
       Priority: 1
     Call Group: 0
   Pickup Group: 0
    Application: AppDial2
           Data: (Outgoing Line)
 Call Identifer: [C-00000008]
      Variables:
AST_OUTGOING_ATTEMPT=1
SCCP_DEVICE_TYPE=Cisco 7975
SCCP_DEVICE_IP=192.168.100.63
SCCP_DEVICE_MAC=SEPxxx
  CDR Variables:
level 1: clid="CIDName" <206>
level 1: src=206
level 1: dst=206
level 1: dcontext=secure
level 1: channel=SCCP/206-00000008
level 1: lastapp=AppDial2
level 1: lastdata=(Outgoing Line)
level 1: start=1599493238.773150
level 1: answer=0.000000
level 1: end=0.000000
level 1: duration=9
level 1: billsec=0
level 1: disposition=1
level 1: amaflags=3
level 1: accountcode=206
level 1: uniqueid=1599493238.10
level 1: linkedid=1599493238.10
level 1: sequence=7

 -- Streams --
Name: audio
    Type: audio
    State: sendrecv
    Group: -1
    Formats: (g722)
    Metadata:
```

DL44227, 2020-09-07:
Thanks @dkgroot for a workaround with local/ channel:

Callfile:

```
Channel: Local/206
Callerid: "Callerid" <290>
MaxRetries: 0
WaitTime: 30
Context: secure
Extension: 290
Priority: 1
```

Dialplan:

```
[default]
exten => _2XX,1,Dial(SCCP/${EXTEN})
```

This way the callerid will be presented on the phone as expected by me.

closed 2020-09-07 by DL44227