chan-sccp/chan-sccp #532: Codec issue (one way audio) with Cisco 6911

gfreccia, 2020-09-07:
Hi All,

I've 4.2.3 STABLE on Asterisk 13.12.1 with a bunch of Cisco 6911.
Sometimes I've codec issues (one way audio).
Do I've to set something on config?
Is there an updated version that solves the issue?
Below my sccp.conf file.

Thank you!

; (SCCP*)Attached there is my sccp.conf file.
;
; An implementation of Skinny Client Control Protocol (SCCP)
;Skinny Client Control Protocol (SCCP). Release: 4.2.3 STABLE - 6728M (built by 'root' on '2018-11-08 21:17:42 UTC')
;http://chan-sccp-b.sourceforge.net/

[general]
servername = Asterisk					; show this name on the device registration
keepalive = 60						; phone keep alive message evey 60 secs. Used to check the voicemail
debug = core						; console debug level or categories
							; examples: debug = 11 | debug = mwi,event,core | debug = all | debug = none or 0
							; possible categories: 
							; core, sccp, hint, rtp, device, line, action, channel, cli, config, feature, feature_button, softkey, indicate, pbx
							; socket, mwi, event, adv_feature, conference, buttontemplate, speeddial, codec, realtime, lock, newcode, high, all, none
context = sccp
dateFormat = D.M.Y					; M-D-Y in any order. Use M/D/YA (for 12h format)
bindaddr = 192.168.3.23					; replace with the ip address of the asterisk server (RTP important param)
port = 2000						; listen on port 2000 (Skinny, default)
disallow=all						; First disallow all codecs
allow=alaw						; Allow codecs in order of preference
allow=ulaw						; 
allow=g729						;
firstdigittimeout = 16					; dialing timeout for the 1st digit 
digittimeout = 3					; more digits
;digittimeoutchar = #					; you can force the channel to dial with this char in the dialing state
autoanswer_ring_time = 1				; ringing time in seconds for the autoanswer, the default is 0
autoanswer_tone = 0x32					; autoanswer confirmation tone. For a complete list of tones: grep SKINNY_TONE sccp_protocol.h
							; not all the tones can be played in a connected state, so you have to try.
remotehangup_tone = 0x32				; passive hangup notification. 0 for none
transfer_tone = 0					; confirmation tone on transfer. Works only between SCCP devices
callwaiting_tone = 0x2d					; sets to 0 to disable the callwaiting tone
musicclass=default					; Sets the default music on hold class
language=it						; Default language setting
;callevents=no 						; generate manager events when phone 
							; performs events (e.g. hold)
;accountcode=skinny					; accountcode to ease billing
deny=0.0.0.0/0.0.0.0					; Deny every address except for the only one allowed. 
permit=192.168.0.0/255.255.0.0			; Accept class C 192.168.1.0
							; You may have multiple rules for masking traffic.
							; Rules are processed from the first to the last.
							; This General rule is valid for all incoming connections. It's the 1st filter.

;localnet = 192.168.1.0/255.255.255.0 			; All RFC 1918 addresses are local networks
;externip = 1.2.3.4					; IP Address that we're going to notify in RTP media stream
;externhost = mydomain.dyndns.org 			; Hostname (if dynamic) that we're going to notify in RTP media stream
; externrefresh = 60					; expire time in seconds for the hostname (dns resolution)
dnd = on						; turn on the dnd softkey for all devices. Valid values are "off", "on" (busy signal), "reject" (busy signal), "silent" (ringer = silent)
sccp_tos = 0x68						; sets the default sccp signaling packets Type of Service (TOS)  (defaults to 0x68 = 01101000 = 104 = DSCP:011010 = AF31)
							; Others possible values : [CS?, AF??, EF], [0x??], [lowdelay, throughput, reliability, mincost(solaris)], none
sccp_cos = 4						; sets the default sccp signaling packets Class of Service (COS) (defaults to 4)
audio_tos = 0xB8					; sets the default audio/rtp packets Type of Service (TOS)       (defaults to 0xb8 = 10111000 = 184 = DSCP:101110 = EF)
audio_cos = 6						; sets the default audio/rtp packets Class of Service (COS)      (defaults to 6)
video_tos = 0x88					; sets the default video/rtp packets Type of Service (TOS)       (defaults to 0x88 = 10001000 = 136 = DSCP:100010 = AF41)
video_cos = 5						; sets the default video/rtp packets Class of Service (COS)      (defaults to 5)
echocancel = on						; sets the phone echocancel for all devices
silencesuppression = off				; sets the silence suppression for all devices
;callgroup=1,3-4					; We are in caller groups 1,3,4. Valid for all lines
;pickupgroup=1,3-5					; We can do call pick-p for call group 1,3,4,5. Valid for all lines
;amaflags = 						; Sets the default AMA flag code stored in the CDR record
trustphoneip = no					; The phone has a ip address. It could be private, so if the phone is behind NAT 
[... 793 lines cut ...]
; 7910 -- Cisco Unified IP Phone 7910G
; 7911 -- Cisco Unified IP Phone 7911G
; 7912 -- Cisco Unified IP Phone 7912G
; 7935 -- Cisco Unified IP Conference Station 7935
; 7936 -- Cisco Unified IP Conference Station 7936
; 7937 -- Cisco Unified IP Conference Station 7937G
; 7920 -- Cisco Unified IP Wireless Phone 7920
; 7921 -- Cisco Unified IP Wireless Phone 7921G
; 7931 -- Cisco Unified IP Phone 7931G
; 7940 -- Cisco Unified IP Phone 7940G
; 7941 -- Cisco Unified IP Phone 7941G/7941G-GE
; 7942 -- Cisco Unified IP Phone 7942G
; 7945 -- Cisco Unified IP Phone 7945G
; 7960 -- Cisco Unified IP Phone 7960G
; 7961 -- Cisco Unified IP Phone 7961G/7961G-GE
; 7962 -- Cisco Unified IP Phone 7962G
; 7965 -- Cisco Unified IP Phone 7965G
; 7970 -- Cisco Unified IP Phone 7970G
; 7971 -- Cisco Unified IP Phone 7971G-GE
; 7975 -- Cisco Unified IP Phone 7975G
; 7985 -- Cisco Unified IP Phone 7985G
; ata -- Cisco ATA-186 or Cisco ATA-188
; kirk -- Kirk telecom ip phones
; cipc -- Cisco IP Communicator
; nokia-icc -- Nokias ICC Cisco client

dkgroot, 2020-09-07:
@gfreccia I would say start by updating/upgrading your chan-sccp version, by checking out the latest version from github.

gfreccia, 2020-09-25:
Thanks dkgroot!

So I've just to follow the Wiki? (https://github.com/chan-sccp/chan-sccp/wiki/Building-and-Installation-Guide)

        _git clone https://github.com/chan-sccp/chan-sccp chan-sccp
        cd chan-sccp
        git pull

    Compiling:
   
        To get started with configure please run (run "./configure --help" to get a list of all the possible configure flags)

        Note: Some of the configure switches have changed during the latest development, please check before running configure (./configure --help).

        ./configure [....any configure flags you like, see below....]
        make            # or 'make -j2' to use 2 cpu's 
        make install
        make reload     # unload and load a previously installed chan-sccp module_

dkgroot, 2020-09-25:
Yes, good luck !

gfreccia, 2020-09-26:
Hi!
The phone is the 6901 (not 6911), but I think it's the same.

dkgroot, 2020-09-26:
Why are you using directrtp=on in your default device config ? Is that actually required in your case. Currently directrtp seems to be quite broken (working on it)

dkgroot, 2021-01-16:
No response for some time ! We closed this issue for now.
If/when necessary, do reopen the issue please.

closed 2021-01-16 by dkgroot