chan-sccp/chan-sccp #533: 2 small problems with dynamic speed dials

drtmk, 2020-09-09:
Invalid speed dial lamp change during ringing state

**Reproduction Steps:**
Alice calls Bob (ringing state)
All speed dials that have Alice extension configured show: `Bob<-Alice` (and blinking orange)
Charlie dials Alice and gets busy tone (as it should under incominglimit=1), but all speed dials that monitors Alice extension, turns solid red instead of blinking orange although Alice line is still in ringing state.

`
`
Invalid speed dial direction after answering Group Pickup

**Reproduction Steps:**
Alice calls Bob. Charlie presses Group Pickup and anwers that call.
Alice is connected with Charlie, and speed dial that has Charlie extension configured shows `Alice -> Charlie`, instead of `Alice <=> Charlie`.

**chan-sccp version:** 4.3.2 develop - c5c56f4
**asterisk version:** 16.12.0

dkgroot, 2020-09-09:
@drtmk They are pretty much cosmetic as you mentioned. The first one is interesting and i will have to have a look. The second one might be out of our control, but it's interesting to see if we can do anything about it.
Thanks for reporting both of them.

dkgroot, 2020-09-09:
Had a look at the first case and came up with a possible solution. We are dealing with an event attached to a line change, which has multiple channels and therefor multiple channelstates. There is no way to combine these channelstates into anything sensible by default. With the new implementation we will set the callerid information on the hinted speeddial to "In use remote" and set the last state provided by the last lineStateChange. This is not ideal in all cases, but give it a try.

dkgroot, 2020-09-09:
Committed a somewhat simple fix for the second issue. It seems to work, although i don't think it's the nicest way to solve the issue.

drtmk, 2020-09-10:
The first problem still exists. Nothing has changed.

The second problem is fixed. After answering a call with Group Pickup by Charlie, the speed dial for a moment shows `Alice -> Charlie` but it quickly updates to `Alice <=> Charlie`.

dkgroot, 2020-09-10:
A lot has changed for the first case, but i checked the callwaiting version. I will check again with calllimit=1

Ok so second one closed.

dkgroot, 2020-09-10:
I retested and case one seems to work for me, both with and without hitting the incomingcalllimit. Can you make sure you are running in the issues/533 branch and recheck the first case. If debug is required, set "sccp debug core, channel, pbx, indicate, hint" so we can see what is going on with these two calls.

drtmk, 2020-09-10:
I'm running 4.3.2 issues/533 - 26b79e6 with incominglimit=1.
When I call from A to B and during ringing state call A again from 3rd phone, speed dial with A extension shows `Bob <- Alice` but lamp becomes solid red.

[issue533.txt](https://github.com/chan-sccp/chan-sccp/files/5200820/issue533.txt)

https://drive.google.com/file/d/1iZeG3EFs7P98Ek1mm-hIgr2Kj3FYWsxc/view?usp=sharing

dkgroot, 2020-09-10:
I recreated you dialplan with the "handle_dialstatus" macro, returning BUSY(x) to the caller. I reproduced the events in your call sequence, but i am getting a different result on the dynamic speeddial. Not sure but it might be a race condition, which in the case of hints is pretty much impossible to prevent.

I am not to worried to much about this exact case, to much, you are getting signalled about a state change on the monitored phone and with aggregate states it's mostly impossible to do the right thing in all cases (do you agree ?). And as you mentioned these are mostly cosmetic. Shall we merge the current result (with case two closed) and circle back to the parking issue ? We can come back to case 1 afterwards.

drtmk, 2020-09-10:
I think we can merge it to development branch. It's so small bug that probably nobody will even notice it.
https://en.wikipedia.org/wiki/Alice_and_Bob

dkgroot, 2020-09-10:
Merged, branch does not exist anymore

drtmk, 2020-09-11:
It's interesting. I'm not getting that red lamp from first case using 4.3.2 issues/528 - fd7d562. Still blinks orange. Tried many times. Maybe it's fixed.

dkgroot, 2020-09-11:
I had that fixed, but i forgot to push the change last time you checked :-) Woops

drtmk, 2020-09-16:
Can you check again that second problem (Invalid speed dial direction after answering Group Pickup). I'm gettin "Alice->Charlie" speed dial again after Group Pickup.

It's happening on 4.3.2 issues/528 - a78dea3 where this problem should be solved.

dkgroot, 2020-09-16:
I will merge #528 first so that it's code is in develop and then re-open and rebase this branch, so that i can have a look

dkgroot, 2020-09-16:
***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/533 origin/issues/533
git checkout issues/533
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

drtmk, 2020-09-17:
Standard dialing and Group Pickup works ok. Call plates and dynamic speed dials behave fine. So both problems are resolved in this issue #533.

I tested parking again, and there are 2 problems. Dialing parked extension shows `701 <=> Alice` in a speed dial when Alice called parked extension. There should be a name of parked party. (call plate is ok, it shows the name).
After parking a call, dynamic speed dial shows `Parkuj -> Alice` for ex. It should display `Parkuj <=> Alice`.

4.3.2 issues/533 - 351fe4b

dkgroot, 2020-09-17:
> Standard dialing and Group Pickup works ok. Call plates and dynamic speed dials behave fine. So both problems are resolved in this issue #533.

:+1: 

> 1. Dialing parked extension shows 701 <=> Alice in a speed dial when Alice called parked extension. There should be a name of parked party. (call plate is ok, it shows the name).

> 2. After parking a call, dynamic speed dial shows Parkuj -> Alice for ex. It should display Parkuj <=> Alice.

I will check both

dkgroot, 2020-09-22:
I do not know how to fix the second issue, to be honest (also i don't find it that problematic). The way remote hints is currently implemented, we don't have that precise controle about what and how remote calls are displayed.

drtmk, 2020-09-23:
Ok. Works fine.
We can leave the second issue. It's not that problematic.

dkgroot, 2020-09-23:
Merged into develop
Thanks again for reporting and testing !

commit 665d210d (2020-09-22): Fix: Dialing parked extension shows 701 <=> Alice in a speed dial when Alice called parked extension.
     There should be the name of parked party. (callplane is ok, it shows the name).

Fixes: #533.1
commit d02d1d50 (2020-09-23): Fix: hint shortcut making it update the hint->previousState and hint->currentState
Fix: devstate_cb checking hint channel states
Fix: Dialing parked extension shows 701 <=> Alice in a speed dial when Alice called parked extension.
     There should be the name of parked party. (callplane is ok, it shows the name).

Fixes: #533
closed 2020-09-23 by dkgroot