chan-sccp/chan-sccp #536: Wrong dynamic speed dials information during call transfer

drtmk, 2020-09-25:
Speed dials show incorrect information at different stages of a call transfer.

**Consult Transfer**

1. Alice calls Bob, Bob answers and presses [Transfer] soft key button. Alice is put on hold. Bob gets a dial tone.
At this moment speed dials with Bob's extension show **Used remotely -> Bob**. It would be better to have just  **Bob** and red lamp at this stage. Used remotely is misleading.
2. Bob enters Charlie's phone number and starts dialing.
Now Bob's speed dial still shows **Used remotely -> Bob** with blinking orange lamp. I guess it should display **Charlie <- Bob**. Charlie's speed dial is ok.
3. Charlie answers.
Bob's speed dials shows **Used remotely <=> Bob** instead of **Charlie <=> Bob**.
4. Bob presses [Transfer] again to connect Alice with Charlie. Alice's and Charlie's speed dials show they are still connected with Bob which is incorrect, because they're now connected with each other.
Alice's speed dial shows **Bob <=> Alice** instead of **Charlie <=> Alice**.
Charlie's speed dial shows **Bob <=> Charlie** instead of **Alice <=> Charlie**.

**Blind Transfer**
1. Alice calls Bob, Bob answers and presses [Transfer] soft key button.
2. Alice is put on hold. Bob gets a dial tone and enters Charlie's phone number.
3. Bob starts dialing Charlie and pressed [Transfer] soft key button again.
4. Bob's line is free. Alice starts dialing to Charlie.
Both Alice's and Charlie's speed dials show incorrent party name: Bob.
Alice's speed dial shows **Bob <- Alice** instead of **Charlie <- Alice**.
Charlie's speed dial shows **Bob -> Charlie** instead of **Alice -> Charlie**.
5. Charlie answers the call.
All speed dials are ok at this moment.

This issue is only related to speed dials. Call plates are working fine.

**chan-sccp version:** 4.3.2 develop - d02d1d5 
**asterisk version:**  16.12.0 x86 64 bit

dkgroot, 2020-09-25:
@drtmk I was afraid making changes to the way hints and dynamic speeddials were handled would have negative side effects in other parts of the code, and here they seem to be. It might be that transfer was not 100% mapped correctly on the dynamic speeddials before (don't remember).
Currently i am working on directrtp, so this will have to wait a little bit.

drtmk, 2020-09-25:
Speed dials during call transfer were also not working properly even before you made changes.