chan-sccp/chan-sccp #537: Directrtp broken

dkgroot, 2020-09-25:
Working on fixing directrtp

References: #405
References: #532
References: #394
References: #391

drtmk, 2020-09-28:
After holding a call, I cannot resume it. Softkey doesn't do anything, and on the console I'm getting warnings:

```
[Sep 27 20:47:15] WARNING[8546][C-00000002]: sccp_session.c:281 sccp_session_waitForPendingRequests: SEPD0C282425243: waitForPendingRequests timed out!
    -- SEPD0C282425243: Set calledPartyNumber '120' calledPartyName 'Bob'
       > 0x7f93680378a0 -- Strict RTP learning after remote address set to: 10.100.0.202:29596
    -- SEPD0C282425243: Hold the channel SCCP/120-00000006
    -- SEPD0C282425243: Resume the channel SCCP/120-00000006
[Sep 27 20:47:20] WARNING[8546][C-00000002]: sccp_session.c:281 sccp_session_waitForPendingRequests: SEPD0C282425243: waitForPendingRequests timed out!
    -- SEPD0C282425243: Set calledPartyNumber '120' calledPartyName 'Bob'
    -- SEPD0C282425243: Hold the channel SCCP/120-00000006
    -- SEPD0C282425243: Resume the channel SCCP/120-00000006
[Sep 27 20:47:25] WARNING[8546][C-00000002]: sccp_session.c:281 sccp_session_waitForPendingRequests: SEPD0C282425243: waitForPendingRequests timed out!
    -- SEPD0C282425243: Set calledPartyNumber '120' calledPartyName 'Bob'
    -- SEPD0C282425243: Hold the channel SCCP/120-00000006
    -- SEPD0C282425243: Resume the channel SCCP/120-00000006
```

dkgroot, 2020-09-29:
Can you set "sccp debug core, channel, rtp, pbx" during testing ? I was not able to reproduce the 'resume' issue you saw, and you debug log was a little short for me to figure out what was going on in this instance.

drtmk, 2020-09-30:
Right now when running  4.3.2 issues/537 - f415823, asterisk shuts down when phone is trying to register:

```
Asterisk Ready.
asterisk*CLI> sccp show version
Skinny Client Control Protocol (SCCP). Release: 4.3.2 issues/537 - f415823 (built by 'root' on '2020-09-30 06:15:44 UTC')
asterisk*CLI>
asterisk*CLI>
asterisk*CLI>
asterisk*CLI>
asterisk*CLI>
    -- SCCP: Accepted Client Connection from 10.100.0.201:49740
  == Alarm Type: LastOutOfServiceInformation
  == Reason Enum: 10
  == Last Event Sent: 0:KeepAlive
  == Last Event Received: 256:KeepAliveAck
    -- SEPD0C28243AE44: nat=Off set manually in config file
    -- SEPD0C28243AE44: Got rtpPort:3500 which the device wants to use for media
asterisk*CLI>
Disconnected from Asterisk server
Asterisk cleanly ending (0).
Executing last minute cleanups
Asterisk ending (0).
root@asterisk:/usr/src/directrtp#
```

I tried many times. Asterisk always shuts down.
I also restarted phones, to be sure they're not in some weird state.

dkgroot, 2020-09-30:
Ooh crap i guess this might be related with work i did in the last couple of days. Can you show me the gdb backtrace for the core dump ?

drtmk, 2020-09-30:
Asterisk works now.
So this is the log when Alice calls Bob. Bob holds a call, and tries to resume it.
[alice, bob, hold, resume.txt](https://github.com/chan-sccp/chan-sccp/files/5305578/alice.bob.hold.resume.txt)

I was able to resume it, but it worked after few seconds after pressing [Resume]. And audio worked only from Alice to Bob.
Same thing happens, when Alice holds a call and tries to resume it.

Basically when I hold a call, and I'm trying to resume it, phone stops responding. Like if it lost connection to asterisk.

dkgroot, 2020-09-30:
@drtmk Thanks for checking and testing. I will have another look at the resume.txt to see what is different between your system and mine. The delay during resume is a bit of an issue, but i am not yet sure how to prevent it. During directrtp we need to start and new openReceiveChannel and have to wait for it to come back, to see the ip-address and port used, before we can continue the handling of resume.

dkgroot, 2020-09-30:
I have changed the handling slightly. Have not looked at the one way audio for now, just trying to get rid of the delay first.

dkgroot, 2020-10-01:
Pushed another update making call resume asynchronous, so the waitForPendingRequest can be processed in a separate thread. This should take care of the timeout events you were seeing. Apparently my machine was slow enough for the phones to beat the race condition :-)

drtmk, 2020-10-01:
Good job. There's no delay right now. Phone behaves ok.
In a call transfer audio also works ok.
There's still a problem in one way audio after resume in cisco phones.
There's also one way audio between 79xx and SIP phone (using direct_media=yes in pjsip). Audio goes only from SIP to cisco. Direction of a call doesn't matter. Both phones are using g722.

Here's a log from a call from 100 (10.100.0.201 cisco) to 121 (SIP phone, 10.100.0.3).
[100 to 121 (SIP).txt](https://github.com/chan-sccp/chan-sccp/files/5309703/100.to.121.SIP.txt)

dkgroot, 2020-10-01:
Ok will look into the one way issue. Also my solution to the delay, still needs a little tweaking.

dkgroot, 2020-10-01:
Made some smaller changes. Fixed the resume function so the channel lock was taken and released correctly. Updated codec framing and ip-address/port updating. Ran a couple of tests with chan_sip. which seemed ok.

When posting your debug log, please grab it from /var/log/asterisk/full with core set debug 3 set, so we can see how/when asterisk is performing the remote bridging part.

dkgroot, 2020-10-01:
Still have not got resume back in working order. All the code is starting to get so convoluted :-(

drtmk, 2020-10-01:
Don't put a pressure on yourself. Most people will not use directmedia. It's not a big deal.

drtmk, 2020-10-02:
The only difference I see is that I can hear audio in both ways calling SIP phone. But I guess that doesn't use direct connection.
[sip.txt](https://github.com/chan-sccp/chan-sccp/files/5316251/sip.txt)
Any idea why they don't want to talk directly?

And here's a log from resume function taken from /var/log/asterisk/full. It's still one way audio.
[resume.txt](https://github.com/chan-sccp/chan-sccp/files/5316255/resume.txt)

dkgroot, 2020-10-02:
I think i finally figured it out. and al lot of stuff i did in the last week might have been superfluous  but we will see. I will have a look in a couple of hours and see what can be done.

dkgroot, 2020-10-03:
@drtmk Had to do quite a bit of moving code about and rethinking the solution to make it clean and nice again.
Have not dared to check how all these changes may have impacted "in-directrtp" to be honest. I hope i have done an ok job. I will check the fall out tomorrow.

drtmk, 2020-10-03:
I will be in my phone lab on monday.

dkgroot, 2020-10-03:
I had to test indirect-rtp with my new code, and low and behold it also works (with a tiny little patch)... => progress !

drtmk, 2020-10-05:
Using 4.3.2 issues/537 - 64a1b08 and directrtp enabled, there's no audio in both ways in a simple call.
But it starts working correctly when I make a hold and resume.
[call.txt](https://github.com/chan-sccp/chan-sccp/files/5325498/call.txt)

I would like to ask about above tests.
By call forward you mean use the CFwdALL softkey?
By conference you mean ConfBridge or that which is created from visual menu on the phone?
What kind of NAT whould you like to test? You want to place two phones behind NAT or Asterisk behind NAT?

dkgroot, 2020-10-05:
@drtmk I will have a look at the call.txt a little later, when i am awake. I did not have simple call issues.

- yes cfwdall
- Conf softkey on the phone. Should (ought to) prevent directrtp and switch back to indirect automatically
- Single NAT on one side to start and then maybe two phones behind nat on a remote network, using directrtp between them (the classical case where directrtp makes the most sense).

drtmk, 2020-10-05:
I will not have much time today to make these tests.

dkgroot, 2020-10-05:
That last change i made is not a good idea. I have not slept yet :-(. I think your system is preventing directrtp during bringing, but we can only review that in the `full` log.

dkgroot, 2020-10-05:
Figured out the cause, if the directrtp fails to be setup (for any cause) it will not fall back to in-direct like it should. Will need extra work. You can skip testing for now.