chan-sccp/chan-sccp #545: Cannot allocate memory. Failed to create new channel thread. Failed to start PBX :(

bknv, 2020-10-05:
Hello.
After 3 week uptime on the server appears problem where all calls not created and in log i see Error/Warning: 
`[2020-10-05 09:09:40] WARNING[2818][C-00003345] pbx.c: Failed to create new channel thread
[2020-10-05 09:09:40] WARNING[2818][C-00003345] chan_sip.c: Failed to start PBX :(
[2020-10-05 09:09:57] ERROR[2811] tcptls.c: TCP/TLS unable to launch helper thread: Cannot allocate memory
[2020-10-05 09:10:25] ERROR[2755] tcptls.c: TCP/TLS unable to launch helper thread: Cannot allocate memory
[2020-10-05 09:09:49] WARNING[2804] pbx.c: Failed to create new channel thread
[2020-10-05 09:09:49] WARNING[2804] chan_iax2.c: Unable to start PBX on IAX2/iaxpc-8743
`
I have this problem for a long time, my hands didn't reach. Problem solved if execute service restart asterisk so i created cron with automatical restart asterisk every weekend.
The other day i update to latest version. What should I pay attention to?

`[root@asteriskpc ~]# ulimit
unlimited
`

Before restart service asterisk 
`[root@asterisk ~]# free -m
              total        used        free      shared  buff/cache   available
Mem:           5805        2280        1863         327        1661        2829
Swap:          3967         264        3703`

after

`free -m
              total        used        free      shared  buff/cache   available
Mem:           5805        1387        2586         327        1831        3731
Swap:          3967         154        3813`

**Skinny Client Control Protocol (SCCP). Release: 4.3.2 develop - 81f140a (built by 'root' on '2020-05-27 18:18:45 UTC')** 

**Asterisk 13.27.1 built by mockbuild @ jenkins7 on a x86_64 running Linux on 2019-09-06 21:11:59 UTC** 

- Platform:
VMware ESXi, 6.5.0, 10884925
RAM : 6GB
CPU : 4 core 1500Mhz
Distro FreePBX

dkgroot, 2020-10-06:
This looks like an asterisk issue. Not (directly) related to chan-sccp.
Most C applications that run into memory allocations problems will exit/quit/give up, because once you run into a memory allocation issue, there is not easy way to get out of it. How would a user space program be able to control freeing some memory for it to use.
Even though you system does still have "free" memory, it might not be available for asterisk to use. For example if `ulimit` was used to set the max amount of allocatable memory for asterisk. It might be a good idea to recompile asterisk and with the `memory/allocation` debug flag enabled, so you can check if there is a memory leak of some kind.
Q: Have you observed that the amount of memory used by asterisk grows over time ?
Note: chan_iax can be a bit of a memory hog, so if you are not using iax, it might be a good idea not to load that module.
Note1: I don't think there is a lot i can do or help with, regarding this issue report (or maybe i am missing something)

dkgroot, 2020-11-12:
Have you seen any more reports of this kind. Did you managed to resolve the issue ?

bknv, 2020-11-12:
Hello @dkgroot.
I did't recompile my asterisk with enabled malloc, because on this server use distro FPBX. I don't know how recompile FPBX with enabled malloc.
But...
I did update (FreePBX Framework and Core FPX to latest version) server 1 week last, because in log i seen Warning:
ast_expr2.fl: ast_yyerror(): syntax error: syntax error, unexpected '=', expecting $end; Input:

Today uptime server 1week 2 days. I waiting uptime 3 week.
If there  is no problem in 3 weeks, you will can consider the issue resolved

dkgroot, 2020-11-12:
> I did update (FreePBX Framework and Core FPX to latest version) server 1 week last, because in log i seen Warning:
ast_expr2.fl: ast_yyerror(): syntax error: syntax error, unexpected '=', expecting $end; Input:

That is mostly likely caused when parsing dialplan logic i think...
Thanks for reporting and being willing to check/test.

bknv, 2020-11-12:
Link to issue https://issues.freepbx.org/browse/FREEPBX-20659

dkgroot, 2020-11-12:
I guess i guessed corectly :-)

closed 2020-11-12 by dkgroot