chan-sccp/chan-sccp #546: make runs into error: unknown field 'nonoptreq'

xJustmy2cents, 2020-11-09:
Fresh Install of Asterisk 18 from Source on an ARMBIAN Stretch.

**Reproduction Steps:**

1. Download, compile and install Asterisk 18 including samples -No DAHDI or Other
2. Download current release of CHAN_SCCP from GH
3. Run configure with "--with-asterisk=[path to source files]"
4. run "make"
5. get error

**chan-sccp version:** chan-sccp-4.3.1
**asterisk version:** asterisk-18.0.1
**Expected behavior:**
finish "make" w/o error to continue with "make install"

**Observed behavior:**
"....
make[3]: Entering directory '/root/chan-sccp-4.3.1/src/pbx_impl/ast114'
  CC       libast114_la-ast114.lo
In file included from ast114.c:37:0:
ast114.c:3665:174: error: unknown field 'nonoptreq' specified in initializer
 L_KEY, AST_MODFLAG_LOAD_ORDER, SCCP_VERSIONSTR,.load = load_module,.unload = unload_module,.reload = module_reload,.load_pri = AST_MODPRI_DEFAULT,.nonoptreq = "chan_local");
                                                                                                                                                   ^
cc1: warning: unrecognized command line option '-Wno-format-truncation'
Makefile:567: recipe for target 'libast114_la-ast114.lo' failed
make[3]: *** [libast114_la-ast114.lo] Error 1
make[3]: Leaving directory '/root/chan-sccp-4.3.1/src/pbx_impl/ast114'
Makefile:1013: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/root/chan-sccp-4.3.1/src'
Makefile:618: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/root/chan-sccp-4.3.1/src'
Makefile:610: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
..."

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [No]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes]
- Platform:
Linux ******* 4.19.59-sunxi #5.91 SMP Mon Jul 15 14:09:32 CEST 2019 armv7l GNU/Linux

dkgroot, 2020-11-09:
Hi xJustmy2cents,

On 10-11-2020 00:25, xJustmy2cents wrote:
> Fresh Install of Asterisk 18 from Source on an ARMBIAN Stretch.
>
> *Reproduction Steps:*
>
>  1. Download, compile and install Asterisk 18 including samples -No
>     DAHDI or Other
>  2. Download current [...]
This is the issue. It's using "ast114" instead of "ast118".
Which git revision are you one. Should/Ought be one of the latest,
because asterisk-18 support was only added 2 weeks ago. From the result
you are showing asterisk-18 is not being recognised and the wrong ast114
directory/file is being used.

Please check your revision and branch.
Current Branch = develop
Latest revision = 73566485
Current Version = 4.3.2 not 4.3.1

Look's like you might be on the wrong branch.

Please remove the version you currently have and run:
git clone https://github.com/chan-sccp/chan-sccp

To get a fresh copy.

Regards,
Diederik

xJustmy2cents, 2020-11-10:
Hello,
I took it from the GH WEbsite at this link:
[image]

I will check for your version and report back.
Thanks

xJustmy2cents, 2020-11-10:
Solved.

I was able to download the "zeta" preRelease and it run w/o any issue.

Nevertheless I had to use "--with-asterisk" and point to the downloaded source folder.

Remark: For some user it is a hurdle to understand, that "asterisk-devel" is nothing else, than the source code libs provided on GH. Maybe you think about the error message, as the search results at the all-knowing-heap-dump returns some misleading results about "yum"/"apt" a.s.o.

For Chan_SCCP it seamlessly accepted my old sccp.conf (from Ast.v11) and my cisco environment is working again.

Thanks for your work!!!!

dkgroot, 2020-11-10:
Hi xJustmy2cents,

>
> Solved.
Good to hear that !
>
> I was able to download the "zeta" preRelease and it run w/o any issue.
Yep that is the one ! I would suggest you use git clone instead of a
prepared "zip" release though. I don't create (pre) releases often enough.
>
> Nevertheless I had to use "--with-asterisk" and point to the
> downloaded source folder.
Aah you compiled asterisk manually, i see. A while ago digium change the
asterisk "make install" and it does not include the header files
anymore. You need to use "make install-headers" separately now, to
install them to /usr/include/asterisk so that thirdparty applications
like chan-sccp can find them.. Not very GNU/autoconf/automake/friendly
like, but that's what they did.
>
> Remark: For some user it is a hurdle to understand, that
> "asterisk-devel" is nothing else, than the source code libs provided
> on GH. Maybe you think about the error message, as the search results
> at the all-knowing-heap-dump returns some misleading results about
> "yum"/"apt" a.s.o.
I guess i can add line stating the requirement to run "make
install-headers" for self-compiled asterisk versions i guess :-) Thanks
for the suggestion !
>
> For Chan_SCCP it seamlessly accepted my old sccp.conf (from Ast.v11)
> and my cisco environment is working again.
>
> Thanks for your work!!!!
No problem,

Diederik

closed 2020-11-10 by dkgroot

dkgroot, 2020-11-10:
I updated the warning generated by the configure script hopefully the new one is a little more clear.
([See 8c04e4a](https://github.com/chan-sccp/chan-sccp/commit/8c04e4adf242688b4204211fface25b92947de47#comments))