chan-sccp/chan-sccp #548: SCCP: Device 'SEP000000000000' not found in realtime table 'sccpdeviceconfig'

kuchka506, 2020-11-27:
Cisco 7940 registration rejected.
Do I've to set something on config?
Thank you!

### Log asterisk
`[2020-11-27 09:28:40] NOTICE[5403]: sccp_session.c:386 session_buffer2msg:     -- SCCP: (session_dissect_msg) Incoming message is smaller(76) than known size(184).
[2020-11-27 09:28:40] NOTICE[5403]: sccp_actions.c:240 sccp_handle_message:     -- 192.168.4.75:2000: >> Got message Register Message (0x1)
[2020-11-27 09:28:40] NOTICE[5403]: sccp_actions.c:123 check_session_message_device: 192.168.4.75:2000: SCCP Handle Message: Register Message(0x0001) 76 bytes length
[2020-11-27 09:28:40] NOTICE[5403]: sccp_actions.c:820 handle_register:  SEP001B0CDB8F24: is registering, Instance: 1, UserId: 0, Type: Cisco 7940 (8), Version: 11 (loadinfo '')
[2020-11-27 09:28:40] NOTICE[5403]: sccp_device.c:3355 __sccp_device_find_realtime:     -- SCCP: Device 'SEP001B0CDB8F24' not found in realtime table 'sccpdeviceconfig'
[2020-11-27 09:28:40] NOTICE[5403]: sccp_actions.c:876 handle_register: SEP001B0CDB8F24: Rejecting device: Device Unknown
[2020-11-27 09:28:40] NOTICE[5403]: sccp_session.c:1238 sccp_session_send2: SCCP: Sending Message: Register Reject Message(0x009D) 36 bytes length
[2020-11-27 09:28:40] NOTICE[5403]: sccp_session.c:785 sccp_session_device_thread:        > SCCP: set poll timeout 60 for session 44
[2020-11-27 09:28:40] NOTICE[5403]: sccp_session.c:836 sccp_session_device_thread:     -- SCCP: Exiting sccp_socket device thread
[2020-11-27 09:28:40] NOTICE[5403]: sccp_session.c:682 sccp_session_device_thread_exit:     -- SCCP: Session without a device attached !
[2020-11-27 09:28:40] NOTICE[5403]: sccp_session.c:685 sccp_session_device_thread_exit:     -- SCCP: cleanup session
[2020-11-27 09:28:40] NOTICE[5403]: sccp_session.c:649 destroy_session:     -- SCCP: Destroy Session 192.168.4.20
[2020-11-27 09:28:40] NOTICE[5403]: sccp_session.c:653 destroy_session:     -- SCCP: Shutdown socket 44
[2020-11-27 09:28:40] NOTICE[5403]: sccp_session.c:655 destroy_session:     -- SCCP: Closing socket 44
`
### Database
`Your MariaDB connection id is 5193
Server version: 5.5.65-MariaDB MariaDB Server
MariaDB [asterisk]> select * from sccpdeviceconfig\G;
*************************** 1. row ***************************
                    button: line,1017,default
                      type: 7940
                     addon: NULL
               description: 3243534
                  tzoffset: 03
              imageversion: NULL
                      deny: 0.0.0.0/0.0.0.0
                    permit: 0.0.0.0/0.0.0.0
                  earlyrtp: none
                   mwilamp: on
                 mwioncall: off
                dndFeature: on
                  transfer: off
                   cfwdall: off
                  cfwdbusy: off
                   private: off
                   privacy: full
                       nat: auto
                 directrtp: off
                softkeyset: default
                 audio_tos: 0xB8
                 audio_cos: 6
                 video_tos: 0x88
                 video_cos: 5
                conf_allow: on
conf_play_general_announce: on
   conf_play_part_announce: on
        conf_mute_on_entry: off
  conf_music_on_hold_class: default
        conf_show_conflist: on
            force_dtmfmode: auto
                    setvar: NULL
           backgroundImage: NULL
       backgroundThumbnail: NULL
                  ringtone: NULL
                      name: SEP001B0CDB8F24
           directed_pickup: on
   directed_pickup_context: NULL
directed_pickup_modeanswer: off
                   _hwlang: null
             useRedialMenu: off
1 row in set (0.00 sec)
`
### Version FreePBX
[image]

###extconfig_custom.conf
`[settings]
;
;
; [alias] => [driver],[dbsection],[dbtablename]
sccpdevice => mysql,general,sccpdeviceconfig                  ; Please note use the '=>' instead of '='
sccpline => mysql,general,sccpline`
###res_config_mysql.conf
`[sccp]
dbhost=localhost
dbname=asterisk
dbuser=
dbpass=
dbport=3306
dbsock=/var/lib/mysql/mysql.sock

[general]
dbhost=localhost
dbname=asterisk
dbuser=
dbpass=
dbport=3306
dbsock=/var/lib/mysql/mysql.sock`
###sccp.conf
`;--------------------------------------------------------------------------------;
;          Do NOT edit this file as it is auto-generated by FreePBX.             ;
;--------------------------------------------------------------------------------;
; For information on adding additional paramaters to this file, please visit the ;
; FreePBX.org wiki page, or ask on IRC. This file was created by the new FreePBX ;
; BMO - Big Module Object. Any similarity in naming with BMO from Adventure Time ;
; is totally deliberate.                                                         ;
;--------------------------------------------------------------------------------;

[general]
debug=all
servername=VPBXSCCP
keepalive=60
context=from-internal
dateformat=D.M.Y
bindaddr=0.0.0.0
port=2000
disallow = all
allow = alaw
allow = ulaw
firstdigittimeout=16
digittimeout=8
autoanswer_ring_time=0
autoanswer_tone=0x32
remotehangup_tone=0x32
transfer_tone=0x32
callwaiting_tone=0x2D
musicclass=default
language=en
deny = 0.0.0.0/0.0.0.0
permit = 0.0.0.0/0.0.0.0
dndFeature=on
sccp_tos=0x68
sccp_cos=0x4
audio_tos=0xB8
audio_cos=0x6
video_tos=0x88
video_cos=0x5
echocancel=off
silencesuppression=off
private=on
callanswerorder=oldestfirst
hotline_enabled=off
hotline_context=default
hotline_extension=*60
callhistory_answered_elsewhere=Missed Calls
hotline_label=hotline
devicetable=sccpdeviceconfig
linetable=sccpline
nat=off
displayconfig=sccpgeneral
localnet = 0.0.0.0/0.0.0.0
directed_pickup_modeanswer=on
mwilamp=on
mwioncall=off
directrtp=off
earlyrtp=none
simulate_enbloc=on
cfwdall=off
cfwdbusy=off
regcontext=sccpregistration
externip=
`

dkgroot, 2021-01-01:
Sorry i overlooked your issue report and i took a while longer to get to it. Is the issue still present or was it already resolved ?

The problem is in sccp.conf:
`devicetable=sccpdeviceconfig`
devicetable should be refering to the name `extconfig_custom.conf` name of the entry which is `sccpdevice`. It should not be pointing at the view in the database.

So update sccp.conf to use:
`devicetable=sccpdevice`
And it should just work.

dkgroot, 2021-01-05:
This should resolve the issue (if you did not fix it already). Closing the issue report for now. If you are still running into a problem after making this change, please reopen the issue.

closed 2021-01-05 by dkgroot