chan-sccp/chan-sccp #549: Unable to run ./configure on Raspberry Pi 4 Model B running Arch Linux, "ASTERISK_VERSION could not be established"

TheROFL98, 2020-12-17:
When running ./configure with or without options `--enable-conference --enable-video --enable-distributed-devicestate --enable-advanced-functions` it terminates with the message "ASTERISK_VERSION could not be established"

**Reproduction Steps:**

1. Install Asterisk, either through the AUR or from source.
2. Attempt to install chan-sccp from source following instructions on Readme (installing through AUR complains of the architecture not being x86)

**chan-sccp version:** Commit 54bda4380173f518f2f6cf7646ad95ed470a4379

**asterisk version:** 18.1.0-2

**Expected behavior:**

./configure should run successfully, allowing the user to run make and subsequently being able to install chan-sccp

**Observed behavior:**

The process terminates with `ASTERISK_VERSION could not be established`, and upon closer inspection, shows that gcc terminated with message `gcc: error: unrecognized command-line option '-fstack-fno-fsigned-char'; did you mean '-fno-signed-char'?` while trying to run `configure:20708: gcc -c -Og -Wall -I.  -isystem /usr/include -O2 -fstack-fno-fsigned-char -I/usr/include/libxml2 -g3 -U_FORTIFY_SOURCE -I.  -isystem /usr/include -O2 -fstack-fno-fsigned-char -I/usr/include/libxml2 -g3 conftest.c >&5`

**Screenshots**

https://imgur.com/12xnEJV (Error Message)
https://imgur.com/RtTJ8gw (Failed command that executed during configure)

**Addition / Optional Information:**
- Platform:
  - Raspberry Pi 4 Model B
  - BCM2711
  - 32-bit armv7l
  - Small Endian
  - Arch Linux ARM armv7l
  - Linux 5.4.80-1-ARCH
- Direct Contact information: therofl98@gmail.com (Email) TheROFL98#8383 (Discord)
- Log Files: https://pastebin.com/U8Vg0108

dkgroot, 2020-12-18:
"-fstack-fno-fsigned-char" definitely looks wrong. I did a grep through the configure sources but could not find any reference to this string or "signed-char". I had though it would have been a faulty string concatenation but that does not seem to be the case.
We do re-use any CFLAGS set in the current environment being run. Can you check your CFLAGS and  CXXFLAGS ? 

Or run:
```
CFLAGS="" CXXCFLAGS="" ./configure --enable-conference --enable-video --enable-distributed-devicestate --enable-advanced-functions
```
To prevent reusing any environment provided ones.

TheROFL98, 2020-12-18:
Running the command with what you provided has given me the same error, so it would appear that this might not be an environment problem. Running the command with the flag `--with-asterisk-version=18.1` bypasses the version check, however this time it errors again with `Couldn't find asterisk/lock.h. No need to go any further. All will fail.` When checking the config.log I can see the same gcc error with `-fstack-fno-fsigned-char`.

dkgroot, 2020-12-18:
@TheROFL98 

Have you tried to see which of the environment flags is the culprit ? You can use `set` to view the current environment flags provided.

If asterisk/lock.h is missing that would mean you do not have the asterisk-devel package installed or if you installed asterisk manually you have forgotten to run `make install-headers` when installing the asterisk binaries (This make command was added recently). The header-files are needed by chan-sccp be able to compile a valid asterisk module.

TheROFL98, 2020-12-18:
I couldn't find anything useful by running `set` with and without grep, and I've already installed the headers with that method. I've actually been able to find the file under /usr/include as well as ast_version.h but still won't register with ./configure.

TheROFL98, 2020-12-19:
That should work. Do you have or use an instant messenger?

dkgroot, 2020-12-19:
@TheROFL98 
We have https://gitter.im/chan-sccp/chan-sccp if that helps. Or i have google hangout if need be.

dkgroot, 2021-01-01:
@TheROFL98
The cause for the issue was a broken asterisk.pc file in /usr/lib/pkgconfig containing a non existing compiler flags:`-fstack-fno-fsigned-char`. After manually editing the file the compilation of chan-sccp worked without a hitch. We need to figure out what/who is responsible for the broken asterisk pkg-config file and inform them. I am assuming that [mkpkgconfig](https://github.com/asterisk/asterisk/blob/master/build_tools/mkpkgconfig) might be the culprit. Can you test this script on your local system, to see if it produces the faulty asterisk.pc file. If it is, then it should be reported to https://issues.asterisk.org/jira/.

dkgroot, 2021-01-03:
@samalero As you seem to be working on raspbian as well, have you noticed a similar issue and/or broken /usr/lib/pkgconfig/asterisk.pc file ?

closed 2021-01-05 by dkgroot