chan-sccp/chan-sccp #550: Segfault in ast_format_cap_append_from_cap

pnlarsson, 2020-12-18:
Segfault with latest develop branch:

[2020-12-18 09:24:23] VERBOSE[15594][C-0000004e] app_dial.c: PJSIP/ngcore-00005786 is making progress passing it to SCCP/wid19985-00000174
[2020-12-18 09:24:37] Asterisk 13.38.0 built by root @ uc168 on a x86_64 running Linux on 2020-12-15 22:09:31 UTC

**chan-sccp version:** Skinny Client Control Protocol (SCCP). Release: 4.3.2 (HEAD - 76000de (built by 'root' on '2020-12-15 22:18:34 UTC')

**asterisk version:** Asterisk 13.38.0 built by root @ uc168 on a x86_64 running Linux on 2020-12-15 22:09:31 UTC

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes]
- Problem can be reliably reproduced, doesn't happen randomly: [No]
- Platform:
  - Debian 10 x86_64

Backtrace (full bt availible):
```[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `/usr/local/ngcore/service/pbx/sbin/asterisk -f -F -g -C /usr/local/ngcore/servi'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000056055f1a111b in ast_format_cap_append_from_cap (dst=0x7f532007d870, src=0x7f5320172e70, type=AST_MEDIA_TYPE_AUDIO) at format_cap.c:311
311			struct format_cap_framed *framed = AST_VECTOR_GET(&src->preference_order, idx);
[Current thread is 1 (Thread 0x7f511b289700 (LWP 15594))]
#0  0x000056055f1a111b in ast_format_cap_append_from_cap (dst=0x7f532007d870, src=0x7f5320172e70, type=AST_MEDIA_TYPE_AUDIO) at format_cap.c:311
        __idx = 0
        framed = 0x7f529c011d60
        idx = 0
        res = 0
#1  0x00007f539fa27a20 in __find_joint_capabilities (c=c@entry=0x7f529c0153f0, peer=peer@entry=0x7f532006cd48, media_type=media_type@entry=AST_MEDIA_TYPE_AUDIO, remoteCapabilities=remoteCapabilities@entry=0x7f529c015718) at ast113.c:463
        buf = "\260\033\002\234R\177\000\000\020X\243\237S\177\000\000\260\033\002\234R\177\000\000\000\032Vt)\240~\342\240\063(\033Q\177\000\000\240\063(\033Q\177\000\000\377\000\000\000\000\000\000\000\223\006-_\005V\000\000H5(\033Q\177\000\000 \335\066_\005V\000\000\310J\002 S\177\000\000\240ǁCT\177\000\000\001\200\255\373Q\177\000\000\310J\002 S\177\000\000\067g\334_\000\000\000\000\243.\a\000\000\000\000\000\027\000\000\000\030\000\000\000\t\000\000\000\022\000\000\000\v\000\000\000x\000\000\000\005\000\000\000`\001\000\000\000\000\000\000\000\000\000\000 \200B\324R\177\000\000\000\000\000\000\000\000\000\000\243.\a", '\000' <repeats 13 times>, "\220"...
        codec_buf = 0x7f511b283290
        owner = 0x7f529c023258
        caps = 0x7f529c01b0b0
        payload_caps = 0x7f5320174080
        remote_caps = 0x7f532007d870
        joint = 0x7f5320002fe0
        best_fmt_cap = 0x0
        best_fmt_native = 0x0
        __PRETTY_FUNCTION__ = "__find_joint_capabilities"
#2  0x00007f539fa28498 in pbx_retrieve_remote_capabilities (c=0x7f529c0153f0) at ast113.c:555
        codec_buf = 0x7f511b283590
        __PRETTY_FUNCTION__ = "pbx_retrieve_remote_capabilities"
        ast = 0x7f529c023258
        remotePeer = 0x7f532006cd48
        iterator = 0x7f5320000960
#3  0x00007f539fa29dae in sccp_astwrap_indicate (ast=0x7f529c023258, ind=14, data=0x0, datalen=0) at ast113.c:676
        res = 0
        c = 0x7f529c0153f0
        ref0 = {ptr = 0x7f511b283650, file = 0x7f539fa66b5f "ast113.c", func = 0x7f539fa63ad0 <__PRETTY_FUNCTION__.24754> "sccp_astwrap_indicate", line = 614}
        __PRETTY_FUNCTION__ = "sccp_astwrap_indicate"
        d = 0x7f529c006e60
        ref1 = {ptr = 0x7f511b283658, file = 0x7f539fa66b5f "ast113.c", func = 0x7f539fa63ad0 <__PRETTY_FUNCTION__.24754> "sccp_astwrap_indicate", line = 621}
        inband_if_receivechannel = false
#4  0x000056055f131aa3 in indicate_data_internal (chan=0x7f529c023258, _condition=14, data=0x0, datalen=0) at channel.c:4782
        condition = AST_CONTROL_PROGRESS
        ts = 0x0
        t38_parameters = 0x7f529c023220
        res = 22021
        __PRETTY_FUNCTION__ = "indicate_data_internal"
#5  0x000056055f131f67 in ast_indicate_data (chan=0x7f529c023258, _condition=14, data=0x0, datalen=0) at channel.c:4948
        res = 32595
        awesome_frame = 0x0
        __PRETTY_FUNCTION__ = "ast_indicate_data"
#6  0x000056055f13139b in ast_indicate (chan=0x7f529c023258, condition=14) at channel.c:4538
No locals.
#7  0x00007f54300bf0c8 in wait_for_answer (in=0x7f529c023258, out_chans=0x7f511b285550, to=0x7f511b28554c, peerflags=0x7f511b2858b8, opt_args=0x7f511b284d90, pa=0x7f511b284e70, num_in=0x7f511b285530, result=0x7f511b284e6c, dtmf_progress=0x0, ignore_cc=0, forced_clid=0x7f511b284c40, stored_clid=0x7f511b284bf0) at app_dial.c:1535
        f = 0x7f51940635e0
        c = 0x7f532006cd48
        o = 0x7f5320077a50
        pos = 2
[... 81 lines cut ...]
        pos = 0
        found = 1
        res = 0
        autoloopflag = 0
        error = 0
        pbx = 0x7f5320025270
        callid = 0x0
        __PRETTY_FUNCTION__ = "__ast_pbx_run"
#14 0x000056055f20117d in pbx_thread (data=0x7f529c023258) at pbx.c:4661
        c = 0x7f529c023258
#15 0x000056055f291cdf in dummy_start (data=0x7f536000f1f0) at utils.c:1239
        __cancel_buf = {__cancel_jmp_buf = {{__cancel_jmp_buf = {0, 7617877189439841998, 139997105940478, 139997105940479, 139986324723456, 139999626680864, 7617877189481785038, 4256089606090974926}, __mask_was_saved = 0}}, __pad = {0x7f511b288ed0, 0x0, 0xe27ea02974561a00, 0x0}}
        __cancel_routine = 0x56055f0c56e1 <ast_unregister_thread>
        __cancel_arg = 0x7f511b289700
        __not_first_call = 0
        ret = 0x7f5434045a20
        a = {start_routine = 0x56055f201158 <pbx_thread>, data = 0x7f529c023258, name = 0x7f5360011e00 "pbx_thread", ' ' <repeats 11 times>, "started at [ 4687] pbx.c ast_pbx_start()"}
#16 0x00007f5443e09fa3 in start_thread (arg=<optimized out>) at pthread_create.c:486
        ret = <optimized out>
        pd = <optimized out>
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {139986324723456, -7558706345709607218, 139997105940478, 139997105940479, 139986324723456, 139999626680864, 7617877189454522062, 7616311966418527950}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#17 0x00007f544389d4cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
No locals.```

dkgroot, 2020-12-18:
Hi Niklas,
I have been working on this exact problem for the last three weeks and we finally had a break through earlier this week. 

***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/segfault origin/issues/segfault
git checkout issues/segfault
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

That should fix the issue. Please let me know if there are any side effects etc. It is a problem that seem to be very sporadic and takes the other company that reported it earlier about 5-10.000 calls to reproduce.

The problem was made aware to me by @tomgilabert.

Regards,
Diederik

dkgroot, 2020-12-19:
@pnlarsson 
In regard of the seriousness and testing already done, i have decided to merge the changes into the develop branch. This will be done a couple of minutes from now. So the issues/segfault branch will be discontinued. You will only have to pull latest develop revision.

commit 9b8357b6 (2020-12-19): Fix: Make sure d->protocol is always set
Enh: Check registrationstate before processing messages
Fix: Remove d->linesRegistered check from handle_line_number
Fix: prevent segfault in __find_joint_capabilities when remote native capabilities is 0x0 or not set
Fix: segfault when handling handle_line_number on a device without any lines registered (possible in cleaning state)
Fix: lock remotePeer channel when retrieving native capabilities
Fix: channel callid creation / assignment
Enh: Use ast_format_cap_empty instead of ast_format_cap_count

Issue reported by @tomgilabert and @pnlarsson
Fixes: # [... cut]
closed 2020-12-19 by dkgroot