chan-sccp/chan-sccp #551: "ICONV_CONST" build error on latest develop

billsimon, 2020-12-31:
Build fails on Debian 10 source install.

```
# make
Making all in src
make[1]: Entering directory '/usr/local/src/chan-sccp/src'
make[2]: Entering directory '/usr/local/src/chan-sccp/src'
  REV      revision.h
make[2]: Leaving directory '/usr/local/src/chan-sccp/src'
make[2]: Entering directory '/usr/local/src/chan-sccp/src'
  GEN      sccp_enum.h
  GEN      sccp_enum.c
make[2]: Leaving directory '/usr/local/src/chan-sccp/src'
make  all-recursive
make[2]: Entering directory '/usr/local/src/chan-sccp/src'
Making all in pbx_impl
make[3]: Entering directory '/usr/local/src/chan-sccp/src/pbx_impl'
  CC       pbx_impl.lo
  CCLD     libpbximpl.la
make[3]: Leaving directory '/usr/local/src/chan-sccp/src/pbx_impl'
Making all in pbx_impl/ast
make[3]: Entering directory '/usr/local/src/chan-sccp/src/pbx_impl/ast'
  CC       libast_la-ast.lo
In file included from ast.c:14:
../../../src/sccp_device.h:290:55: error: unknown type name ‘ICONV_CONST’; did you mean ‘OLDCONST’?
  void (*copyStr2Locale) (constDevicePtr d, char *dst, ICONV_CONST char *src, size_t dst_size);  /*!< copy string to device converted to locale if necessary */
                                                       ^~~~~~~~~~~
                                                       OLDCONST
make[3]: *** [Makefile:573: libast_la-ast.lo] Error 1
make[3]: Leaving directory '/usr/local/src/chan-sccp/src/pbx_impl/ast'
make[2]: *** [Makefile:1101: all-recursive] Error 1
make[2]: Leaving directory '/usr/local/src/chan-sccp/src'
make[1]: *** [Makefile:660: all] Error 2
make[1]: Leaving directory '/usr/local/src/chan-sccp/src'
make: *** [Makefile:605: all-recursive] Error 1
```

`configure` output is as follows:

```
Operating System : Linux
          Kernel : 4.19.0-10-amd64
    Architecture : x86_64
Asterisk Version : 11600 / 116
  Asterisk Repos : TRUNK
      Build user : root
         Version : 4.3.3_develop_r4a23494e
Compiler Version : 8
              CC : gcc
             CPP : gcc -E
         HOST_CC : gcc
          CFLAGS : -Og -Wall -I. -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -std=gnu11 -fgnu89-inline  -isystem /usr/include -DHAVE_ASTERISK -g3 -ggdb3 -pthread    
        CPPFLAGS : -U_FORTIFY_SOURCE  -isystem /usr/include -DHAVE_ASTERISK  -g3 -ggdb3     
         LDFLAGS :  -lssl -lcrypto -lpthread -lc -lltdl       -L/usr/lib        
        PBX_TYPE : Asterisk
      PBX_PREFIX : /usr
         PBX_ETC : /etc/asterisk
         PBX_LIB : /usr/lib
      PBX_VARLIB : /var/lib/asterisk
     PBX_DATADIR : /var/lib/asterisk
     PBX_INCLUDE : /usr/include/asterisk
Module Directory : /usr/lib/asterisk/modules
```

dkgroot, 2020-12-31:
@billsimon I will look into this tomorrow. Need to figure out how to reproduce the error. 
Do you have "gettext" (aka "iconv") installed include it's header files (/usr/include/iconv.h) ? Normally when the configure script finds the gettext/iconv installed files, it will generate an entry in src/config.h containing something like:
`#define ICONV_CONST const`
or
`#define ICONV_CONST`
depending on if it's needed or not.

billsimon, 2020-12-31:
It's my mistake; I did not have gettext installed.

configure had shown this, which scrolled away and did not cause the script to abort:

./configure: line 18232: AM_ICONV: command not found
checking for main in -liconv... no

billsimon, 2020-12-31:
[config.log](https://github.com/chan-sccp/chan-sccp/files/5758277/config.log)

dkgroot, 2020-12-31:
Ooh ok so issue "fixed", but i should still make sure the configure script aborts when it detects the missing gettext issue.
Good luck setting up and using chan-sccp. I will leave the issue open until tomorrow so i can commit a fix for the configure script.

commit 066a14a1 (2021-01-01): Fix: Extend aclocal.m4 iconv detection
Fix: continue if iconv/gettext not supported on the local platform

Fixes: #551
closed 2021-01-01 by dkgroot