chan-sccp/chan-sccp #553: No audio on both sides with devices on different subnets

mpaland, 2021-01-05:
This is a bug I observed after upgrading from asterisk 13 to 16. I don't use this scenario very often, so this bug is not so hard for me, but it may result in other RTP problems somewhere.
Two different subnets are involved here, so localnet is set to `localnet = 192.168.0.0/255.255.0.0`, but other settings didn't change the result.

The setup is:
```
192.168.1.210 ------- VPN ------ 192.168.2.1 ------ 192.168.2.48
<23> SEP00270DBD6712             Asterisk           <12> SEPE80462129313
```
**Reproduction Steps:**
1. <23> is calling <12>
2. <12 > answers the call

**chan-sccp version:**
066a14a1d081fdba52cb30f87a9c760d336e8ae9 from 2021-01-01

**asterisk version:**
16.15.1

**Expected behavior:**
RTP working in both directions

**Observed behavior:**
Both partys can't hear each other.

Calling outside parties (via PJSIP) or calling test services on asterisk itself (echo test) works fine from every device/subnet.
Calling from <16> to <12> which are  both on the same subnet (192.168.2.0) works fine.

**Addition / Optional Information:**
Problem started after upgrading from asterisk 13 to 16, sccp.conf and VPN config remained the same.
I attached a log file in which you can see, **that no RTP packets** are sent from asterisk (only "Got  RTP packet from" messages), which is strange. 
At least there are no "Send RTP packet to" in the asterisk log (haven't used tcpdump to verify).
But I can see:
`[Jan  5 17:34:36] NOTICE[22455]: sccp_rtp.c:392 sccp_rtp_set_phone:     -- SEP00270DBD6712: Tell PBX   to send RTP/UDP media from 192.168.2.1:35656 to 192.168.1.210:16956 (NoNat)
`
[23 to 12 log.txt](https://github.com/chan-sccp/chan-sccp/files/5771959/23.to.12.log.txt)

sccp.conf:
```
bindaddr = 192.168.2.1
port = 2000
deny=0.0.0.0/0.0.0.0
permit = internal
localnet = 192.168.0.0/255.255.0.0
externip = 0.0.0.0

[defaultdevice](!)
type = device
park = off
transfer = on
cfwdall = on
cfwdbusy = on
cfwdnoanswer = on
dndFeature = on
dnd = silent
private = on
directed_pickup = on
directed_pickup_context = "default"
mwioncall = off
deny=0.0.0.0/0.0.0.0
permit=internal
directrtp = off
earlyrtp = progress
keepalive = 60
```
directrtp and NAT is off for all devices.

Perhaps it's just a SCCP misconfiguration, but I don't suspect the VPN or firewall here.
Perhaps you have a quick idea.

dkgroot, 2021-01-05:
@mpaland I will review the log tomorrow, to see what is going on. Sorry i have no quick idea. I am guessing we are using a wrong RTP source or destination somewhere in the flow. Thanks for reporting the issue and providing a detailed log.

[dkgroot@devserver:/data … rces/chan-sccp/src] develop(+1/-1)+* 😊> grep "192.168" mpaland.log
```
[Jan  5 17:34:19] NOTICE[22455]: ast116.c:2479 sccp_astwrap_createRtpInstance:     -- SCCP/23-0000001E: rtp server instance created at 192.168.2.1:35656
[Jan  5 17:34:19] NOTICE[22455]: sccp_rtp.c:161 sccp_rtp_createServer:     -- SCCP/23-0000001E: (sccp_rtp_createRtpServer) setting new phone Audio RTP destination to: 192.168.2.1:35656, family:IPv4, map                                            ped: False
[Jan  5 17:34:36] NOTICE[22496]: ast116.c:2479 sccp_astwrap_createRtpInstance:     -- SCCP/12-0000001F: rtp server instance created at 192.168.2.1:36518
[Jan  5 17:34:36] NOTICE[22496]: sccp_rtp.c:161 sccp_rtp_createServer:     -- SCCP/12-0000001F: (sccp_rtp_createRtpServer) setting new phone Audio RTP destination to: 192.168.2.1:36518, family:IPv4, map                                            ped: False
[Jan  5 17:34:36] NOTICE[22496]: sccp_actions.c:3420 handle_openReceiveChannelAck:     -- SEPE80462129313: Got OpenChannel ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.2.48:32606', Type:I                                            ndirect RTP, PassThruPartyId:4294967264, CallID:31
       > 0x77fed40185b0 -- Strict RTP learning after remote address set to: 192.168.2.48:32606
[Jan  5 17:34:36] NOTICE[22496]: ast116.c:2622 sccp_astwrap_setPhoneRTPAddress:     -- SCCP: (setPhoneRTPAddress) Update PBX to send RTP/UDP media to '192.168.2.48:32606' (new remote) (NAT: no)
[Jan  5 17:34:36] NOTICE[22496]: sccp_rtp.c:392 sccp_rtp_set_phone:     -- SEPE80462129313: Tell PBX   to send RTP/UDP media from 192.168.2.1:36518 to 192.168.2.48:32606 (NoNat)
[Jan  5 17:34:36] NOTICE[22496]: sccp_channel.c:1014 sccp_channel_startMediaTransmission:     -- SEPE80462129313: Tell Phone to send RTP/UDP media from 192.168.2.48:32606 to 192.168.2.1:36518 (NAT: (Aut                                            o)Off)
[Jan  5 17:34:36] NOTICE[22496]: sccp_actions.c:3484 handle_startMediaTransmissionAck:     -- SEPE80462129313: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.2.48                                            :32606', Type:Indirect RTP, PassThruPartyId:4294967264, CallID:31, CallID1:31
       > 0x77fed40185b0 -- Strict RTP switching to RTP target address 192.168.2.48:32606 as source
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016446, ts 1925362451, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016447, ts 1925362771, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016448, ts 1925363091, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016449, ts 1925363411, len 000640)
[Jan  5 17:34:36] NOTICE[22455]: sccp_actions.c:3420 handle_openReceiveChannelAck:     -- SEP00270DBD6712: Got OpenChannel ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.1.210:16956', Type:                                            Indirect RTP, PassThruPartyId:4294967265, CallID:30
       > 0x77fecc01b950 -- Strict RTP learning after remote address set to: 192.168.1.210:16956
[Jan  5 17:34:36] NOTICE[22455]: ast116.c:2622 sccp_astwrap_setPhoneRTPAddress:     -- SCCP: (setPhoneRTPAddress) Update PBX to send RTP/UDP media to '192.168.1.210:16956' (new remote) (NAT: no)
[Jan  5 17:34:36] NOTICE[22455]: sccp_rtp.c:392 sccp_rtp_set_phone:     -- SEP00270DBD6712: Tell PBX   to send RTP/UDP media from 192.168.2.1:35656 to 192.168.1.210:16956 (NoNat)
[Jan  5 17:34:36] NOTICE[22455]: sccp_channel.c:1014 sccp_channel_startMediaTransmission:     -- SEP00270DBD6712: Tell Phone to send RTP/UDP media from 192.168.1.210:16956 to 192.168.2.1:35656 (NAT: (Au                                            to)Off)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016450, ts 1925363731, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016451, ts 1925364051, len 000640)
[Jan  5 17:34:36] NOTICE[22455]: sccp_actions.c:3484 handle_startMediaTransmissionAck:     -- SEP00270DBD6712: Got startMediaTransmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.1.21                                            0:16956', Type:Indirect RTP, PassThruPartyId:4294967265, CallID:30, CallID1:30
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016452, ts 1925364371, len 000640)
       > 0x77fecc01b950 -- Strict RTP switching to RTP target address 192.168.1.210:16956 as source
Got  RTP packet from    192.168.1.210:16956 (type 25, seq 006168, ts 3961341810, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016453, ts 1925364691, len 000640)
Got  RTP packet from    192.168.1.210:16956 (type 25, seq 006169, ts 3961342130, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016454, ts 1925365011, len 000640)
Got  RTP packet from    192.168.1.210:16956 (type 25, seq 006170, ts 3961342450, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016455, ts 1925365331, len 000640)
Got  RTP packet from    192.168.1.210:16956 (type 25, seq 006171, ts 3961342770, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016456, ts 1925365651, len 000640)
Got  RTP packet from    192.168.1.210:16956 (type 25, seq 006172, ts 3961343090, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016457, ts 1925365971, len 000640)
Got  RTP packet from    192.168.1.210:16956 (type 25, seq 006173, ts 3961343410, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016458, ts 1925366291, len 000640)
Got  RTP packet from    192.168.1.210:16956 (type 25, seq 006174, ts 3961343730, len 000640)
Got  RTP packet from    192.168.2.48:32606 (type 25, seq 016459, ts 1925366611, len 000640)
....
```

dkgroot, 2021-01-05:
Ooh you already figured out the cause. 

chan-sccp is creating the rtp instance on the wrong interface. It should have created something like 192.168.1.210:36518 instead. 

A quick fix on your server would if you would allow traffic forwarding between the two local interfaces in sysctl.conf: `net.ipv4.ip_forward = 1`.

But chan-sccp has to be fixed to do the right thing. I will try to figure this out somewhere tomorrow.

mpaland, 2021-01-06:
`net.ipv4.ip_forward` was/is already set, as the system is mainly a router, but a good idea anyway.
The whole issue is a bit strange, I attach a second log, where <23> is calling an outside party via PJSIP.
This is working as expected. [23 to ext party log.txt](https://github.com/chan-sccp/chan-sccp/files/5775842/23.to.ext.party.log.txt) So the tunnel interface seems not to be the problem here.
Unfortunately I don't have a test machine on this network.

dkgroot, 2021-01-06:
I assumed as much, and net.ipv4.ip_forward should of course not be a requirement for things to work. What is a little strange is that asterisk is receiving packets, but not relaying them. Even if the rtp_instance is connected to the wrong interface.
Steps to fix:
1. figure out a way to reproduce the effect / network situation. Might be as simple as setting up two ip-addresses on one interface and connecting CIPC to both of them.
1. fix the ip-address used for the asterisk server side rtp_instance
2. figure out why asterisk-13 is handling this differently from asterisk-16 and up (if needed)

Note: Possible Temp solution: If net.ipv4.ip_forward is present, you might be able to force chan-sccp to only use one of the interfaces and rely on the os to the the ip forwarding (ie:make it only use 192.168.2.x for example).

mpaland, 2021-01-07:
Seems to be codec selection problem, somehow. I'm still testing.

mpaland, 2021-01-08:
Adding codec setup in the [general] section fixed the problem.
```
disallow = all
allow = g722,alaw,g729,ilbc,ulaw
```
It seems, that without these lines alaw is taken as default codec, which leads to the rtp problem.
I leave this issue open to investigate this a little further.

dkgroot, 2021-01-08:
Hi Marco,

Some sections and entries have defaults and they do have an inheritance tree, to propagate defaults and or settings. If i understand correctly you had no codecs set in the [general] section so that would default to `allow=ulaw, alaw` i presume. I take it you did not have disallow/allow on the device level and only set it on the line level (correct).

I do not understand how that could cause:
- wideband/slin16 to be used directly on the device  (which can be a bit of a problematic codec)
- no translations was being performed if necessary
- packets were received but never transmitted by asterisk

We should still fix the rtp_instance ip-address issue mentioned before.

mpaland, 2021-01-09:
You are right, I had no codecs in the [general] or in the [line] section, just in the device section template.
I assumed, the device would override the [general] level.
Okay, this was wrong an therefore an invalid setup, my mistake.

But I don't understand, like you, why one way RTP happened.

Anyway, for the following test, I'm using this (valid) config:
```
[general]
disallow = all
allow = g722,alaw,g729,ilbc,ulaw

[defaultline](!)
disallow = all
allow = g722,alaw,g729,ilbc,ulaw

[21](defaultline)
disallow = all
allow = ilbc
```

Line 21 status gives only the ilbc codec, Fine.

Test is a call from <21> to the system asterisk echo test, here on *43. ilbc codec is installed in asterisk, so any transcoding concerning ilbc should work.
Problem is, that no RTP flow takes place, At the end of the log, the call is kind of established, but no audio/RTP.
When I put the handset on hook, nothing happens, I need to press "End call" to hang up. This is not normal.
@dkgroot , are you able to observe the same behavior, if you take the ilbc line codec as the only one like above?

I think, the no audio RTP problem is not a problem of the tunnel or the interfaces, as a snom SIP phone is working fine over the tunnel with pjsip.
Perhaps it's related to the initial wrong config and a codec mismatch by chance between the two phones <12> and <23>.

```
ipfire*CLI>
    -- SEP002699F01CFA: Accessory 'Handset' is 'Off Hook'
    -- SEP002699F01CFA: Got Softkey: NewCall (2) line=0 callid=0
    -- SEP002699F01CFA: Looking for line with instance 1.
    -- SCCP: Found line 21
    -- SEP002699F01CFA: Getting the active channel on device.
    -- SEP002699F01CFA: No active channel on device.
SEP002699F01CFA: (getEmptyChannel) on line:21, maybe_c:
    -- SEP002699F01CFA: Getting the active channel on device.
    -- SEP002699F01CFA: No active channel on device.
 SCCP: Adding channel 7 to line 21
    -- SEP002699F01CFA: Set the active channel 7 on device
    -- SEP002699F01CFA: Set the active line 21
    -- 1021: New channel number: 7 on line 21
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 21
    -- SCCP:             cid_num: 21
    -- SCCP:            cid_name: Marco
    -- SCCP:         accountcode: (null)
    -- SCCP:               exten:
    -- SCCP:             context: from-internal
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/21-00000007
    -- SCCP: combined audio caps: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
    -- SCCP: reduced audio prefs: [ilbc]
    -- SCCP: combined video caps: []
    -- SCCP: reduced video prefs: []
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/21-00000007
    -- allocPBXChannel: using nativeformats=(ilbc) fmt=ilbc
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
  == SCCP/21-00000007: update rtpmap: format:CISCO-DTMF, payload:101, mime:audio, rate:0
    -- 1021: Allocated asterisk channel SCCP/21-00000007
    -- SEP002699F01CFA: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/21-00000007, lineInstance:1 (previous channelstate:DOWN)
    -- SEP002699F01CFA: Send speaker mode 'on'
    -- SEP002699F01CFA: Send and Set the call state offhook(1) on call 7 (visibility:default)
    -- SEP002699F01CFA: Send activate call plane on line 1
    -- SEP002699F01CFA: ( sccp_indicate.c:90:__sccp_indicate ) sccp_dev_displayprompt '▒ ' for line 1 (10)
    -- SEP002699F01CFA: Display prompt on line 1, callid 7, timeout 10
[... 29 lines cut ...]
    -- SEP002699F01CFA: Send activate call plane on line 1
    -- SEP002699F01CFA: ( sccp_device.c:2946:sccp_device_indicate_connected ) sccp_dev_displayprompt '▒' for line 1 (10)
    -- SEP002699F01CFA: Display prompt on line 1, callid 7, timeout 10
    -- SCCP/21-00000007, Could not find a common prefered codec (yet), using ilbc (86)
    -- SCCP/21-00000007: (findBestJoint) returning prefered codec ilbc (86)
    -- SCCP/21-00000007: (setNativeAudioFormats) (ilbc)
    -- SEP002699F01CFA - SCCP/21-00000007: (recalculateAudioCodecformat)
        channel capabilities: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
        channel preferences [ilbc]
        line preferences: [ilbc]
        remote caps: []
        Format:ilbc
    -- SCCP/21-00000007, OpenReceiveChannel with format G.729B Low/ILBC, payload 86, echocancel: YES, passthrupartyid: 4294967288, pbx_channel_name: SCCP/21-00000007
    -- SEP002699F01CFA: Request 'Open Receive Channel' to device Pending
    -- SEP002699F01CFA: Stop tone on line 1 with callid 7
    -- SEP002699F01CFA: Set softkeyset to CONNTRANS(5) on line 1  and call 7
    -- SEP002699F01CFA: validKeyMask 4294967295
    -- SEP002699F01CFA: Finish to indicate channel state:CONNECTED on call:SCCP/21-00000007, lineInstance:1. New channel state:CONNECTED
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SCCP/21-00000007: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
 SEP002699F01CFA: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/21-00000007 (rtp_instance:yes, reception.state:1ion.state:0/G.729B Low/ILBC)
    -- Executing [*43@from-internal:2] Wait("SCCP/21-00000007", "1") in new stack
    -- Executing [*43@from-internal:3] Playback("SCCP/21-00000007", "demo-echotest") in new stack
    -- <SCCP/21-00000007> Playing 'demo-echotest.slin' (language 'en')
```

mpaland, 2021-01-10:
Seems to be an ilbc codec related problem. Other codecs work fine.

dkgroot, 2021-01-10:
Hi Marco,
I never really used/test with ilbc. I need to make a little time, to research this. I did have less this weekend then i hoped.

mpaland, 2021-01-10:
Hi Diederik,
no rush. I used ilbc a lot over VPN connections, cause it has a good sound quality while having low bandwidth, a fairly goog jitter / packet loss tolerance and is supported out of the box on most Cisco phones.
I have no G729 license (therefore I removed it from the codec list), so that g722, alaw, ulaw and ilbc remained. To test transcoding, I locked the line to ilbc - which is not working as it seems.

Strange is the last line, which contains G.729B, but it isn't in any list.
```
    -- SEP00270DBD6712 - SCCP/23-00000014: (recalculateAudioCodecformat)
        channel capabilities: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
        channel preferences [ilbc]
        line preferences: [ilbc]
        remote caps: []
        Format:ilbc
    -- SCCP/23-00000014, OpenReceiveChannel with format G.729B Low/ILBC, payload 86, echocancel: YES, passthrupartyid: 4294967275, pbx_channel_name: SCCP/23-00000014
    -- SEP00270DBD6712: Request 'Open Receive Channel' to device Pending
    -- SEP00270DBD6712: Stop tone on line 1 with callid 20
    -- SEP00270DBD6712: Set softkeyset to CONNTRANS(5) on line 1  and call 20
    -- SEP00270DBD6712: validKeyMask 4294967295
    -- SEP00270DBD6712: Finish to indicate channel state:CONNECTED on call:SCCP/23-00000014, lineInstance:1. New channel state:CONNECTED
    -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    -- SCCP/23-00000014: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
 SEP00270DBD6712: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/23-00000014 (rtp_instance:yes, reception.state:1/G.729B Low/ILBC, transmission.state:0/G.729B Low/ILBC)
```

mpaland, 2021-01-11:
Interesting, I didn't know, how codecs are mapped - my first guess would have been by a fixed codec code.
I don't understand why Cisco calls iLBC "G.729B Low/ILBC", as iLBC is not related to G729.
Anyway, I see the problem. Can you substitute Cisco's name by "ilbc" before processing it any further in asterisk?
Perhaps we should open a new issue for that.

I still think this issue here is/was the result of an invalid config plus some "unlucky" side effects.

dkgroot, 2021-01-11:
> Interesting, I didn't know, how codecs are mapped - my first guess would have been by a fixed codec code.

It's complicated :-)

> I don't understand why Cisco calls iLBC "G.729B Low/ILBC", as iLBC is not related to G729.

as you know we don't have any of the cisco source code, so we don't really know what they call it. Most off these values are based on somewhat educated guesses, phone debug output, asterisk output etc. Maybe because the ILBC codec was added later and inserted in the middle of an existing enum (first free position) and we didn't know exactly what it was yet i guessed that it must be related.
I just have updated/change the enum name and debug output so it will say ILBC instead. I have only updated the name. The mapping still seems to be going wrong though when testing with a sip->sccp call.

dkgroot, 2021-01-11:
@mpaland Did you use ilbc before without any trouble. If so i might have a look at that revision + asterisk combination to see what changed.

mpaland, 2021-01-11:
Jeeez, I see.
Unfortunately I haven't used ilbc that explicit like in the test, it was always in the list among the others, so I don't know what coded was finally chosen, but I guess it wasn't ilbc.
I used it explicit a few years ago with my old UC520 but not since then. I just had the idea to test and use it again, therefore the test.
I didn't intend to open up a huge problem/programming session here. 😞

dkgroot, 2021-01-11:
Hmm ok. I can run a short asterisk-13 test tomorrow. Maybe ilbc never really worked in combination with chan-sccp and asterisk. It's not a supported codec in chan_skinny (so cannot cheat a look how they do it).

I would like ilbc to work as it sounds like a good tradeoff between size and quality. I just never looked into it that much. I have a couple of idea, but i just tested sip (jami)-sip (ikega) and the audio was messed up as well, so maybe choosing ekiga was not a good idea. Jami doesn't support ilbc. Ilbc support on softclients seems to be somewhat sparse. I would like to make sure sip-sip works first before slowly introducing chan-sccp back in the mix.

dkgroot, 2021-01-11:
I besides the codec & payload it looks like cisco also sets certain parameters for some of the codecs (like max bitrate / frames size / etc). I currently don't know what they are but someone might be able to capture some tcpdumps.

I also set the payload number to 86 for ilbc on the asterisk side to use the cisco number, instead of the asterisk value for the payload which is 97. I am not sure if that has any effect, could not test with only one phone :-).

mpaland, 2021-01-12:
I tested branch/553 and it's not working, no RTP flow.
Below is the trace from <23> to *43, device to device is the same.
```
    -- SEP00270DBD6712: Accessory 'Handset' is 'Off Hook'
    -- SEP00270DBD6712: Got Softkey: NewCall (2) line=0 callid=0
    -- SEP00270DBD6712: Looking for line with instance 1.
    -- SCCP: Found line 23
    -- SEP00270DBD6712: Getting the active channel on device.
    -- SEP00270DBD6712: No active channel on device.
SEP00270DBD6712: (getEmptyChannel) on line:23, maybe_c:
    -- SEP00270DBD6712: Getting the active channel on device.
    -- SEP00270DBD6712: No active channel on device.
 SCCP: Adding channel 10 to line 23
    -- SEP00270DBD6712: Set the active channel 10 on device
    -- SEP00270DBD6712: Set the active line 23
    -- 1021: New channel number: 10 on line 23
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 23
    -- SCCP:             cid_num: 23
    -- SCCP:            cid_name: Marco
    -- SCCP:         accountcode: (null)
    -- SCCP:               exten:
    -- SCCP:             context: from-internal
    -- SCCP:            amaflags: 0
    -- SCCP:           chan/call: SCCP/23-0000000A
    -- SCCP: combined audio caps: [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
    -- SCCP: reduced audio prefs: [ilbc]
    -- SCCP: combined video caps: []
    -- SCCP: reduced video prefs: []
    -- SCCP: (allocPBXChannel) Create New Channel with name: SCCP/23-0000000A
    -- allocPBXChannel: using nativeformats=(ilbc) fmt=ilbc
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP CoS mark 6
  == SCCP/23-0000000A: update rtpmap: format:CISCO-DTMF, payload:101, mime:audio, rate:0
    -- 1021: Allocated asterisk channel SCCP/23-0000000A
    -- SEP00270DBD6712: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/23-0000000A, lineInstance:1 (previous channelstate:DOWN)
    -- SEP00270DBD6712: Send speaker mode 'on'
    -- SEP00270DBD6712: Send and Set the call state offhook(1) on call 10 (visibility:default)
    -- SEP00270DBD6712: Send activate call plane on line 1
    -- SEP00270DBD6712: ( sccp_indicate.c:90:__sccp_indicate ) sccp_dev_displayprompt '▒ ' for line 1 (10)
    -- SEP00270DBD6712: Display prompt on line 1, callid 10, timeout 10
    -- SEP00270DBD6712: Sending tone Inside Dial Tone (33) on line 1 with callid 10 (direction: User)
    -- SEP00270DBD6712: Set softkeyset to OFFHOOK(4) on line 1  and call 10
    -- SEP00270DBD6712: validKeyMask 4294967295
    -- SEP00270DBD6712: Finish to indicate channel state:OFFHOOK on call:SCCP/23-0000000A, lineInstance:1. New channel state:OFFHOOK
    -- SCCP/23-0000000A: schedule digittimeout 16
       > SEP00270DBD6712: keepalive:120, keepaliveinterval:54
    -- SEP00270DBD6712: EnblocCall, party: *43, lineInstance 1
    -- SEP00270DBD6712: Getting the active channel on device.
    -- SEP00270DBD6712: (sccp_pbx_softswitch) New call on line 23
    -- SEP00270DBD6712: Indicate SCCP new state:DIALING, current channel state:OFFHOOK on call:SCCP/23-0000000A, lineInstance:1 (previous channelstate:DOWN)
    -- SEP00270DBD6712: Stop tone on line 1 with callid 10
    -- SEP00270DBD6712: Send the dialed number:*43, callid:10, lineInstance:1
    -- SEP00270DBD6712: Send and Set the call state proceed(12) on call 10 (visibility:default)
    -- SEP00270DBD6712: Finish to indicate channel state:DIALING on call:SCCP/23-0000000A, lineInstance:1. New channel state:DIALING
    -- +- pbx extension matcher (*43            ): ---+
    -- |ignore     |exists     |can match  |match more|
    -- | no        |yes        |yes        | no       |
    -- +----------------------------------------------+
 SEP00270DBD6712: (sccp_pbx_softswitch) channel SCCP/23-0000000A is dialing number *43
 SEP00270DBD6712: (sccp_pbx_softswitch) finished
 SCCP: (sccp_pbx_softswitch) exit
    -- Executing [*43@from-internal:1] Answer("SCCP/23-0000000A", "") in new stack
    -- SCCP/23-0000000A: Remote has answered the call.
    -- SEP00270DBD6712: (sccp_pbx_remote_answer) Outgoing call SCCP/23-0000000A being answered by remote party
    -- SEP00270DBD6712: Indicate SCCP new state:CONNECTED, current channel state:DIALING on call:SCCP/23-0000000A, lineInstance:1 (previous channelstate:OFFHOOK)
    -- SEP00270DBD6712: Send ringer mode Off(1) on device
    -- SEP00270DBD6712: Send speaker mode 'on'
    -- SEP00270DBD6712: Stop tone on line 1 with callid 10
    -- SEP00270DBD6712: Send and Set the call state connected(5) on call 10 (visibility:default)
    -- SEP00270DBD6712: Send activate call plane on line 1
[... 53 lines cut ...]
    -- SEP00270DBD6712: validKeyMask 4294967295
       > SEP00270DBD6712: clear message stack 3
    -- SEP00270DBD6712: Send speaker mode 'off'
    -- SEP00270DBD6712: Send ringer mode Off(1) on device
    -- SEP00270DBD6712: Finish to indicate channel state:ONHOOK on call:SCCP/23-0000000A, lineInstance:1. New channel state:DOWN
    -- SEP00270DBD6712: Device is Requesting Connections Statistics And Clear
    -- SCCP: Cleaning channel SCCP/23-0000000A
    -- SEP00270DBD6712: Set the active channel 0 on device
    -- SEP00270DBD6712: Set the active line (NULL)
    -- SEP00270DBD6712: Looking for selected channel (10)
    -- SEP00270DBD6712: Set the active line (NULL)
    -- SEP00270DBD6712: Clear the status prompt on line 0 and callid 0
    -- SEP00270DBD6712: ( sccp_device.c:2142:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '9' for line 0 (0)
[... cut ...]
         Last Call        : CallID: 10 Packets sent: 551 rcvd: 550 lost: 0 jitter: 1 latency: 0
         Last Quality     : MLQK=4.4438;MLQKav=4.4250;MLQKmn=4.3874;MLQKmx=4.4438;MLQKvr=0.95|ICR=0.0032;CCR=0.0022;ICRmx=0.0034|CS=2;SCS=0
         Mean Statistics  : #Calls: 6 Packets sent: 551 rcvd: 550 lost: 0 jitter: 1 latency: 0
         Mean Quality     : MLQK=4.4438;MLQKav=4.4250;MLQKmn=4.3874;MLQKmx=4.4438;MLQKvr=0.95|ICR=0.0032;CCR=0.0022;ICRmx=0.0034|CS=2;SCS=0
       ]
       > SEP00270DBD6712: keepalive:72, keepaliveinterval:54
```

dkgroot, 2021-01-12:
Hi Marco,
I was not expecting it to work / change anything. It's a work in progress. Sorry about making your test it. I think we might need some tcpdump from a work call manager to see how cisco sets up ilbc on the phones.

mpaland, 2021-01-12:
Ah, I see. Okay, that's no problem, I grab my old Cisco UC520, hope that any Flashcard is still working, create an iLBC call and record it.
By the way, if you have any usage for further analyzing this PBX, I can donate it to you.

dkgroot, 2021-01-12:
Hi Marco, Thanks for gathering that capture information. I am not really looking to collect more hardware at the moment :-) Also to keep keep to the reverse engineering state, it might be best not to have direct access to any cisco hardware like that.

mpaland, 2021-01-18:
I'm really sorry, my old UC520 doesn't exist anymore, I wasn't able to find it, probably it got trashed in the last cleanup. 😞 
Perhaps someone else with access to a call manager could provide the necessary trace, we should ask on the mailing list.

dkgroot, 2021-01-20:
I guess we will have to ask around. Sorry to hear you misplaced your UC520