chan-sccp/chan-sccp #554: Phone freezes after it receives 2-way auto-pickup call from another phone

mwllgr, 2021-01-05:
I observed this behaviour several times now. A Cisco 7970 just kinda "freezes" and I have to restart my Asterisk server to get everything working again when I call it using auto pickup. I currently don't know if it's only because of the aa2w and will get some more debug information when the problem occurs again.

**Reproduction Steps:**

1. _203_ calls _2202_ (which is _202_ with `aa=2w`)

**chan-sccp version:** 4.3.3 develop - 066a14a

**asterisk version:** 13.14.1~dfsg-2+deb9u4

**Expected behavior:**
Phone _202_ should pick up the call from _203_.

**Observed behavior:**

_202_'s MWI lamp is constantly blinking. The phone shows the name number for _203_.  
Call duration is displayed, even though the only to softkeys available are _Answer_ and _Hangup_ (_Annehmen_ and _Auflegen_ in German). The line key for 202 is constantly green, same for the speaker button.

_203_ only shows an empty bar with the number _2202_ and does not respond to the _Hangup_ softkey.

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes (around 3 weeks from now)
- Problem can be reliably reproduced, doesn't happen randomly: No
- Platform: Raspberry Pi 3 Model B+ (1 GB, 1.3)
- Log Files
  - [202 debug log, normal log level](https://p.kll.li/?53ff814629d1cea8#AUTHY8GPVdrn8dRTF5esD4JxKBcXAGNoJ58sgo7wGWxz)
  - [203 debug log, normal log level](https://p.kll.li/?c875c7d751bf6d6d#EBUUKeHRHF1pDAydbaBxmu2eGxeQutDdEz3WzWv1a5bk)
- Comments
  - 202: Cisco CP-7970G, jar70sccp.9-4-2ES9.sbn
  - 203: Cisco CP-7911G, jar11sccp.9-4-2ES9.sbn

dkgroot, 2021-01-05:
@mwllgr Assuming the state is not easily reproducable it might take a while for the issue to re-occur and it might take a while for us to figure out the cause. Please report back when the state occurs again. If possible do not reset the device immediately so we may collect a little extra data.

mwllgr, 2021-01-05:
The bug just reappeared. This time I am calling from another 7970 with extension 201. Here's an Asterisk log with `sccp debug core, channel, pbx, action`: [https://p.kll.li/?f44e0837a369be53#EFho1UqWG9NymT98gXesU1GgJsSmrRgx4upMjza6z35d](https://p.kll.li/?f44e0837a369be53#EFho1UqWG9NymT98gXesU1GgJsSmrRgx4upMjza6z35d)

This time, the softkeys on 203 just disappeared after some time and I wasn't able to press one of them.
Pressing Hangup on 201 results in the following log entries:

```
    -- SEP00192FEEEEEE: Got Softkey: EndCall (9) line=1 callid=12
    -- SEP00192FEEEEEE: Getting the active channel on device.
    -- SCCP/201-0000000C: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:-1) (hangup_id:-1) (cfwd_noanswer_id:-1)
  == SEP00192FEEEEEE: Ending call SCCP/201-0000000C (state:DIALING)
    -- SEP00192FEEEEEE: Sending hangupRequest to Call SCCP/201-0000000C (state: DIALING)
```

Nothing happens besides that, the busy lamp for line 202 (as visible on 201) is constantly in "ringing" state. Phone 201 is now stuck in the following screen (notice the second line button indicates the ringing phone 202):

[image]
Translations: Wahlw./Redial, Auflegen/Hangup, Rufuml./CfwdAll, an/to

Two minutes after I took the screenshot (00:40) the phone 201 looks just like 202, only the empty screen is visible:

[image]

Also, I got some more log entries now. The last digit of the IP address is always the last digit of the extension. (10.3.3.7X = 20X)
```
    -- SCCP: Accepted Client Connection from 10.3.3.72:51184
  == Alarm Type: LastOutOfServiceInformation
  == Reason Enum: 15
  == Last Event Sent: 39:Unregister
  == Last Event Received: 325:DisplayPromptStatusV2
 SEP0019563DDDDD: is registering, Instance: 0, UserId: 0, Type: Cisco 7970 (30006), Version: 22 (loadinfo 'term70.default')
[2021-01-06 00:43:23] WARNING[6428]: sccp_session.c:1311 sccp_session_check_crossdevice: Session(0x74102380) and Device Session(0x741020d8) are out of sync.
[2021-01-06 00:43:23] WARNING[6428]: sccp_actions.c:841 handle_register: SEP0019563DDDDD: Cleaning previous session, come back later, state:OK
  == 10.3.3.7:2000: Session 0x741022c4 needs to be closed!
```

dkgroot, 2021-01-05:
@mwllgr I will look into this tomorrow or the day after, to see what i can find. The softkey translations are not really need btw, i do speak and read German just fine, but thanks anyway :-)

dkgroot, 2021-01-06:
As this is an auto answer call, it might be a simple matter of an unreleased refcount that is incremented on the first call and then causing trouble on subsequent calls. I think that should be easy to check and hopefully equally easy to fix.

dkgroot, 2021-01-06:
I see this in your error report
```
  == SEP00192FEEEEEE: Ending call SCCP/201-0000000C (state:DIALING)
    -- SEP00192FEEEEEE: Sending hangupRequest to Call SCCP/201-0000000C (state: DIALING)
```
And "DIALING" is clearly the wrong state, because the call was auto answered and should be in the CONNECTED state instead. I am having a little trouble reproducing the issue though.

Zw1d, 2022-03-09:
I started creating a new issue, but found this one, let me get this in here:
Hello everyone (and Diederik!)
I have recently updated SCCP from 33da09010fa8b9f139fa302df5ac7776a297b0d7 to c1533a6b5ab5aa77fd55f432884a56d1c9601588 (quite a gap - ~2 years), because I was getting some dropped calls. That issue seems to be resolved, however just found out a new, serious one.
I have been using a custom extension to get auto answer mechanism on some phones, which effectively called this:
SCCP/4010/aa=2wb/ringer=silent
Now if try that, whole PBX hangs, and I have to killall asterisk to get it working again. Phones hang as well. Could you please tell me what logs to provide to help fix this?