chan-sccp/chan-sccp #555: Problem with configuration template

mpaland, 2021-01-07:
I'm observing a (IMHO confusing) behavior in the configuration when using a template.

**Reproduction Steps:**
Following template (codec selection is relevant) ;
```
[defaultdevice](!)
type = device
park = off
transfer = on
cfwdall = on
cfwdbusy = on
cfwdnoanswer = on
dndFeature = on
dnd = silent
private = on
directed_pickup = on
directed_pickup_context = "default"
mwioncall = off
deny=0.0.0.0/0.0.0.0
permit=internal
directrtp = off
earlyrtp = progress
keepalive = 60
disallow = all
allow = g722,alaw,g729,ilbc,ulaw

[SEP00270DBD6712](defaultdevice)
description = 23
devicetype = 7975
button = line, 23, default                      ; Assign Line 23 to device and use this as default line
button = speeddial, Amt,, 80@ext-hints          ; Amt
softkeyset = softkeyset1
disallow = all
allow = ilbc
```

Looking at the device:
```
ipfire*CLI> sccp show device SEP00270DBD6712

--- SCCP channel driver device settings ----------------------------------------------------------------------------------
MAC-Address                                    : SEP00270DBD6712
Protocol Version                               : Supported '22', In Use '22'
Protocol In Use                                : SCCP Version 22
Device Features                                : 0x7285,000000000111001010000101
Tokenstate                                     : None
Keepalive                                      : 60
Registration state                             : OK
State                                          : On Hook
Addons                                         :
MWI state                                      : off (0/7)
MWI light-type                                 : On
MWI During call                                : turn off
Description                                    : 23 (9525237)
Config Phone Type                              : 7975
Skinny Phone Type                              : Cisco 7975(437)
Softkey support                                : yes
Softkeyset                                     : softkeyset1 => softkeyset1 (0x77fec42b19a0)
BTemplate support                              : yes
linesRegistered                                : yes
Image Version                                  : SCCP75.9-4-2SR3-1S
Timezone Offset                                : 0
Audio Capabilities                             : [slin16, g722/64k, ulaw/64k, alaw/64k, g722/56k, g722/48k, ilbc, g729b, g729ab, g729, g729a]
Audio Preferences                              : [g722/64k, g722/56k, g722/48k, alaw/64k, alaw/56k, g729, g729a, g729b, g729ab, g729/annex/b, ulaw/64k, ilbc]
```

**chan-sccp version:**
Latest

**asterisk version:**
16.15.1

**Expected behavior:**
```
Audio Preferences : [ilbc]
```

**Observed behavior:**
IMHO the detailed/specific device config should override the general default template config.
The default defines all codecs here, but on this special device I just want to use ilbc.
This seems to be ignored.
Further ilbc and ulaw precedence is switched.

Releated to this:
In which types/sections can codes be selected? global, device, line?
If no codec (option not set) is configured, is this the same like `allow=all` - and what exactly is taken than?

dkgroot, 2021-01-08:
Hi Marco,

I agree with you that if disallow is repeated in the derived section, it should cancel out all previously set values. I did run several tests in the past to make sure this happens, apparently this is not the case at this moment (based on your tests), and it should be fixed.

Note: disallow/allow has move to the line level as of the beginning of last year, to allow you more control over the codec preferences used on a per line basis. The codec on device level has becomes deprecated. 

On Defaults and Inheritance:
- if no codec is specified in the general section we take the absolute minimal set of [alaw/ulaw] during call creation, however in cli output we will show that no preference was specified.
- device and lines inherit from general
- device codec preferences are deprecated
- asterisk config templates do not provide detail about where the data comes from, and only presents a concatenated representation of the section asked for. Which presents some issues getting the information out correctly.

For now i would advice to only set codec selection in the final line section and remove the entry from the device and line templates. I will have to recheck the code to see to it, that your example works the way it's expected to.

dkgroot, 2021-01-08:
Partially related to #553

dkgroot, 2021-01-08:
Branch: `issues/555`

Tested
```
; New Line Template Method
[defaultline](!)                                        ; default template for lines
type = line                                             ; specifies that this template is for lines will be inherited
disallow=all
allow=g722,ilbc

[98061](defaultline)                               
type = line                                             ; specifies that this template is for lines will be inherited
disallow = all                                          ; reset previous set values in the 'defaultline' template
description = test                                      ; mixing other options between disallow/allow options
allow = ulaw                                            ; one per line
allow = alaw
description = label
allow= ulaw,alaw                                        ; no spaced values
allow=g729
allow = !ulaw                                           ; using "!" to invert the meaning of "allow"
allow = !all, ulaw, alaw                                ; using "!" to invert option behaviour, spaced values, mixed
allow = !ulaw                                           ; remove ulaw from previously allowed values
```
Parsing result:
```
[Jan  8 17:51:41] VERBOSE[75771] :     -- SCCP: parsing line parameter: disallow = 'all'  (line: 576)
[Jan  8 17:51:41] VERBOSE[75771] :        > Create new variable set (disallow=all)
[Jan  8 17:51:41] VERBOSE[75771] :        > Add to variable set (allow=g722,ilbc)
[Jan  8 17:51:41] VERBOSE[75771] :        > Add to variable set (disallow=all)
[Jan  8 17:51:41] VERBOSE[75771] :        > Add to variable set (allow=ulaw)
[Jan  8 17:51:41] VERBOSE[75771] :        > Add to variable set (allow=alaw)
[Jan  8 17:51:41] VERBOSE[75771] :        > Add to variable set (allow=ulaw, alaw)
[Jan  8 17:51:41] VERBOSE[75771] :        > Add to variable set (allow=g729)
[Jan  8 17:51:41] VERBOSE[75771] :        > Add to variable set (allow=!ulaw)
[Jan  8 17:51:41] VERBOSE[75771] :        > Add to variable set (allow=!all, ulaw, alaw)
[Jan  8 17:51:41] VERBOSE[75771] :        > Add to variable set (allow=!ulaw)
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: disallow, value:all
[Jan  8 17:51:41] VERBOSE[75771] : SCCP: disallow=all => reset codecs
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: allow, value:g722,ilbc
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'g722/64k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'g722/56k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'g722/48k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'ilbc'
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: disallow, value:all
[Jan  8 17:51:41] VERBOSE[75771] : SCCP: disallow=all => reset codecs
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: allow, value:ulaw
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'ulaw/64k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'ulaw/56k'
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: allow, value:alaw
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'alaw/64k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'alaw/56k'
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: allow, value:ulaw, alaw
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'ulaw/64k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'ulaw/56k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'alaw/64k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'alaw/56k'
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: allow, value:g729
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'g729'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'g729a'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'g729b'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'g729ab'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'g729/annex/b'
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: allow, value:!ulaw
[Jan  8 17:51:41] VERBOSE[75771] :  matched !, token=!ulaw
[Jan  8 17:51:41] VERBOSE[75771] :  removing codec 'ulaw/56k'
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: allow, value:!all, ulaw, alaw
[Jan  8 17:51:41] VERBOSE[75771] :  matched !, token=!all
[Jan  8 17:51:41] VERBOSE[75771] : SCCP: disallow=all => reset codecs
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'ulaw/64k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'ulaw/56k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'alaw/64k'
[Jan  8 17:51:41] VERBOSE[75771] :  appending codec 'alaw/56k'
[Jan  8 17:51:41] VERBOSE[75771] : sccp_config_parse_codec preference: name: allow, value:!ulaw
[Jan  8 17:51:41] VERBOSE[75771] :  matched !, token=!ulaw
[Jan  8 17:51:41] VERBOSE[75771] :  removing codec 'ulaw/64k'
[Jan  8 17:51:41] VERBOSE[75771] :  removing codec 'ulaw/56k'
```
sccp show line 98031
```
Audio Preferences                              : [alaw/64k, alaw/56k]
```

mpaland, 2021-01-09:
Hi Diederik,
thanks a lot for this useful information. I didn't know, that codec selection in the device section has been deprecated. Good to know (and makes sense).

My old setup was just an entry in the device section and as this is deprecated, I don't treat it as an valid setup anymore.

New setup is now:
```
[general]
disallow = all
allow = g722,alaw,g729,ilbc,ulaw

[defaultline](!)
disallow = all
allow = g722,alaw,g729,ilbc,ulaw

[23](defaultline)
disallow = all
allow = ilbc

```
I run several tests with branch/555, and from the point of the config parser is looks fine. Good work!!
I think, you can merge this fix and close this issue after that.

However, there seem to be (new) codec problems which I track in #553 tomorrow.

dkgroot, 2021-01-09:
Hi Marco,
Thanks for switching to line level codecs and testing the config parser change. I will merge the change in a little bot. I will still have to figure out the 553 issue with default values in the general block.

commit fce2f49a (2021-01-09): Fix multivalue config parameters for concatenated entries like disallow|allow

Fixes: #555
closed 2021-01-09 by dkgroot