chan-sccp/chan-sccp #556: Unable to build realtime line. line 1702 does not exists

bknv, 2021-01-20:
I Have is 4 device cisco 7965. SEPA4563041E538,SEPA4563041E763 Link to line 1 extension 1702 and SEPA4563041DDB8,SEPA4563041E3C9 Link to line 1 extension 1623. After start asterisk i get problem with connected line to device.

For example:

After start server, device SEPA4563041E538 (link to 1702) and SEPA4563041E3C9 (link to 1623) load normaly, but device SEPA4563041E763 (link to 1702) and SEPA4563041DDB8 (link to 1623) load but don't have line. Below log with problem.

`[2021-01-19 22:36:06] ERROR[16975]: sccp_line.c:660 __sccp_line_find_realtime_byname: SCCP: Unable to build realtime line '1702'`
`[2021-01-19 22:36:06] WARNING[16975]: sccp_actions.c:1072 sccp_make_button_template: SEPA4563041E763: line 1702 does not exists`
`[2021-01-19 22:36:06] ERROR[16960]: sccp_line.c:660 __sccp_line_find_realtime_byname: SCCP: Unable to build realtime line '1623'`
`[2021-01-19 22:36:06] WARNING[16960]: sccp_actions.c:1072 sccp_make_button_template: SEPA4563041DDB8: line 1623 does not exists`

In realtime all right:
result "realtime load sccpdevice name SEPA4563041E763" - https://pastebin.freepbx.org/view/356fafa4

But In the settings device problem. 
Default line instance                          : 0
result "asterisk -rx 'sccp show device SEPA4563041E763'" - https://pastebin.freepbx.org/view/ad9faa80

After restart phone - all right.
result "asterisk -rx 'sccp show device SEPA4563041E763'" - https://pastebin.freepbx.org/view/0b1287f1

asterisk 13.36.0
Skinny Client Control Protocol (SCCP). Release: 4.3.3 develop - 003663c

dkgroot, 2021-01-20:
@bknv
Just like you check the sccpdevice information, can you run:
`realtime load sccpline name 1702`
`realtime load sccpline name 1623`

bknv, 2021-01-21:
@dkgroot result realtime load sccpline name 1702
https://pastebin.freepbx.org/view/7f471a55

dkgroot, 2021-01-21:
If that does generate output like it seems it does, then the registration process should just work. I don't understand why chan-sccp would not be able to find the sccpline table in question. What does sccp.conf
```
[general]
...
devicetable=
linetable=
```
contain. You should normally not have them in your sccp.conf to get the default values. If you set linetable= to something else then sccpline then it would make sence it wouldn't work.

Please attach: extconfig.conf / res_config_mysql.conf and sccp.conf (and the additional and custom versions of those files as well). I think there is something wrong in one of them.

bknv, 2021-01-25:
[chan-sccp.zip](https://github.com/chan-sccp/chan-sccp/files/5864768/chan-sccp.zip)

dkgroot, 2021-01-25:
@bknv
The extconfig_custom.conf is wrong
```
[settings]
sccpdevice=mysql,sccp,sccpdeviceconfig
sccpline= mysql,sccp,sccpline
```
should be:
```
[settings]
sccpdevice => mysql, sccp, sccpdeviceconfig
sccpline => mysql, sccp, sccpline
```

bknv, 2021-01-29:
@dkgroot Hello, you are right. 
I cannot understand how this could happen ... 
I will be watching this file. If I find a reason then I will definitely inform you

dkgroot, 2021-01-29:
Ok closing the issue for now

bknv, 2021-02-02:
Hello @dkgroot I was a little hasty with conclusions. I apologize
See what i found.

When asterisk start, device SEPA4563041E538 get line 1702, before device SEPA4563041E763 can't get line because "Line '1702' already exists."

`[2021-02-02 22:44:51] VERBOSE[18439] :     -- Added line '1702' to Glob(lines)`
`[2021-02-02 22:44:51] VERBOSE[18439] :     -- SEPA4563041E538: Sending ConfigStatMessage, lines 1, speeddials 0`
`[2021-02-02 22:44:51] VERBOSE[18439] :     -- SEPA4563041E538: Device has 9 Capabilities`
`[2021-02-02 22:44:51] VERBOSE[18418] :     -- SCCP: Line '1702' already exists.`
`[2021-02-02 22:44:51] ERROR[18418] sccp_line.c: SCCP: Unable to build realtime line '1702'`
`[2021-02-02 22:44:51] WARNING[18418] sccp_actions.c: SEPA4563041E763: line 1702 does not exists`

Further, if restart device, then all right

`[2021-02-02 22:45:15] VERBOSE[18418] :  SCCP: Clean Device SEPA4563041E763, remove from global:Yes, restart_device:No`
`[2021-02-02 22:45:15] VERBOSE[18418] :     -- Removed device 'XXXA4563041E763' from Glob(devices)`
`[2021-02-02 22:45:15] VERBOSE[18418] :   == SCCP: Unregister Device XXXA4563041E763`
`[2021-02-02 22:45:15] VERBOSE[18418] :     -- XXXA4563041E763: (devstate::deviceRegisterListener) device unregistered`
`[2021-02-02 22:45:15] VERBOSE[18432] :     -- SCCP: Unregister request Received (Reason: Powersave)`
`[2021-02-02 22:45:18] VERBOSE[18041] :     -- SCCP: Accepted Client Connection from 10.77.23.90:52529`
`[2021-02-02 22:45:18] VERBOSE[18730] :     -- SEPA4563041E763: nat=Off set manually in config file`
`[2021-02-02 22:45:18] VERBOSE[18731] :   == Alarm Type: LastOutOfServiceInformation`
`[2021-02-02 22:45:18] VERBOSE[18731] :   == Reason Enum: 23`
`[2021-02-02 22:45:18] VERBOSE[18731] :   == Last Event Sent: 39:Unregister`
`[2021-02-02 22:45:18] VERBOSE[18731] :   == Last Event Received: 280:UnregisterAck`
`[2021-02-02 22:45:18] VERBOSE[18730] :     -- SEPA4563041E763: Got rtpPort:3500 which the device wants to use for media`
`[2021-02-02 22:45:18] WARNING[18730] sccp_config.c: SCCP: Unknown param at line:0:videomode='off'`
`[2021-02-02 22:45:18] VERBOSE[18730] :     -- Added line '1702' to Glob(lines)`
`[2021-02-02 22:45:18] VERBOSE[18730] :     -- SEPA4563041E763: Sending ConfigStatMessage, lines 1, speeddials 0`

I also cannot understand. I made a copy of the current server and tried to reproduce the problem there, without success.
The only difference is the number of connected phones

dkgroot, 2021-02-04:
@bknv Could it be you have two entries for this line, like one in the database and one in the sccp.conf file or something ?

bknv, 2021-02-04:
In sccp.conf no device records. 
Can they be duplicated somewhere else?

dkgroot, 2021-02-04:
I don't see how/where. You can restart asterisk and make sure no devices register. then do `sccp show devices` and `sccp show lines` it should return an empty list for both, if so you are only using realtime db. If you do get output, then lines/devices are stored in somewhere in /etc/asterisk/sccp*.conf or included into it.

bknv, 2021-02-04:
[image]
I use sccp_manager. Can he influence?

dkgroot, 2021-02-04:
That should not matter. sccp_manager only uses realtime db, And only having "Hotline" in there is perfectly fine. You can switch hotline off, if you don't use it.
So on the first register the line is not found and then on a restart of the phone it is. Very strange. but don't know exactly how that could happen to be honest.

bknv, 2021-02-05:
I don't use hotline. It disable 
[image]
Maybe there is a check for duplicate devices with lines in the sccp code?
Suspicious that it swears SCCP: Line '1702' already exists.

dkgroot, 2021-02-05:
@bknv Maybe we can setup a little ssh connection in the coming days, so i can do a little checking ?

bknv, 2021-02-09:
@dkgroot Yes. I can 10.02. If you will be restart asterisk, then only after 21:00 (+3).

dkgroot, 2021-02-11:
I have committed the patch against the latest version of the develop branch. I will wait for your tests to see if the race condition during realtime line creation is fixed before merging it back into develop.
---
Я применил патч к последней версии ветки разработки. Я буду ждать ваших тестов, чтобы увидеть, исправлено ли состояние гонки во время создания линии в реальном времени, прежде чем объединять его обратно в разработку.

bknv, 2021-02-12:
@dkgroot Hello. I switch to branch issue556
Skinny Client Control Protocol (SCCP). Release: 4.3.3 issues/556 - 85868e5 (built by 'root' on '2021-02-12 19:40:05 UTC')
Line not adding and now in log vanished message about error `sccp_line.c: SCCP: Unable to build realtime line '1702'`
in log has only  
[2021-02-12 22:55:48] WARNING[5110] sccp_actions.c: SEPA4563041E763: line 1702 does not exists

dkgroot, 2021-02-13:
@bknv If everything works fine and you only see this message, then it might just be a spurious warning,  I did not make any change to the source in sccp_action.c. If it doesn't work correctly, let me know.
Note you switching to branch issues/556 will mean you are open to the AGI issue you have seen before.

bknv, 2021-02-13:
@dkgroot This is doesn't work. Line not adding

dkgroot, 2021-02-24:
@bknv Oh sorry it wasn't 100% clear. I am not sure what is causing your issue or what exactly is going on or how to help.