chan-sccp/chan-sccp #557: Access visual parking lot by the Park softkey [enhancement]

tomquinn04, 2021-01-20:
Following a conversation with @dkgroot in gitter

> If you don't have any calls connected on the phone and you press park we could push the visual parkinglot xml to the phone

For using the visual parkinglot without taking up a BLF key for phones such as the 7940 with limited BLF keys.

dkgroot, 2021-01-20:
We could re-use the standard 'park' softkey to show the visual parkinglot information when there is no call present on the phone. It would be similar to the visualparkinglot button functionality with 'RetrieveSingle' park turned off.

dkgroot, 2021-01-25:
posted a first attempt to implement visual parkinglot feature using the park softkey (in branch: "feature/visualpark").

Seems to be working nicely, except when dealing with multiple parked calls on a 8945. The cXML doesn't seem to be processed correctly. This needs a little more research.

- Added sccp.conf new parkinglot field for device to set the parkinglot name, standard-value:"default"
- Added park softket to the onhook softkeyset entry.