chan-sccp/chan-sccp #558: ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.

npanagiotidis81, 2021-01-26:
[Short description of problem here]
hi , while my pbx is working without issues i can call i can recieve calls , on logfile i got this message for every phone (7960g) that got a live line  , my logs is flooding from this , can we think what can be ?
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.
[2021-01-26 15:59:03] NOTICE[20962][C-0000023a] ast111.c: SEP00235EB6C453: Asked to transmit frame type 2 ('dahdi_read') with no samples.

**chan-sccp version:** 4.3.3

**asterisk version: 11

dkgroot, 2021-01-26:
@npanagiotidis81 
I guess this would do the trick:
```diff
diff --git a/src/pbx_impl/ast111/ast111.c b/src/pbx_impl/ast111/ast111.c
index 1cbed173..cf5cbd9d 100644
--- a/src/pbx_impl/ast111/ast111.c
+++ b/src/pbx_impl/ast111/ast111.c
@@ -857,8 +857,8 @@ static int sccp_astwrap_rtp_write(PBX_CHANNEL_TYPE * ast, PBX_FRAME_TYPE * frame
                        }
                        if(!strcasecmp(frame->src, "ast_prod")) {
                                sccp_log((DEBUGCAT_PBX | DEBUGCAT_CHANNEL))(VERBOSE_PREFIX_3 "%s: Asterisk prodded channel %s.\n", c->currentDeviceId, pbx_channel_name(ast));
-                       } else {
-                               pbx_log(LOG_NOTICE, "%s: Asked to transmit frame type %d ('%s') with no samples.\n", c->currentDeviceId, (int)frame->frametype, frame->src);
+                       //} else {
+                       //      pbx_log(LOG_NOTICE, "%s: Asked to transmit frame type %d ('%s') with no samples.\n", c->currentDeviceId, (int)frame->frametype, frame->src);
                        }
                        if (!frame->samples) {
                                if(!strcasecmp(frame->src, "ast_prod")) {
```

npanagiotidis81, 2021-01-26:
well its easy for you , its not easy for me , im a starter ! thanks for the reply but
as my pbx is issabel the location of ast111.c is /usr/src/chan-sccp-chan-sccp-968caa4/src/pbx_impl/ast111/ast111.c
i guess that if i gonna add this lines below it will be ok ? at line 840
[image]

dkgroot, 2021-01-26:
@npanagiotidis81 
Hi there,
Sorry about making assumptions.
In the sccp_astwrap_rtp_write function you can remove lines 860 and 861, which contain:
```
                       } else {
                                pbx_log(LOG_NOTICE, "%s: Asked to transmit frame type %d ('%s') with no samples.\n", c->currentDeviceId, (int)frame->frametype, frame->src);
```
That should suppress the extra logging. In the meantime i will think about removing them in general.

dkgroot, 2021-01-26:
@npanagiotidis81
I have reconsidered the issue, and committed a patch to the repository.
You can do:
```
cd /usr/src/chan-sccp-chan-sccp-968caa4/
git checkout develop    # to undo your local changes
git pull                          # to fetch the latest version
```

closed 2021-01-26 by dkgroot

npanagiotidis81, 2021-01-26:
thanks for the support !

commit e0cb339a (2021-01-26): Fix: Remove extreneous logging regarding receipt of empty frames

Fixes: #558