chan-sccp/chan-sccp #559: AMI SCCPShowSoftkeySets returns duplicate ActionID in first table entry

steve-lad, 2021-01-27:
Using AMI

_Action: SCCPShowSoftkeySets
ActionID: 123123_

The first SCCPSoftKeySetEntry has the ActionID field duplicated.

_Event: SCCPSoftKeySetEntry
ChannelType: SCCP
ActionID: 123123
ChannelObjectType: SoftKeySet
ActionID: 123123
Set: default
Mode: ONHOOK
Description: On Hook
LblID: 0
Label: Redial_

All subsequent SCCPSoftKeySetEntry  are correct with only a single ActionID

_Event: SCCPSoftKeySetEntry
ChannelType: SCCP
ActionID: 123123
ChannelObjectType: SoftKeySet
lSet: default
Mode: ONHOOK
Description: On Hook
LblID: 1
Label: NewCall_

This is not a major issue as should not cause any issues

dkgroot, 2021-01-27:
Created a branch https://github.com/chan-sccp/chan-sccp/tree/issues/559 can you have a look:
Have not found the stack overflow yet.

dkgroot, 2021-01-28:
I know you closed #560 but i came up with a nice solution that will allow use to make modifications like this, without to much trouble in the future.

Enh: Added support for AMI Query Field "TableFormatVersion: 2" on table generating AMI Queries.

This will cause the TableStart Event to output the "TableFormatVersion: 2" when requested, like:
```
Event: TableStart
TableName: SoftKeySets
TableFormatVersion: 2
ActionID: 1234
```

And will change the generated AMI Table output to use: "Event: SoftKeySet_Entry" instead of "Event: SCCPSoftKeySetEntry", like:
```
Event: SoftKeySet_Entry
ChannelType: SCCP
ChannelObjectType: SoftKeySet
Set: default
Mode: ONHOOK
Description: On Hook
LblId: 0
Label: Redial
ActionID: 1234
```

Note: Supplying "TableFormatVersion: 2" on older revisions of chan-sccp will logically have no effect and generate the old output like before.

steve-lad, 2021-01-28:
This is a nice format, but unnecessary. I am working on the receiving end and when I have it right, will ping you

dkgroot, 2021-01-28:
@steve-lad It does provide an optional path forward, in case we have more Format Changes that need to be made compatible on both ends.

***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/559 origin/issues/559
git checkout issues/559
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

dkgroot, 2021-01-28:
Please let me know if everything is working correctly now and if we can merge this change. Thanks for making the issue known to me.

dkgroot, 2021-01-28:
@steve-lad I had to rebase the latest commit and force the subsequent push, to make sure the travis.org->travis.com transition was included (which is besides the point). It you did not pull the previous commit (6hrs ago) you should be fine. If you did pull the previous commit, there might be a slight issue, when trying to pull. You can delete the local branch (using `git branch -d issues/559`) in that case and pull a clean one from upstream (if you need help, let me know).

dkgroot, 2021-02-01:
@steve-lad Thanks for reporting the issue(s). Change has been merged into develop now.

closed 2021-02-01 by dkgroot