chan-sccp/chan-sccp #560: AMI SCCPShowSoftKeySets : No correlation between TableName and subsequent entries.

steve-lad, 2021-01-27:
_action: SCCPShowSoftkeySets
actionid: 123123_

returns as the first message:

_Event: TableStart
TableName: SoftKeySets
ActionID: 123123_

and subsequent entries are

_Event: SCCPSoftKeySetEntry_

It would facilitate subsequent parsing and verification of message integrity if the event was directly linked to the TableName. ie

TableName: SoftKeySet
Event: SoftKeySet_Entry

steve-lad, 2021-01-27:
Please close this issue - it will give enormous backward compatibility issues with SCCP Manager.

closed 2021-01-27 by steve-lad

dkgroot, 2021-01-28:
Provided a backward compatible solution that is available for you to use as of this commit.

Enh: Added support for AMI Query Field "TableFormatVersion: 2" on table generating AMI Queries.

This will cause the TableStart Event to output the "TableFormatVersion: 2" when requested, like:
```
Event: TableStart
TableName: SoftKeySets
TableFormatVersion: 2
ActionID: 1234
```

And will change the generated AMI Table output to use: "Event: SoftKeySet_Entry" instead of "Event: SCCPSoftKeySetEntry", like:
```
Event: SoftKeySet_Entry
ChannelType: SCCP
ChannelObjectType: SoftKeySet
Set: default
Mode: ONHOOK
Description: On Hook
LblId: 0
Label: Redial
ActionID: 1234
```
Note: Supplying "TableFormatVersion: 2" on older revisions of chan-sccp will logically have no effect and generate the old output like before.

steve-lad, 2021-01-28:
Thanks but this is not necessary. Either way have to create a new eventClass so please put on hold for the moment

If you look at the present method, I already handle dynamic event types, but do not manage new event classes. I am working on this