chan-sccp/chan-sccp #562: NOTICE[6113][C-00000001]: ast111.c:860 sccp_astwrap_rtp_write: SEP0011BB30A993: Asked to transmit frame type 2 ('RTP') with no samples.

npanagiotidis81, 2021-01-27:
[Short description of problem here]
hello again !!!!
i did what u proposed me , the previous notice stop to appears but now it appears this one , my machine is issabel pbx , with asterisk 11 and the latest version of sccp 
i got some plans to update to asterisk 16 but till then any help is appriciated .

this appears in log while every sepxxxxx is on active line and for every second that passes

dkgroot, 2021-01-27:
@npanagiotidis81 
Line 860 currently does not have that message any more. Make sure you are on revision:e0cb339 or revision:bbeed397 using
`git log -n1`

If need be, do a clean clone of chan-sccp and run configure/make again.

> i got some plans to update to asterisk 16 but till then any help is appriciated .

:+1:

dkgroot, 2021-02-10:
@npanagiotidis81 Any update ?